litex-hub/linux-on-litex-vexriscv #145: [RFC] Extracting the DT file generation to an external repo

mateusz-holenko, 2020-07-29:
At the moment, `linux-on-litex-vexriscv` uses the [json2dts.py](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/json2dts.py) script to dynamically generate the DT file from the current LiteX configuration.
There are other LiteX-related projects (like [litex-buildenv](https://github.com/timvideos/litex-buildenv), [zephyr-on-litex-vexriscv](https://github.com/litex-hub/zephyr-on-litex-vexriscv) and possibly more in the future) that would also benefit from using this approach.

I know about at least two standalone repositories providing scripts simillar to `json2dts.py` - [litex-devicetree](https://github.com/litex-hub/litex-devicetree) and [litex-renode](https://github.com/litex-hub/litex-renode). 
I was thinking about gathering (and possibly merging together) all those scripts into a single repo that could be reused by all those projects.

The only problem I see is that `linux-on-litex-vexriscv` does not have any dependencies (except for the LiteX itself) and extracting the `json2dts.py` script to an external repo would require the user to install it manually.
The options I see that could address this, are:
* to add the repo with the DT-generator as a git submodule - this could also lead to turning the LiteX into a submodule (in order to improve reproducibility of the setups),
* to add a setup script (similar to the `litex_setup.py` from LiteX) that automatically clones all the dependencies,
* to add the DT-generation directly to LiteX (although I'm not sure if that's the best idea -- at the moment LiteX is rather software-agnostic)

What do you think about this?

enjoy-digital, 2020-07-29:
@mateusz-holenko: that's indeed a good idea to merge things. The initial idea with `json2dts.py` was to prototype things here and integrate it in LiteX later because i'm not sure there is a real need to create a submodule for this. This could be a tool located in `litex/tools`, that woud be installed with LiteX as the other tools and that we would use like this:` litex_json2dts --os=linux (zephyr, etc...) soc.json > soc.dts` ?

mithro, 2020-07-29:
I think pulling it into litex would make sense. We already have the svd generation in upstream litex and multiple things use dts (Linux, Zephyr, -- possible MicroPython in the future, etc)...

mateusz-holenko, 2020-08-05:
The `json2dts.py` script has been moved from linux-on-litex-vexriscv to LiteX.

closed 2020-08-05 by mateusz-holenko