litex-hub/linux-on-litex-vexriscv #148: Update Image Load Command

mescoba1, 2020-08-13:
Looks like `image.json` has been changed to `buildroot/boot.json`, perhaps README should be updated to reflect this. 
```
$ lxterm --images=images.json /dev/ttyUSBX --speed=1e6 --no-crc
```
to 
```
$ lxterm --images=buildroot/boot.json /dev/ttyUSBX --speed=1e6 --no-crc
```

mescoba1, 2020-08-13:
I could be mistaken, looks like `images.json` and `buildroot.json` look similar. Can someone elaborate on the differences?

enjoy-digital, 2020-08-13:
`images.json` is still used for serialboot, `boot.json` is used for netboot/sdcardboot. We could try to avoid having two separate files in the future, but for now the README is still up to date on this.

closed 2020-08-13 by enjoy-digital

mescoba1, 2020-08-13:
Thanks for the clarification