litex-hub/linux-on-litex-vexriscv #152: Generating the Linux binaries (optional)

ToTamir, 2020-09-17:
Generating the Linux binaries don't work on my PC. I succesfully generated Linux binaries around two weeks ago, but since this time i can't do this again. I also deleted all files and setup all again step by step, but i get the same result. Last few lines before error:

```
running configure fragment for sysdeps/unix/sysv/linux
checking installed Linux kernel header files... 3.2.0 or later
configure: WARNING: minimum kernel version reset to 5.4.0
checking for kernel header at least 5.4.0... too old!
configure: error: *** The available kernel headers are older than the requested
*** compatible kernel version
package/pkg-generic.mk:231: recipe for target '/mnt/e/linux/buildroot/output/build/glibc-2.32.9000-69-gbd394d131c10c9ec22c6424197b79410042eed99/.stamp_configured' failed
make: *** [/mnt/e/linux/buildroot/output/build/glibc-2.32.9000-69-gbd394d131c10c9ec22c6424197b79410042eed99/.stamp_configured] Error 1 
```

ToTamir, 2020-09-22:
In [buildroot](https://github.com/buildroot/buildroot) there were two commits [package/glibc: use upstream glibc for RISC-V 32-bit](https://github.com/buildroot/buildroot/commit/a4889545aa653e7f001f212eda169e7d676aefc0) and [configs/qemu_riscv32_virt: use Linux 5.4 kernel](https://github.com/buildroot/buildroot/commit/acc1330d4a838193eb276e02e6642cf297c0e9c1). Last working commit is [linux: add support for ZSTD compression](https://github.com/buildroot/buildroot/commit/88a268354daaaa6422ed875ad8c580dbaf1d81a1) (one before). 

Generating the Linux binaries require [buildroot](https://github.com/buildroot/buildroot) with commit 88a268354daaaa6422ed875ad8c580dbaf1d81a1 or (probably) update [linux-on-litex-vexriscv](https://github.com/litex-hub/linux-on-litex-vexriscv) config files.

enjoy-digital, 2020-09-25:
Thanks @ToTamir for reporting, we'll try to investigate on this soon.

enjoy-digital, 2020-11-05:
https://github.com/litex-hub/linux-on-litex-vexriscv/commit/3156f1290fde3c64897904437c6c7527ab8b19bf adds a reset to the correct buildroot git commit to prevent user to encounter this issue, but we should still try to update the project to upstream version.

Dolu1990, 2020-11-18:
@enjoy-digital About updating the kernel. On a // project, @svancau were trying to update to run linux 5.10.rc3, got it to work, it is still very fresh, we will have to test a bit the interrupts + heavy load to be sure it is stable.

But anyway, we had to fix one HW issue in Vex to get it working. This HW update will come very soon in the SMP branch at the same time than the AES acceleration.

enjoy-digital, 2020-11-18:
@Dolu1990: ah great, thanks for the update!

svancau, 2020-12-11:
@enjoy-digital I have been working on Buildroot for the SaxonSoC, I have a working 5.10-rc7 kernel with SMP.
You can find our working version of Linux here : https://github.com/SpinalHDL/linux/tree/spinal-5.10-latest
And the Buildroot external tree here : https://github.com/SpinalHDL/buildroot-spinal-saxon

Cheers,
Sébastien

enjoy-digital, 2020-12-11:
Thanks for sharing @svancau, this will be useful!

svancau, 2020-12-12:
@enjoy-digital BTW, OpenSBI 0.8 is in the buildroot external tree and works well with Linux

enjoy-digital, 2020-12-15:
This has been updated (using litex-rocket-rebase branch of https://github.com/litex-hub/linux) and will be part of https://github.com/litex-hub/linux-on-litex-vexriscv/pull/165. Upstream Buildroot is used. OpenSBI is still built separately but we could integrate the build to Buildroot in the future if useful.

enjoy-digital, 2020-12-16:
https://github.com/litex-hub/linux-on-litex-vexriscv/pull/165 has been merged, we can now close this.

closed 2020-12-16 by enjoy-digital