litex-hub/linux-on-litex-vexriscv #153: Cannot boot KC705 from sdcard

Logicwax, 2020-09-23:
I can't seem to get my kc705 board to boot from SDcard.   Serial method words fine with lxterm.

I copied the buildroot directory files, and emulator.bin over to the SDcard, which is formatted with mbr partition table and FAT32.  I've tried even without partition tables. 

The bios reports back:

```
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
Ethernet phy mode: GMII
Booting from network...
Local IP : 192.168.1.50
Remote IP: 192.168.1.100
Fetching from: UDP/69
Unable to download Image over TFTP
Network boot failed
Unable to download Linux images, falling back to boot.bin
Unable to download boot.bin over TFTP
Network boot failed
No boot medium found

--============= Console ================--
litex> 
```

What am I missing here?  It's like the bios doesn't even want to try reading from the SDcard.

enjoy-digital, 2020-09-23:
@Logicwax: are you using the prebuilt bitstream? IIRC SDCard was not yet enabled on the last prebuilt bitstreams so you should re-built the bitstream or wait that an update of the prebuilt bitstream (that i could do soon).

Logicwax, 2020-09-25:
@enjoy-digital Yep, with the prebuilt I actually was getting sdram errors.   So building was what I had to do.   here's an overview of my environment:

Ubuntu 20.04
```
sudo apt install build-essential device-tree-compiler wget git python3-setuptools verilator libevent-dev libjson-c-dev libtool automake pkg-config libusb-1.0-0-dev
mkdir litex
cd litex
git clone https://github.com/enjoy-digital/linux-on-litex-vexriscv
wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
chmod +x litex_setup.py
./litex_setup.py init
sudo ./litex_setup.py install
wget https://static.dev.sifive.com/dev-tools/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14.tar.gz
tar -xvf riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14.tar.gz
export PATH=$PATH:$PWD/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/
git clone https://github.com/ntfreak/openocd.git
cd openocd
./bootstrap
./configure --enable-ftdi
make
sudo make install
cd ..
git clone https://github.com/enjoy-digital/linux-on-litex-vexriscv-prebuilt
cp -r linux-on-litex-vexriscv-prebuilt/* linux-on-litex-vexriscv
cd linux-on-litex-vexriscv
./make.py --board=kc705 --build
./make.py --board=kc705 --load
```
For SDcard, I format with gpt/FAT, then
```
cd ..
cp -r linux-on-litex-vexriscv/buildroot/* /media/sdcard/
cp linux-on-litex-vexriscv/emulator/emulator.bin /media/sdcard/
```

After doing the boot command, I then look at UART output:
```
screen /dev/ttyUSB2 500000
````
Which leaves me with this:
```
7, b7: |00000000000000000000000000000000| delays: -
best: m7, b5 delays: 16+-05
SDRAM now under hardware control
Memtest OK
Memspeed Writes: 471Mbps Reads: 540Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
Ethernet phy mode: GMII
Booting from network...
Local IP : 192.168.1.50
Remote IP: 192.168.1.100
Fetching from: UDP/69
Unable to download Image over TFTP
Network boot failed
Unable to download Linux images, falling back to boot.bin
Unable to download boot.bin over TFTP
Network boot failed
No boot medium found

--============= Console ================--
litex> 
```

I was able to boot Linux from sdcard about two months ago, but life got in the way and I totally forgot what I was doing, so when returning to mess around with this, it doesn't seem to work.    Are there any jumper settings I should have on my KC705 board?

enjoy-digital, 2020-09-25:
@Logicwax: thanks for additional information, i'm going to look at this.

enjoy-digital, 2020-09-25:
@Logicwax: i just rebuilt the design with upstream repositories and got it working. Here are attached the bitstream+linux images used and the log of the boot. Can you try to update LiteX and the cores (with litex_setup.py update or manual pull of each dependency)?.

````
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep 25 2020 09:09:26
 BIOS CRC passed (0b11199d)

 Migen git sha1: 06c8ea9
 LiteX git sha1: a1c023b9

--=============== SoC ==================--
CPU:		VexRiscv_Linux @ 125MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		8-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		1048576KiB 64-bit @ 1000Mbps/pin

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  Cmd/Clk scan:
  |00000000000001111111100000000000| best: 13
  Setting Cmd/Clk delay to 13 taps.
  Data scan:
  m0: |01111111111111100000000000| delay: 01
  m1: |01111111111110000000000000| delay: 01
  m2: |00001111111111111000000000| delay: 04
  m3: |00001111111111111000000000| delay: 04
  m4: |00000000011111111111111000| delay: 09
  m5: |00000000011111111111111000| delay: 09
  m6: |00000000000111111111111100| delay: 11
  m7: |00000000000111111111111100| delay: 11
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000000000| delays: -
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |11111110000000000000000000000000| delays: 03+-03
  m0, b4: |00000000001111111111000000000000| delays: 15+-04
  m0, b5: |00000000000000000000000111111111| delays: 28+-04
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b04 delays: 15+-05
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000000000| delays: -
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |11111100000000000000000000000000| delays: 03+-03
  m1, b4: |00000000001111111110000000000000| delays: 14+-04
  m1, b5: |00000000000000000000000011111111| delays: 28+-04
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b04 delays: 15+-04
  m2, b0: |00000000000000000000000000000000| delays: -
  m2, b1: |00000000000000000000000000000000| delays: -
  m2, b2: |00000000000000000000000000000000| delays: -
  m2, b3: |11111000000000000000000000000000| delays: 02+-02
  m2, b4: |00000000011111111100000000000000| delays: 13+-04
  m2, b5: |00000000000000000000001111111110| delays: 26+-04
  m2, b6: |00000000000000000000000000000000| delays: -
[... 149 lines cut ...]
mount: mounting tmpfs on /dev/shm failed: Invalid argument
mount: mounting tmpfs on /tmp failed: Invalid argument
mount: mounting tmpfs on /run failed: Invalid argument
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Saving random seed: [    3.842198] random: dd: uninitialized urandom read (512 bytes read)
OK
Starting network: OK

Welcome to Buildroot
buildroot login: root
                __   _
               / /  (_)__  __ ____ __
              / /__/ / _ \/ // /\ \ /
             /____/_/_//_/\_,_//_\_\
                   / _ \/ _ \
   __   _ __      _\___/_//_/ __         ___  _
  / /  (_) /____ | |/_/__| | / /____ __ / _ \(_)__ _____  __
 / /__/ / __/ -_)>  </___/ |/ / -_) \ // , _/ (_-</ __/ |/ /
/____/_/\__/\__/_/|_|    |___/\__/_\_\/_/|_/_/___/\__/|___/

````

[kc705_2020_09_25.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5281217/kc705_2020_09_25.zip)

Logicwax, 2020-09-25:
Writing your files to disk, and using your bit file it works great!  (thanks!).    But as soon as I go about building it myself, I get:
```
Switching SDRAM to hardware control.
Memtest at 0x40000000...
  Write: 0x40000000-0x40200000 (2/2MiB)
  Read:  0x40000000-0x40200000 (2/2MiB)
  bus errors:  58/256
  addr errors: 0/8192
  data errors: 261094/524288
Memtest KO
Memory initialization failed
```
I decided to just blast away everything, and start fresh.   I wrote a quick bash script to setup my entire environment again.   Is there anything I'm missing here?

```
#!/bin/bash

mkdir litex
cd litex
git clone https://github.com/enjoy-digital/linux-on-litex-vexriscv
wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
chmod +x litex_setup.py
./litex_setup.py init
sudo ./litex_setup.py install
wget https://static.dev.sifive.com/dev-tools/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14.tar.gz
tar -xvf riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14.tar.gz
export PATH=$PATH:$PWD/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/
git clone https://github.com/ntfreak/openocd.git
cd openocd
./bootstrap
./configure --enable-ftdi
make
sudo make install
cd ..
git clone https://github.com/enjoy-digital/linux-on-litex-vexriscv-prebuilt
cp -r linux-on-litex-vexriscv-prebuilt/* linux-on-litex-vexriscv
cd linux-on-litex-vexriscv
./make.py --board=kc705 --build
./make.py --board=kc705 --load
cp linux-on-litex-vexriscv/buildroot/boot.json /media/sdcard/
cp linux-on-litex-vexriscv/buildroot/Image /media/sdcard/
cp linux-on-litex-vexriscv/buildroot/rootfs.cpio /media/sdcard/
cp linux-on-litex-vexriscv/buildroot/rv32.dtb /media/sdcard/
cp linux-on-litex-vexriscv/emulator/emulator.bin /media/sdcard
```

enjoy-digital, 2020-09-25:
Ok, can you provide the Vivado version you are using, the full KC705 boot log (with the DDR3 initialization) and also the bitstream to allow me to do some test on hardware with your version?

Logicwax, 2020-09-25:
Vivado v2020.1 (64-bit) SW Build: 2902540 on Wed May 27 19:54:35 MDT 2020 IP Build: 2902112 on Wed May 27 22:43:36 MDT 2020

[kc705_files.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5282794/kc705_files.zip)

enjoy-digital, 2020-09-25:
The bash script seems correct yes. It seem one of the DRAM module is not initializing correctly on your board with this bitstream (it is working on mine). I'll try to have a closer look in the next days.

Logicwax, 2020-09-26:
whats your vivado version?   do you use a debian OS?

enjoy-digital, 2020-09-28:
@Logicwax: i generated the previous version with Vivado 2018.2 and Ubuntu 18.04. It could be an electrical setting that needs to be adjusted. Can you download https://raw.githubusercontent.com/enjoy-digital/litedram/master/bench/ddr3_mr_gen.py and execute it with: `./ddr3_mr_gen.py --cl=7 --cwl=6`, then copy the sdram_mr_write command to the LiteX BIOS and do a `sdram_cal`?:

````
litex> sdram_mr_write 0 2608

Switching SDRAM to software control.
Writing 0x0a30 to MR0
Switching SDRAM to hardware control.

litex> sdram_mr_write 1 2050

Switching SDRAM to software control.
Writing 0x0802 to MR1
Switching SDRAM to hardware control.

litex> sdram_mr_write 2 520

Switching SDRAM to software control.
Writing 0x0208 to MR2
Switching SDRAM to hardware control.

litex> sdram_cal

Switching SDRAM to software control.
Write leveling:
  Cmd/Clk scan:
  |00000000000001111111100000000000| best: 13
  Setting Cmd/Clk delay to 13 taps.
  Data scan:
  m0: |01111111111111100000000000| delay: 01
  m1: |01111111111110000000000000| delay: 01
  m2: |00001111111111111000000000| delay: 04
  m3: |00001111111111111000000000| delay: 04
  m4: |00000000011111111111111000| delay: 09
  m5: |00000000011111111111111000| delay: 09
  m6: |00000000000111111111111100| delay: 11
  m7: |00000000000111111111111100| delay: 11
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000000000| delays: -
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |11111111000000000000000000000000| delays: 04+-04
  m0, b4: |00000000000111111111000000000000| delays: 15+-05
  m0, b5: |00000000000000000000000011111111| delays: 28+-04
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b04 delays: 15+-04
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000000000| delays: -
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |11111110000000000000000000000000| delays: 03+-03
  m1, b4: |00000000001111111110000000000000| delays: 14+-04
  m1, b5: |00000000000000000000000011111111| delays: 28+-04
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b04 delays: 14+-04
  m2, b0: |00000000000000000000000000000000| delays: -
  m2, b1: |00000000000000000000000000000000| delays: -
  m2, b2: |00000000000000000000000000000000| delays: -
  m2, b3: |11111000000000000000000000000000| delays: 02+-02
  m2, b4: |00000000011111111100000000000000| delays: 13+-04
  m2, b5: |00000000000000000000000111111100| delays: 27+-04
  m2, b6: |00000000000000000000000000000000| delays: -
  m2, b7: |00000000000000000000000000000000| delays: -
  best: m2, b04 delays: 13+-04
  m3, b0: |00000000000000000000000000000000| delays: -
  m3, b1: |00000000000000000000000000000000| delays: -
  m3, b2: |00000000000000000000000000000000| delays: -
  m3, b3: |11111000000000000000000000000000| delays: 02+-02
  m3, b4: |00000000111111111000000000000000| delays: 12+-04
  m3, b5: |00000000000000000000011111111100| delays: 26+-04
  m3, b6: |00000000000000000000000000000000| delays: -
  m3, b7: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: 12+-04
  m4, b0: |00000000000000000000000000000000| delays: -
  m4, b1: |00000000000000000000000000000000| delays: -
  m4, b2: |00000000000000000000000000000000| delays: -
  m4, b3: |00000000000000000000000000000000| delays: -
  m4, b4: |00111111111100000000000000000000| delays: 07+-05
  m4, b5: |00000000000000011111111110000000| delays: 20+-05
  m4, b6: |00000000000000000000000000000111| delays: 30+-01
  m4, b7: |00000000000000000000000000000000| delays: -
  best: m4, b04 delays: 07+-05
  m5, b0: |00000000000000000000000000000000| delays: -
  m5, b1: |00000000000000000000000000000000| delays: -
  m5, b2: |00000000000000000000000000000000| delays: -
  m5, b3: |00000000000000000000000000000000| delays: -
  m5, b4: |00111111111000000000000000000000| delays: 06+-04
  m5, b5: |00000000000000011111111100000000| delays: 19+-04
  m5, b6: |00000000000000000000000000000111| delays: 30+-01
  m5, b7: |00000000000000000000000000000000| delays: -
  best: m5, b04 delays: 07+-05
  m6, b0: |00000000000000000000000000000000| delays: -
  m6, b1: |00000000000000000000000000000000| delays: -
  m6, b2: |00000000000000000000000000000000| delays: -
  m6, b3: |00000000000000000000000000000000| delays: -
  m6, b4: |11111111000000000000000000000000| delays: 04+-04
  m6, b5: |00000000000001111111100000000000| delays: 17+-04
  m6, b6: |00000000000000000000000000111111| delays: 29+-03
  m6, b7: |00000000000000000000000000000000| delays: -
  best: m6, b04 delays: 04+-04
  m7, b0: |00000000000000000000000000000000| delays: -
  m7, b1: |00000000000000000000000000000000| delays: -
  m7, b2: |00000000000000000000000000000000| delays: -
  m7, b3: |00000000000000000000000000000000| delays: -
  m7, b4: |11111111110000000000000000000000| delays: 05+-05
  m7, b5: |00000000000011111111111000000000| delays: 17+-05
  m7, b6: |00000000000000000000000001111111| delays: 28+-03
  m7, b7: |00000000000000000000000000000000| delays: -
  best: m7, b05 delays: 17+-05
Switching SDRAM to hardware control.

litex> 
````

and then also test commands from: `./ddr3_mr_gen.py --cl=7 --cwl=6 --rtt_nom=disabled` to see if you are able to have a valid read leveling on all modules? (delays found during the read leveling).

If this isn't working, can you try with other electical settings? `./ddr3_mr_gen.py  --list` will list the supported ones.

Logicwax, 2020-09-29:
Perhaps its the fact that I have newer Vivado version?

Not sure what I'm looking for exactly....

I ran litex/litedram/bench/ddr3_mr_gen.py --cl=7 --cwl=6:
```
DDR3 Timing Settings:
cl:  7
cwl: 6
DDR3 Electrical Settings:
rtt_nom: 60ohm
rtt_wr:  60ohm
ron:     34ohm
Commands to be used with LiteX BIOS:
sdram_mr_write 0 2608
sdram_mr_write 1 2054
sdram_mr_write 2 520
```

So then I typed those three commands into the BIOS, hit enter after pasting each one in, and got this:
```
lite>sdram_mr_write 0 2608

Switching SDRAM to software control.

Writing 0x0a30 to MR0

Switching SDRAM to hardware control.

lite>sdram_mr_write 1 2054

Switching SDRAM to software control.

Writing 0x0806 to MR1

Switching SDRAM to hardware control.

lite>sdram_mr_write 2 520

Switching SDRAM to software control.

Writing 0x0208 to MR2

Switching SDRAM to hardware control.

lite>sdram_cal

Switching SDRAM to software control.

Write leveling:

  Cmd/Clk scan:

  |00000000000000111111110000000000| best: 14

  Setting Cmd/Clk delay to 14 taps.

  Data scan:

  m0: |00111111111111100000000000| delay: 02

  m1: |01111111111111000000000000| delay: 01

  m2: |00000111111111111100000000| delay: 05

  m3: |00000111111111111100000000| delay: 05

  m4: |00000000001111111111111100| delay: 10

  m5: |00000000001111111111111000| delay: 10
[... 386 lines cut ...]
 - 13
 - 14
cwl:
 - 5
 - 6
 - 7
 - 8
 - 9
 - 10
Supported DDR3 Electrical settings:
rtt_nom:
 - disabled
 - 60ohm
 - 120ohm
 - 40ohm
 - 20ohm
 - 30ohm
rtt_wr:
 - disabled
 - 60ohm
 - 120ohm
ron:
 - 40ohm
 - 34ohm
```

Logicwax, 2020-09-29:
I should also mention, that when I manually force a sdcardboot from the bios I get this:
```
litex> sdcardboot

Booting from SDCard in SD-Mode...
Booting from boot.json...
Copying Image to 0x40000000 (4545524 bytes)...
[########################################]
Copying rootfs.cpio to 0x40800000 (8029184 bytes)...
[########################################]
Copying rv32.dtb to 0x41000000 (2231 bytes)...
[########################################]
Copying emulator.bin to 0x41100000 (9724 bytes)...
[########################################]
Executing booted program at 0x41100000

--============= Liftoff! ===============--
```
and it just hangs there...

Logicwax, 2020-09-30:
@enjoy-digital could you test building with Vivado v2020.1?

enjoy-digital, 2020-10-01:
@Logicwax: the boot is hanging because the DDR3 has not been initialized correctly, so there is some data corruption.
I will try to test with Vivado 2020.1 in the next days. Could you also redo a test with upstream LiteX/LiteDRAM (i'm working on some improvements).

Logicwax, 2020-10-02:
by upstream, you mean
`cd litedram`,` git checkout master`, `git pull `?
i also ran the `./litex_setup.py update` script in the parent directory too.  I rebuilt, and have the same errors.

Logicwax, 2020-10-02:
@enjoy-digital I just noticed you commited more code to the litedram repo.   I just upgraded that repo to b16e9bfd5c4d5f9a49c80e6190ba755d74172b7a and rebuilt, and now its working!   So whatever you just did fixed my problem.   Would be interested to know what the fix was.

enjoy-digital, 2020-10-03:
@Logicwax: thanks for testing and for the feedback. The fix is probably https://github.com/enjoy-digital/litedram/commit/16480d9aed87db71ef4e95630a1b945753a3bc8d that adjusts the DQ/DQS Tristrate control. Can you test without this to confirm it?

Logicwax, 2020-10-03:
@enjoy-digital yup!  Confirmed this is what breaks / fixes it.    Thanks so much for looking in this!      

why did this affect just me, but yours worked fine?

enjoy-digital, 2020-10-04:
@Logicwax: thanks for the feedback, i found this while doing some cleanup to the PHYs before adding new features. There are small variations between boards and i didn't noticed the issue before since the boards i was testing with were not exhibiting it. (the fact DQs/DQs are controlled for 3 `sys_clk` cycles + the software calibration was probably masking it).

closed 2020-10-04 by enjoy-digital