litex-hub/linux-on-litex-vexriscv #154: DE10-Lite Memory initialization failed (?)

pphilippos, 2020-10-06:
Hello, (this is a new user speaking)

I was trying to load the prebuilt bitstream on my board (Intel DE10-Lite) and works fine, but when I built it with Quartus it fails.

The closest information I could find is this, although I tried the suggestions here http://lists.lca2018.linux.org.au/pipermail/fpga-miniconf/2018-January/000055.html . I also tried to modify the rom_size/sdram_size values to match the first bitstream with no success. What could I be missing? 

Many thanks,

Text from the pre-built bitstream:
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Mar 23 2020 16:51:09
 BIOS CRC passed (04debf02)

 Migen git sha1: 3f9809b
 LiteX git sha1: 4adac90d

--=============== SoC ==================--
CPU:       VexRiscv @ 50MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  65536KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under hardware control
Memtest OK
Memspeed Writes: 110Mbps Reads: 147Mbps

--============== Boot ==================--
...

Text from the pre-built bitstream:

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Oct  6 2020 00:40:07
 BIOS CRC passed (c5678997)

 Migen git sha1: --------
 LiteX git sha1: 6916674f

--=============== SoC ==================--
CPU:		VexRiscv_Linux @ 50MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		8-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		65536KiB 16-bit @ 50MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000...
  Write: 0x40000000-0x40200000 (2/2MiB)
  Read:  0x40000000-0x40200000 (2/2MiB)
  bus errors:  256/256
  addr errors: 8191/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed
...

Quartus version: Version 20.1.0 Build 711 06/05/2020 SJ Lite Edition
One warning after the bitstream generation: (could be "normal"(?))
build/de10lite/de10lite.dts:63.45-68.19: Warning (simple_bus_reg): /soc/interrupt-controller: missing or empty reg/ranges property

enjoy-digital, 2020-11-05:
Hi @pphilippos,

sorry for the delay. I don't have the hardware to investigate on this, but i just re-generated the bistream on my machine to first try to understand if this is related to a LiteX/LiteDRAM regression or to the version of Quartus. Could  you try this bitstream on your hardware? [de10lite_2020_11_05.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5492905/de10lite_2020_11_05.zip) Depending on the results, i could provide other bitstreams to test to understand the regression.

pphilippos, 2020-11-05:
Hello, sure here is the output for initalisation this time.

--========== Initializtching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
MiB   e: 0x40000000-0x401e0000 1MiB   e: 0x40000000-0x401a0000 1MiB   e: 0x40000000-0x400c0000 768KiB   
MiB   d: 0x40000000-0x401c0000 1MiB   d: 0x40000000-0x40120000 1MiB     
  bus errors:  256/256
  addr errors: 8191/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

enjoy-digital, 2020-11-05:
@pphilippos: thanks, so this is a regression. Here is a new bitstream where i reverted a change in LiteDRAM that i suspect could cause this, can you test it? [de10lite_2020_11_05_10h43.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5493372/de10lite_2020_11_05_10h43.zip)

pphilippos, 2020-11-05:
thanks, this did not seem to work either
--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
  bus errors:  256/256
  addr errors: 8191/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

enjoy-digital, 2020-11-05:
Thanks, so this could be related to the CRG changes. Can you try this one with the same CRG than the prebuilt version? 
[de10lite_2020_11_05_11h11.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5493531/de10lite_2020_11_05_11h11.zip)

pphilippos, 2020-11-05:
sure, same result :)
--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
  bus errors:  256/256
  addr errors: 8191/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

enjoy-digital, 2020-11-05:
OK, i now also reverted the platform file: [de10lite_2020_11_05_11h30.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5493641/de10lite_2020_11_05_11h30.zip) If this is still not working, i'll try to test on a similar board (MAX10/SDRAM) or will order the De10Lite.

pphilippos, 2020-11-05:
It is still not working. It is easy for me if you want me to try some more bistreams. (note that the pre-built bitstream in the github project was working fine)

 BIOS built on Nov  5 2020 11:25:30
 BIOS CRC passed (042f129d)

 Migen git sha1: cb88406
 LiteX git sha1: 3dffdbf6

--=============== SoC ==================--
CPU:            VexRiscv_Linux @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            8-bit data
ROM:            64KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          65536KiB 16-bit @ 50MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB
   Read: 0x40000000-0x40200000 2MiB
  bus errors:  256/256
  addr errors: 8191/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

enjoy-digital, 2021-03-30:
The DE10Lite support has been removed for now with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/ba487402cf15dcbfe6093893933477e70fbbffe2. To Re-enable it, someone would need to investigate the issue or send the hardware.

closed 2021-03-30 by enjoy-digital

robinsonb5, 2022-01-15:
I have a DE10-Lite and a pretty good working knowledge of SDRAM.  Can you offer any hints as to where I should start in trying to get this working?

pphilippos, 2022-01-23:
@robinsonb5 Thank you for your contribution. I personally do not know/remember more than what has been discussed above, as I now mainly use other tools and platforms.

Lefteris-B, 2022-08-07:
> 65536KB

What arguments did you use here? 
if I use --integrated-main-ram  with >0x8000 I always get an error.