litex-hub/linux-on-litex-vexriscv #156: Error: unknown type name __uint128_t while building uclibc 1.0.36 in buildroot

kmezynski, 2020-11-03:
After generating Linux binaries from [buildroot GH repository](http://github.com/buildroot/buildroot) as it is described in the [README](https://github.com/litex-hub/linux-on-litex-vexriscv#generating-the-linux-binaries-optional):
```
git clone http://github.com/buildroot/buildroot
cd buildroot
make BR2_EXTERNAL=../linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig
make
```
I get the following error during the **uclibc 1.0.36 build process** (after executing make):
```
<built-in>: note: this is the location of the previous definition
In file included from ./include/sys/procfs.h:34,
                 from ./libpthread/nptl/../nptl_db/thread_db.h:28,
                 from ./libpthread/nptl/descr.h:30,
                 from ./libpthread/nptl/pthreadP.h:25,
                 from <stdin>:2:
./include/sys/user.h:32:3: error: unknown type name '__uint128_t'
   __uint128_t  vregs[32];
   ^~~~~~~~~~~
In file included from /home/kamil/git/linux-on-litex-vexriscv-build/buildroot/output/build/linux-headers-0192f8256a3966ac16ab5a2f3d88a2a4323078ca/usr/include/asm/sigcontext.h:17,
                 from ./include/bits/sigcontext.h:30,
                 from ./include/signal.h:396,
                 from ./libc/sysdeps/linux/common/internal-signals.h:17,
                 from ./libpthread/nptl/pthreadP.h:33,
                 from <stdin>:2:
/home/kamil/git/linux-on-litex-vexriscv-build/buildroot/output/build/linux-headers-0192f8256a3966ac16ab5a2f3d88a2a4323078ca/usr/include/asm/ptrace.h:27:8: error: redefinition of 'struct user_regs_struct'
 struct user_regs_struct {
        ^~~~~~~~~~~~~~~~
In file included from ./include/sys/procfs.h:34,
                 from ./libpthread/nptl/../nptl_db/thread_db.h:28,
                 from ./libpthread/nptl/descr.h:30,
                 from ./libpthread/nptl/pthreadP.h:25,
                 from <stdin>:2:
./include/sys/user.h:22:8: note: originally defined here
 struct user_regs_struct
        ^~~~~~~~~~~~~~~~
```
It seems during the build 32-bit gcc is being used, which doesn't support `__uint128_t` type. I've tried to use 64-bit riscv toolchain extracted on my machine from this repository by setting **buildroot's menuconfig** manually, which haven't worked work well. After I've adjusted the settings to use the external toolchain (that I'm using on my machine) instead of the built-in one, I'm getting error:
```
Cannot execute cross-compiler '/opt/riscv64/bin/riscv64-unknown-elf-gcc/bin/riscv32-linux-gcc'
```
Which is surprising, since I've set in menuconfig path to `/opt/riscv64/bin` only. The ramaining path is being added without regard to the set toolchain's path in the menuconfig. To make sure I've extracted my toolchain and added it to the PATH properly, I've made simple C file to test whether gcc understands `__uint128_t` type. Program compiled successfully.
Could you give me any suggestion how this issue could be solved? I would appreciate any kind of help :slightly_smiling_face:
I'm using Debian 10 Buster.

kmezynski, 2020-11-04:
I've tested buildroot's commit 88a268354daaaa6422ed875ad8c580dbaf1d81a1 from 2020-09-03, which turned out to be working. I'd suggest adjusting README until this problem is fixed... even more, to guarantee no errors during any tutorial like this, there should be repositories' specific releases instead of simply cloning master branch like so:
```
git clone http://github.com/buildroot/buildroot
cd buildroot
git reset --hard 88a268354daaaa6422ed875ad8c580dbaf1d81a1
make BR2_EXTERNAL=../linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig
make
```

enjoy-digital, 2020-11-05:
Thanks @kmezynski, this has been integrated in the README with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/3156f1290fde3c64897904437c6c7527ab8b19bf. I agree with you on using fixed commit for dependencies, since this project wa initially just an experiment we were using upstream version of the tools to simplify things (and also has a way to catch regressions) but we should indeed now try to improve this.

enjoy-digital, 2020-11-05:
I'm closing this but we can continue the discussion in https://github.com/litex-hub/linux-on-litex-vexriscv/issues/152.

closed 2020-11-05 by enjoy-digital