litex-hub/linux-on-litex-vexriscv #159: AES custom instruction

Dolu1990, 2020-11-18:
I'm currently working on adding a custom instruction to acceleration AES in the SMP branch.

```
SoC configuration :
./make.py --board=arty_a7 --cpu-count=2 --dcache-width=64 --icache-width=64 --dcache-size=8192 --dcache-ways=2 --icache-size=8192 --icache-ways=2 --aes-instruction=True --local-ip=192.168.0.159 --remote-ip=192.168.0.24 --build --load

Test :
openssl speed -elapsed -evp aes-128-cbc aes-256-cbc

SW :
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
aes-256-cbc        466.55k      683.84k      787.37k      829.75k      838.31k      838.25k
aes-128-cbc        532.21k      798.63k      945.24k     1002.84k     1001.55k     1003.66k

HW AES custom instruction :
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
aes-256 cbc       1781.52k     2834.07k     3323.07k     3486.72k     3465.22k     3440.10k
aes-128-cbc       1408.26k     2907.39k     3990.78k     4392.96k     4398.10k     4390.91k
```
So a good 4x gain :)

Nothing is pushed yet excepter the Aes CPU Plugin :
https://github.com/SpinalHDL/VexRiscv/blob/dev/src/main/scala/vexriscv/plugin/AesPlugin.scala

The foot print is light : 
Per CPU : 2KB ROM + 90 LUT + 75 FF

Question now is how do we want to integrate it into buildroot.
Basicaly, it require a patch for libopenssl and dropbear. Question is :
Is there a way (already used) to make the buildroot build depend on the SoC configuration ?

enjoy-digital, 2020-11-18:
Great  @Dolu1990, we currently use the same buildroot defconfig for all configurations, we could try to generate the defconfig dynamically. As an intermediate step we could for now create an AES version with custom rules to build the patched lipopenssl/dropbear.

Dolu1990, 2020-12-11:
Added a few comments on how the Aes instruction works : 
https://github.com/SpinalHDL/VexRiscv/blob/dev/src/main/scala/vexriscv/plugin/AesPlugin.scala#L8

Also, i ported it to libressl. 

To test it, you need the following in the defconfig : 

```
BR2_PACKAGE_LIBATOMIC_OPS_ARCH_SUPPORTS=y
BR2_PACKAGE_LIBATOMIC_OPS=y
BR2_PACKAGE_OPENSSL=y
BR2_PACKAGE_LIBRESSL=y
BR2_PACKAGE_LIBRESSL_BIN=y
BR2_PACKAGE_HAVEGED=y
BR2_PACKAGE_VEXRISCV_AES=y
```

If you want to turn off the hardware acceleration in software, just turn BR2_PACKAGE_VEXRISCV_AES to n

enjoy-digital, 2020-12-11:
Thanks @Dolu1990, as discussed together, I'll try to reproduce your results in the next days.

bunnie, 2021-01-03:
@Dolu1990 do you happen to have a link to your port to libressl? An example of a library that uses these instructions would be really helpful for building a driver for us.

Dolu1990, 2021-01-03:
Hi @bunnie,

There it is : 
https://github.com/litex-hub/linux-on-litex-vexriscv/blob/vexriscv-smp/buildroot/patches/libressl/0000-vexriscv-aes.patch

it contains the instruction driver (https://github.com/litex-hub/linux-on-litex-vexriscv/blob/vexriscv-smp/buildroot/patches/libressl/0000-vexriscv-aes.patch#L1042)

And basicaly, the port in buildroot add a new option in libressl : 
https://github.com/litex-hub/linux-on-litex-vexriscv/blob/vexriscv-smp/buildroot/patches/libressl/0000-vexriscv-aes.patch#L8

and copy the aes_core into aes_core_vexriscv with the proper modifications : 
https://github.com/litex-hub/linux-on-litex-vexriscv/blob/vexriscv-smp/buildroot/patches/libressl/0000-vexriscv-aes.patch#L14

Dolu1990, 2021-01-03:
@bunnie About the endianness of the expended key, it is the same than linux kernel crypto, but the reverse than default openssl/libressl/dropbear. That's why https://github.com/litex-hub/linux-on-litex-vexriscv/blob/vexriscv-smp/buildroot/patches/libressl/0000-vexriscv-aes.patch#L1054 was added.

Logicwax, 2021-02-05:
this is amazing!  would love to see efforts to add sha256 support as well!

closed 2021-04-01 by Dolu1990

amr-25, 2022-08-23:
@Dolu1990, Suppose I want to add my own instruction, for example: https://github.com/SpinalHDL/VexRiscv/tree/dev#add-a-custom-instruction-to-the-cpu-via-the-plugin-system, What are the necessary changes to be made on the buildroot side while doing a make with BR2 external= def_config. I assume, only then I could compile the C code with the custom instruction. Thank you

Dolu1990, 2022-08-23:
> What are the necessary changes to be made on the buildroot side

None, i mean, in general, if i do a custom instruction, the way i use it in software is by forcing its opcode in the binary. I do not rely on GCC being able to support it.

Here is an example : 
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/driver/aes_custom.h#L58
defined :
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/driver/aes_custom.h#L5
defined : 
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/driver/riscv.h#L205

Then, if you want to add support into the toolchaine in a "clean" way, i realy have no idea ^^

amr-25, 2022-08-23:
> 

Curious about BR2_PACKAGE_LIBATOMIC_OPS_ARCH_SUPPORTS=y
BR2_PACKAGE_LIBATOMIC_OPS=y , why are they needed? 

I have an application which has __sync__compare__and__swap function call (works fine with g++ compiler). And it fails during riscv compilation saying: undefined reference to '__sync__compare__and__swap_8'. I tried including atomic library in riscv32-buildroot-linux-gnu/10.04, which has alternative atomic_compare_exchange. 

So, maybe the BR2_PACKAGE_LIBATOMIC_OPS=y might help? Right now BR2_PACKAGE_LIBATOMIC_OPS is commented. 

Thank you

Dolu1990, 2022-08-23:
@amr-25 Hoo i realy have no idea / opinon about this, i probably copy paster stuff found on the web ^^
Else, it is possible that at a given moment, that package was broken for RISC-V 32 bits, and didn't compiled.
A few years ago, things where much less "stable" than now ^^

mntmn, 2025-08-29:
BTW for the current buildroot with openssl, the first hunk in `0000-vexriscv-aes.patch` needs to be adjusted / rebased slightly.