litex-hub/linux-on-litex-vexriscv #161: Linux s

abdullahyildiz, 2020-11-25:


abdullahyildiz, 2020-11-25:
I am trying to boot Linux on ZCU104 evaluation board. However, UART terminal hangs after the `Liftoff` message.

`$ lxterm /dev/ttyUSB4 --images=images.json --serial-boot --speed=1e6 --no-crc`

```
         __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Nov 24 2020 11:03:47
 BIOS CRC passed (5c61a71b)

 Migen git sha1: a5cc037
 LiteX git sha1: 8e39060d

--=============== SoC ==================--
CPU:		VexRiscv_Linux @ 125MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		8-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		1048576KiB 64-bit @ 1000MT/s (CL-9 CWL-9)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  Cmd/Clk scan:
  |00000000  |0000  |0000  |0000| best: -1
  Setting Cmd/Clk delay to -1 taps.
  Data scan:
  m0: |1111111111111111111111| delay: -
  m1: |1111111111111111111111| delay: -
  m2: |1111111111111111111111| delay: -
  m3: |1111111111111111111111| delay: -
  m4: |1111111111111111111111| delay: -
  m5: |1111111111111111111111| delay: -
  m6: |1111111111111111111111| delay: -
  m7: |1111111111111111111111| delay: -
Write latency calibration:
m0:0 m1:0 m2:0 m3:0 m4:0 m5:0 m6:0 m7:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000000000| delays: -
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000000000| delays: -
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b00 delays: -
  m2, b0: |00000000000000000000000000000000| delays: -
  m2, b1: |00000000000000000000000000000000| delays: -
  m2, b2: |00000000000000000000000000000000| delays: -
  m2, b3: |00000000000000000000000000000000| delays: -
[... 57 lines cut ...]
Memtest KO
Memory initialization failed

--============= Console ================--

litex> serialboot

Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0x40000000 (4545524 bytes)...
[LXTERM] Upload complete (86.2KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x41100000 (9600 bytes)...
[LXTERM] Upload complete (86.3KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0x40800000 (8029184 bytes)...
[LXTERM] Upload complete (86.2KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0x41000000 (1987 bytes)...
[LXTERM] Upload complete (85.3KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
KExecuting booted program at 0x41000000

--============= Liftoff! ===============--
```

enjoy-digital, 2020-11-27:
Hi @abdullahyildiz,

the DDR4 is not calibrating correctly, so it will not work. It seems the ZCU104 does not comes with DIMM in the SODIMM Socket (https://www.xilinx.com/products/boards-and-kits/zcu104.html#hardware: *DIMM not included with kit). Have you added one? If so, which one are you using? I don't have a ZCU104 personnaly, but know it has been tested recently.

abdullahyildiz, 2020-11-28:
@enjoy-digital,

You are right. This is the first time that I use this board and I missed that point.

I will try to boot Linux again after plugging a suitable DDR4 memory into SODIMM socket.

closed 2020-11-28 by abdullahyildiz

sususjysjy, 2023-10-06:
> @enjoy-digital,
> 
> You are right. This is the first time that I use this board and I missed that point.
> 
> I will try to boot Linux again after plugging a suitable DDR4 memory into SODIMM socket.

Sorry to bother you, I saw this issue today and I would like to ask how you found the appropriate phy in LitedRAM for the sodim you inserted

sususjysjy, 2023-10-06:
> Hi @abdullahyildiz,
> 
> the DDR4 is not calibrating correctly, so it will not work. It seems the ZCU104 does not come with a DIMM in the SODIMM socket (https://www.xilinx.com/products/boards-and-kits/zcu104.html#hardware: *DIMM not included with kit). Have you added one? If so, which one are you using? I don't have a ZCU104 personnaly, but know it has been tested recently.

Sorry to bother you, I saw this issue today and I would like to ask how you found the appropriate phy in LitedRAM for the sodim you inserted