litex-hub/linux-on-litex-vexriscv #164: Prebuilt Bitstreams and Linux/OpenSBI images.

enjoy-digital, 2020-12-15:
Prebuilt Bitstreams and Linux/OpenSBI images can be found below to get started quickly and easily:

Linux (uncompress in buildroot directory):
[linux_2020_12_15.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5695229/linux_2020_12_15.zip)

OpenSBI (uncompress in opensbi directory):
[opensbi_2020_12_15.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5695244/opensbi_2020_12_15.zip)

SanadaShinken, 2020-12-17:
Hi Sir:

when I use linux_2020_12_15.zip on qmtech wukong board, the Linux kernel will boot failed.
but using https://github.com/litex-hub/linux-on-litex-vexriscv/files/5281217/kc705_2020_09_25.zip, this version will boot successfully.

BR, Sanada

frantony, 2021-03-27:
@enjoy-digital 

Please attach device tree file.