litex-hub/linux-on-litex-vexriscv #167: Boot issue on qmtech wukong

enjoy-digital, 2020-12-17:
Hi Sir:

when I use linux_2020_12_15.zip on qmtech wukong board, the Linux kernel will boot failed.
but using https://github.com/litex-hub/linux-on-litex-vexriscv/files/5281217/kc705_2020_09_25.zip, this version will boot successfully.

BR, Sanada

_Originally posted by @SanadaShinken in https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164#issuecomment-747561179_

enjoy-digital, 2020-12-17:
Hi @SanadaShinken,

the prebuilt linux_2020_12_15 is only compatible with >= 2020_12_15 bitstreams. You will have to rebuild the bitstream for your board to use it.

SanadaShinken, 2020-12-19:
> Hi @SanadaShinken,
> 
> the prebuilt linux_2020_12_15 is only compatible with >= 2020_12_15 bitstreams. You will have to rebuild the bitstream for your board to use it.

Hi @enjoy-digital :+1: 

Yes, I did the update to 2020-1218 for all of the litex related project. 
the file is the boot up message. the most interesting parts in the file is the following part.
[    0.000000] Failed to find INTC node [/cpus/cpu@0]
[    0.000000] Failed to initialize '/cpus/cpu@0': -19
[    0.000000] timer_probe: no matching timers found

wukong_boot_ok.txt is using kc705_2020_09_25.zip's Image and rootfs.cpio
[wukong_boot_ok.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5719622/wukong_boot_ok.txt)

in the workable Image and rootfs.cpio, there are some messages about litex-spiflash show that the device did not work properly. 

[    4.595471] litex-spiflash f0005800.spiflash: unrecognized JEDEC id bytes: ff, ff, ff
[    4.596087] litex-spiflash f0005800.spiflash: SPI_NOR_SCAN FAILED
[    4.596929] litex-spiflash: probe of f0005800.spiflash failed with error -2

BTW, the DDR3 memory R/W speed is very slow with compare to using the same DDR3 chip, such as like atry-7. is there any idea about improve memory access speed?

BR, Sanada

SanadaShinken, 2020-12-26:
Hi @enjoy-digital :

Many Thanks for Litex and @enjoy-digital !!! Greate Work!!

finally, Using  liunx image which built @ 20201215 is booted!!

this issue could be closed.

BR, Sanada

enjoy-digital, 2021-01-04:
Closing since resolved.

closed 2021-01-04 by enjoy-digital