litex-hub/linux-on-litex-vexriscv #169: vexrisc-smp size issue on OrangeCrab

geertu, 2020-12-18:
Commit 55306a3ebc302656 ("Merge branch 'vexriscv-smp'") broke the build for OrangeCrab, even with --cpu-count=1:

 Info: Device utilisation:
-Info: 	       TRELLIS_SLICE: 10338/12144    85%
-Info: 	          TRELLIS_IO:    68/  197    34%
+Info: 	       TRELLIS_SLICE: 11537/12144    95%
+Info: 	          TRELLIS_IO:    70/  197    35%
 Info: 	                DCCA:     5/   56     8%
-Info: 	              DP16KD:    33/   56    58%
+Info: 	              DP16KD:    66/   56   117%
 Info: 	          MULT18X18D:     4/   28    14%
 Info: 	              ALU54B:     0/   14     0%
 Info: 	             EHXPLLL:     2/    2   100%
 Info: 	             EXTREFB:     0/    1     0%
 Info: 	                DCUA:     0/    1     0%
 Info: 	           PCSCLKDIV:     0/    2     0%
-Info: 	             IOLOGIC:    49/  128    38%
+Info: 	             IOLOGIC:    55/  128    42%
 Info: 	            SIOLOGIC:     0/   69     0%
 Info: 	                 GSR:     0/    1     0%
 Info: 	               JTAGG:     0/    1     0%
 Info: 	                OSCG:     0/    1     0%
 Info: 	               SEDGA:     0/    1     0%
 Info: 	                 DTR:     0/    1     0%
 Info: 	             USRMCLK:     0/    1     0%
 Info: 	             CLKDIVF:     1/    4    25%
 Info: 	           ECLKSYNCB:     1/   10    10%
 Info: 	             DLLDELD:     0/    8     0%
 Info: 	              DDRDLL:     1/    4    25%
 Info: 	             DQSBUFM:     2/    8    25%
 Info: 	     TRELLIS_ECLKBUF:     3/    8    37%
 Info: 	        ECLKBRIDGECS:     1/    2    50%
 
-Info: Placed 128 cells based on constraints.
-Info: Creating initial analytic placement for 8487 cells, random placement wirelen = 672620.
-Info:     at initial placer iter 0, wirelen = 10582
+Info: Placed 136 cells based on constraints.
+ERROR: Unable to place cell 'mem.4.0.0', no Bels remaining of type 'DP16KD'
+6 warnings, 1 error

Interestingly, the prebuilt bitstream from #164 works, and I can build my own from commit 5e042549bfad25a6 ("README: update prebuilt bitstreams/images description/link."), too (needs only 55/56 DP16KD).

enjoy-digital, 2020-12-18:
@geertu: thanks for the feedback. It was building correctly on my setup so could be related the Yosys/Nextpnr version. With https://github.com/litex-hub/linux-on-litex-vexriscv/commit/944bb719c0eb358a15dbcce362daf58c7d802c34 I reverted the old `integrated_rom_size` used for the OrangeCrab so it should now also build correctly for you even with the different Yosys/Nextpnr version.

geertu, 2020-12-18:
@enjoy-digital Thanks, that fixed my build for OrangeCrab.

BTW, I'm using fpga-toolchain nightly-20200904.  Will check if upgrading makes a difference.

closed 2020-12-18 by geertu

geertu, 2020-12-18:
Using nightly-20201218 on commit 07325a0a70b9ccd2 ("README: update ci badge.") also works.