litex-hub/linux-on-litex-vexriscv #170: Prebuilt images hang after "liftoff!" line

geertu, 2020-12-18:
Using either the prebuilt bitstream from #164, or my own built from commit 5e04254 ("README: update prebuilt bitstreams/images description/link."), and the prebuilt images from #164, booting Linux fails with:

[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading images/Image to 0x40000000 (7287696 bytes)...
[LXTERM] Upload complete (273.2KB/s).
[LXTERM] Uploading images/rv32.dtb to 0x40ef0000 (2300 bytes)...
[LXTERM] Upload complete (281.3KB/s).
[LXTERM] Uploading images/opensbi.bin to 0x40f00000 (53640 bytes)...
[LXTERM] Upload complete (281.1KB/s).
[LXTERM] Uploading images/rootfs.cpio to 0x41000000 (5639680 bytes)...
[LXTERM] Upload complete (273.2KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
Executing booted program at 0x41000000

--============= Liftoff! ===============--

What am I doing wrong?
Thanks!

geertu, 2020-12-18:
After building my own bitstream from commit 944bb719c0eb358a ("make.py/OrangeCrab: reduce integrated_rom_size to reduce DP16KD usage."), Linux is booting again, but panics later:

```
[    5.904119] litex-mmc f0004800.mmc: invalid resource
[    5.910400] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[    5.911760] Oops [#1]
[    5.912456] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.10.0-rc6 #32
[    5.913231] epc: c0573378 ra : c01cb3ec sp : c082dcc0
[    5.914034]  gp : c06f2c90 tp : c0830000 t0 : c10673c8
[    5.914761]  t1 : 00000114 t2 : 00000000 s0 : c082dcd0
[    5.915483]  s1 : 00000000 a0 : 00000000 a1 : 00000000
[    5.916186]  a2 : 00000000 a3 : 00000000 a4 : 00000000
[    5.916921]  a5 : 00000000 a6 : c06f4000 a7 : 00000110
[    5.917657]  s2 : 00000000 s3 : 00000000 s4 : 00000000
[    5.918369]  s5 : c1066054 s6 : c1fd16c0 s7 : c06f5000
[    5.919143]  s8 : 00000007 s9 : 00000008 s10: c0001000
[    5.919835]  s11: 00000000 t3 : 00000000 t4 : 00000001
[    5.920517]  t5 : 00000001 t6 : c0804ec9
[    5.921174] status: 00000120 badaddr: 00000000 cause: 0000000d
[    5.922822] ---[ end trace 2a950a1dd99f80f7 ]---
[    5.924133] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    5.925182] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---
```

Might be related to:

`build/orangecrab/orangecrab.dts:102.22-110.5: Warning (simple_bus_reg): /soc/mmc@f0005000: simple-bus unit address format error, expected "f0004800"`

The mmc0 node indeed looks fishy:

```
mmc0: mmc@f0005000 {
        compatible = "litex,mmc";
        bus-width = <4>;
        reg = < 
                0xf0004800 0x100
                0xf0005000 0x100
        >;
        status = "okay";
};  
```

enjoy-digital, 2020-12-18:
@geertu: the .dts generation indeed needs to be updated/fixed. To fix the boot for now you can disable the `sdcard` (comment it here: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L309) or change it to `spisdcard`.

enjoy-digital, 2020-12-21:
I just switched back all boards to SDCard SPI-Mode since SD-Mode is not yet validated. We'll track SD-Mode progress in https://github.com/litex-hub/linux-on-litex-vexriscv/issues/171.

closed 2020-12-21 by enjoy-digital