litex-hub/linux-on-litex-vexriscv #171: Test/Fix SDCard in SD-Mode

enjoy-digital, 2020-12-21:
SDCard is currently working in SPI-Mode, SD-Mode should also be working (boot from SD-Card is already working) but still need to be tested/validated.

kgugala, 2020-12-28:
@enjoy-digital what actually has to be tested? Linux support?

pgielda, 2020-12-28:
It would really be great to get this sorted, this revert to spisdcard broke our internal Renode tests @enjoy-digital

enjoy-digital, 2020-12-28:
@kgugala: yes that's the Linux support. If you can provide help testing it with current Linux kernel: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/buildroot/configs/litex_vexriscv_defconfig#L26-L33 and current [litex_json2dts](https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_json2dts.py) this would be very useful: make sure it works and if not try to understand why, then share tested bitstream, images and steps so that I can reproduce the results (on Arty for example).

enjoy-digital, 2020-12-28:
@pgielda: sorry if it broke your internal Renode tests, this has been reverted since issues were reported by users and I was lacking time on the moment to investigate. SDCard is now working but SDCard access is slower which is a temporary compromise. 

Note that Linux-on-LiteX-VexRiscv is an application project that can be used as an example to create Linux SoCs based on VexRiscv but I would not recommend having CI running directly based on it since it's important here to allow things to change/evolve.

enjoy-digital, 2020-12-28:
From a discussion with @gsomlo on IRC, it seems the issue is reproduced on linux-on-litex-rocket project with 8-bit CSRs (32-bit CSRs is working fine).

gsomlo, 2020-12-29:
On Mon, Dec 28, 2020 at 12:25:01PM -0800, enjoy-digital wrote:
> From a discussion with @gsomlo on IRC, it seems the issue is reproduced on
> linux-on-litex-rocket project with 8-bit CSRs (32-bit CSRs is working fine).

I tried to build the original Antmicro litesdcard driver from here:

https://github.com/litex-hub/linux/tree/wip/litex-vexriscv--linux5.0_sd_card_dma

for my 8-bit-csr-wide litex+rocket gateware. But even after the
hackery required to get it to probe, and updating the DMA base width
to 64 bit width to accomodate current litex defaults, it's not recognizing
the sdcard.

The modified/updated version I have in litex-rocket-rebase (and now in
litex-rebase) actually gets a bit further than the original even on
8-bit-csr (rocket) gateware.

https://github.com/litex-hub/linux/tree/litex-rebase

Not sure whether it was me who broke it (since I can't get the original
to work at all), or it's a mismatch w.r.t. the current litex gateware
(regardless of CPU) that causes it to fail on 8-bit csr-data-width.

But I think we might have an easier time debugging it from where we're
at (in litex-rebase), than trying to figure out where I might have broken
it for 8-bit-csr in the process of getting it to work for me in the first
place (on 32-bit csr litex+rocket).

So that's what I'm going to try, as soon as I have a few spare cycles
(unless anyone else beats me to it :).

enjoy-digital, 2021-01-14:
This has been validated with VexRiscv-SMP, kernel generated from litex-rebase branch (5.11.0-rc2 ) and with `config-frequency `added to the `.dts`: https://github.com/enjoy-digital/litex/commit/b1cad93e62ecf57f1cd98489ca592dcf6d1cf3a2. 

SD-mode on boards has also been reverted with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/732371976ece9a21e2490322f6f6c556232c9883.

Thanks a lot @gsomlo for suggesting looking at `config-frequency`.

enjoy-digital, 2021-01-14:
Boot from SDCard on Arty and test of the SDCard under Linux:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jan 14 2021 08:55:48
 BIOS CRC passed (4464a638)

 Migen git sha1: d42aa6f
 LiteX git sha1: 83fb79fb

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		0KiB
SDRAM:		262144KiB 16-bit @ 800MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000011111111111111000000| delays: 19+-07
  m0, b2: |00000000000000000000000000001111| delays: 29+-02
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 19+-07
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000011111111111111000000| delays: 19+-07
  m1, b2: |00000000000000000000000000001111| delays: 30+-02
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 19+-07
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
Memtest OK
Memspeed at 0x40000000 (2MiB)...
  Write speed: 31MiB/s
   Read speed: 26MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from SDCard in SD-Mode...
Booting from boot.json...
Copying Image to 0x40000000 (7293192 bytes)...
[########################################]
[... 142 lines cut ...]
                  / /  (_)__  __ ____ __
                 / /__/ / _ \/ // /\ \ /
                /____/_/_//_/\_,_//_\_\
                      / _ \/ _ \
   __   _ __      _  _\___/_//_/         ___  _
  / /  (_) /____ | |/_/__| | / /____ __ / _ \(_)__ _____  __
 / /__/ / __/ -_)>  </___/ |/ / -_) \ // , _/ (_-</ __/ |/ /
/____/_/\__/\__/_/|_|____|___/\__/_\_\/_/|_/_/___/\__/|___/
                  / __/  |/  / _ \
                 _\ \/ /|_/ / ___/
                /___/_/  /_/_/
  32-bit RISC-V Linux running on LiteX / VexRiscv-SMP.

login[69]: root login on 'console'
root@buildroot:~# mount /dev/mmcblk0p1 /mnt
[ 1095.791299] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
root@buildroot:~# cd /mnt/
root@buildroot:/mnt# ls
Image                 boot.json             rootfs.tar
LINUX_~2.ZIP          dmesg.txt             rv32.dtb
OPENSBI.BIN           linux_2020_12_30.zip  uname_a.txt
OPENSBI.BIN           opensbi.bin
RV32.DTB              rootfs.cpio
root@buildroot:/mnt#
```

closed 2021-01-14 by enjoy-digital

zguig52, 2021-01-20:
Hi,
Don't know if this is the right place to give this info, but after loading the latest linux on ecpix5, I get this message every seconds:
[   11.953487] Command (cmd 8) failed, status 2
[   11.964504] Command (cmd 55) failed, status 2
[   11.975476] Command (cmd 55) failed, status 2
[   11.986474] Command (cmd 55) failed, status 2
[   11.997961] Command (cmd 55) failed, status 2
[   12.009593] Command (cmd 1) failed, status 2
[   12.011242] litex-mmc f0005000.mmc: Requested clk_freq=0: set to 273437 via div=256
[   13.067506] litex-mmc f0005000.mmc: Requested clk_freq=12500000: set to 8750000 via div=8

When I disable the sdcard option in the board, no problem at all. Please note that otherwise, linux is working fine.

gsomlo, 2021-01-20:
That's probably because of the polling that's been enabled in the
driver:

https://github.com/litex-hub/linux/blob/c926b7e5819cf8050d17400e770cb955634f58b7/drivers/mmc/host/litex_mmc.c#L446

combined with the lack of a card-detect pin on the ecpix5's sdcard
pads:

https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/ecpix5.py#L93

(Contrast with, e.g., nexys4ddr, which *does* have a "cd" pin:
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/nexys4ddr.py#L83
)

In the cd-pin-less scenario, the litesdcard gateware returns "present"
by default, which means linux thinks there should be a card in the
socket, and attempts to initialize it, and you see those log entries
when that initialization fails (because there's really no card present)

There might be a way to use gpio to signal the driver whether a card
is present or not. There might actually exist a bona-fide cd pin on
the ecpix5 that hasn't been added to the platform file (probably not,
but I've seen it happen on other platforms).

Either way, the "easiest" solution is to treat your sd card as
"non-removable", have it inserted at boot, and don't ever eject
(because as it is right now, the linux driver can't tell the
difference, and has to assume the card is always plugged in).

HTH,
--Gabriel

zguig52, 2021-01-20:
This seems to be a very good reason, thanks Gabriel. I will try to find one and put it in to confirm.
Maybe also, we can fix the board conf, there is a cd pin on port L22?
http://docs.lambdaconcept.com/ecpix-5/features/microsd.html
`microSD card pin | FPGA pin | Description
-- | -- | --
CMD | M24 | Command/response
CLK | P24 | Clock
CD | L22 | Card Detect
DAT0 | N26 | Data line (bit 0)
DAT1 | N25 | Data line (bit 1)
DAT2 | N23 | Data line (bit 2)
DAT3 | N21 | Data line (bit 3)
D0+ | AD19 | UHS-II lane 0 (Positive)
D0- | AD20 | UHS-II lane 0 (Negative)
D1+ | AF18 | UHS-II lane 1 (Positive)
D1- | AF19 | UHS-II lane 1 (Negative)`

zguig52, 2021-01-20:
Hey Gabriel, your point was totally right! Thanks again!

When I added to line 97:
'Subsignal("cd",        Pins("L22")),'

Everything is working fine:

litex> sdcard_detect
SDCard not inserted.

... put an SDCard

litex> sdcard_detect
SDCard inserted.

Rebooted to load linux and no more error message :)

It is properly detecting SDCard insertion / removal:
[   45.863892] litex-mmc f0005000.mmc: Requested clk_freq=0: set to 273437 via div=256
[   45.883607] litex-mmc f0005000.mmc: Requested clk_freq=12500000: set to 8750000 via div=8
[   45.999188] litex-mmc f0005000.mmc: Requested clk_freq=25000000: set to 35000000 via div=2
[   46.000838] mmc0: new SDHC card at address 0007
[   46.023843] blk_queue_max_hw_sectors: set to minimum 8
[   46.024811] blk_queue_max_segment_size: set to minimum 4096
[   46.026116] mmcblk0: mmc0:0007 SD16G 14.5 GiB 
[   46.068268]  mmcblk0: p1 p2
[   51.209503] Command (cmd 13) failed, status 2
[   51.220409] Command (cmd 13) failed, status 2
[   51.231305] Command (cmd 13) failed, status 2
[   51.242192] Command (cmd 13) failed, status 2
[   51.253797] Command (cmd 13) failed, status 2
[   51.265192] Command (cmd 13) failed, status 2
[   51.276082] Command (cmd 13) failed, status 2
[   51.287069] Command (cmd 13) failed, status 2
[   51.297955] Command (cmd 13) failed, status 2
[   51.309520] Command (cmd 13) failed, status 2
[   51.310503] mmc0: card 0007 removed
[   51.392703] litex-mmc f0005000.mmc: Requested clk_freq=0: set to 273437 via div=256

Still a small error message when removing it: Command (cmd 13) failed, status 2, put not persistent. I don't know if this is normal?

If it is OK, I will try to push this fix.

gsomlo, 2021-01-20:
I think the failed cmd13 attempts following a removal are normal.
Please go ahead and submit a PR for the CD pin on the ecpix5.

Best,
--Gabriel