litex-hub/linux-on-litex-vexriscv #172: Install error: root priviledges

dadede, 2020-12-21:
Hello,
I am following the installation guide at [(https://github.com/enjoy-digital/litex/wiki/Installation)] and I got the error below. I ran as root and everything worked until I got more errors so I had to backtrack after seeing this post [https://github.com/enjoy-digital/litex/issues/273]. I deleted the linux-on-litex-vexriscv folder and started from the beginning.

Afrer running `./litex_setup.py init install --user`, this is the error I get below:
`[checking litex_setup.py]...
[installing migen]...
running develop
error: can't create or remove files in install directory

The following error occurred while trying to add or remove files in the
installation directory:

    [Errno 13] Permission denied: '/usr/local/lib/python3.8/dist-packages/test-easy-install-11520.write-test'

The installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was:

    /usr/local/lib/python3.8/dist-packages/

Perhaps your account does not have write access to this directory?  If the
installation directory is a system-owned directory, you may need to sign in
as the administrator or "root" account.  If you do not have administrative
access to this machine, you may wish to choose a different installation
directory, preferably one that is listed in your PYTHONPATH environment
variable.

For information on other options, you may wish to consult the
documentation at:

  https://setuptools.readthedocs.io/en/latest/easy_install.html

Please make the appropriate changes for your system and try again.

Traceback (most recent call last):
  File "./litex_setup.py", line 170, in <module>
    subprocess.check_call("python3 setup.py develop", shell=True)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'python3 setup.py develop' returned non-zero exit status 1.
`
Any suggestion helps, thanks!

enjoy-digital, 2020-12-22:
Hi @dadede,

can you try?:
```
./litex_setup.py init
sudo ./litex_setup.py install
```

dadede, 2020-12-22:
Wow, that worked. Thanks!

Can you please let me know what is different between 
`./litex_setup.py init
sudo ./litex_setup.py install` and the instructions in the installation guide are?

After running `lxsim --cpu-type=vexriscv`  as follows in the guide I get the following error

> Traceback (most recent call last):
>   File "/usr/local/bin/lxsim", line 11, in <module>
>     load_entry_point('litex', 'console_scripts', 'lxsim')()
>   File "/home/dut/litex/litex/tools/litex_sim.py", line 423, in main
>     vns = builder.build(
>   File "/home/dut/litex/litex/soc/integration/builder.py", line 204, in build
>     self._generate_includes()
>   File "/home/dut/litex/litex/soc/integration/builder.py", line 104, in _generate_includes
>     for k, v in export.get_cpu_mak(self.soc.cpu, self.compile_software):
>   File "/home/dut/litex/litex/soc/integration/export.py", line 88, in get_cpu_mak
>     ("TRIPLE", select_triple(triple)),
>   File "/home/dut/litex/litex/soc/integration/export.py", line 83, in select_triple
>     raise OSError(msg)
> OSError: Unable to find any of the cross compilation toolchains:
> - riscv64-unknown-elf
> - riscv64-unknown-linux-gnu
> - riscv32-unknown-elf
> - riscv32-unknown-linux-gnu
> - riscv64-elf
> - riscv32-elf
> - riscv-none-embed
> - riscv64-linux
> - riscv64-linux-gnu
> - riscv-sifive-elf
> - riscv64-none-elf

Could they be related? Any suggestion is appreciated, thank you.

enjoy-digital, 2020-12-22:
It seems you are not able to use `develop` without  `root` privileges on your machine. It could be related to previous installation you made.

You now need to install a RISC-V GCC toolchain. You can get one with `litex_setup.py  gcc` and after that need to be sure it's in your `PATH.`

dadede, 2020-12-22:
Ok, understood. After running `litex_setup.py gcc`, it downloads a file and extracts to a folder named `riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-ubuntu14`.

I have added that to my path, however I suspect there may be something that i am missing to install the toolchain, that command does not install and I am still getting the same error. Do I have to run `litex_setup.py gcc`  as root?

enjoy-digital, 2020-12-22:
Adding the` RISC-V GCC` compiler to your `PATH` has to be done manually and should be printed when executing `litex_setup.py`:
https://github.com/enjoy-digital/litex/blob/master/litex_setup.py#L182-L183

dadede, 2020-12-22:
I've done that and everything works now. Thank you for your prompt assistance with this, great work you're doing.

closed 2020-12-22 by dadede