litex-hub/linux-on-litex-vexriscv #173: Simulation (sim.py) hangs after OpenSBI

kammoh, 2020-12-21:
Tried both the prebuild opensbi binary (v0.8-1-gecf7701) and `0.8-linux-on-litex-vexriscv` branch.
```
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

OpenSBI v0.8-2-ga9ce3ad
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : LiteX / VexRiscv-SMP
Platform Features   : timer,mfdeleg
Platform HART Count : 8
Boot HART ID        : 0
Boot HART ISA       : rv32imas
BOOT HART Features  : time
BOOT HART PMP Count : 0
Firmware Base       : 0x40f00000
Firmware Size       : 124 KB
Runtime SBI Version : 0.2

MIDELEG : 0x00000222
MEDELEG : 0x0000b10
```
^^^ hangs here

Dolu1990, 2020-12-21:
Hi, 

Did you used sim.py with some arguements ?

kammoh, 2020-12-22:
Hi,
I'm using sim.py without any arguments.
Using Verilator version `v4.106`, litex master (`5ec55547`), Python 3.9.1, macOS 11.1, clang 12.0.0 x86_64

enjoy-digital, 2020-12-22:
Hi @kammoh, @Dolu1990,

I'm going to look at this.

kammoh, 2020-12-22:
Thanks @enjoy-digital! Please let me know if adding vcd traces is going to be helpful.

enjoy-digital, 2020-12-22:
This is fixed with https://github.com/enjoy-digital/litex/commit/d90d3e043bc81b434d6424c8585c757d77853d2c and https://github.com/litex-hub/linux-on-litex-vexriscv/commit/2758b5df5d2f96eb3eaae4ae9f26832dbb7bf64c. The SDRAM contents that is pre-initialized from files was corrupted by some tests that have been added recently in `liblitedram `and that were not disabled:```

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS CRC passed (5457362e)

 Migen git sha1: 9a37a58
 LiteX git sha1: d90d3e04

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 1MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		8-bit data
ROM:		32KiB
SRAM:		8KiB
L2:		0KiB
SDRAM:		65536KiB 32-bit @ 1MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

OpenSBI v0.8-1-gecf7701
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : LiteX / VexRiscv-SMP
Platform Features   : timer,mfdeleg
Platform HART Count : 8
Boot HART ID        : 0
Boot HART ISA       : rv32imas
BOOT HART Features  : time
BOOT HART PMP Count : 0
Firmware Base       : 0x40f00000
Firmware Size       : 124 KB
Runtime SBI Version : 0.2

MIDELEG : 0x00000222
MEDELEG : 0x0000b101
[    0.000000] Linux version 5.10.0-rc6 (florent@panda) (riscv32-buildroot-linux-gnu-gcc.br_real (Buildroot 2020.11-281-g69e5046e7b) 10.2.0, GNU ld (GNU Binutils) 2.33.1) #32 SMP Tue Dec 15 10:55:28 CET 2020
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] Initial ramdisk at: 0x(ptrval) (8388608 bytes)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x0000000043ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000043ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x0000000043ffffff]
[    0.000000] SBI specification v0.2 detected
[    0.000000] SBI implementation ID=0x1 Version=0x8
[    0.000000] SBI v0.2 TIME extension detected
[    0.000000] SBI v0.2 IPI extension detected
[    0.000000] SBI v0.2 RFENCE extension detected
[    0.000000] SBI v0.2 HSM extension detected
[    0.000000] riscv: ISA extensions aim
[    0.000000] riscv: ELF capabilities aim
```
[...]

@kammoh: can you give it a try?

kammoh, 2020-12-22:
@enjoy-digital Linux is booting (be it quite slow)! Amazing! Thanks!
How can I access Verilator/compiler flags to see if it can run any faster? 
Next step is to try it on my shiny new ulx3s though :D

closed 2020-12-22 by kammoh

suppamax, 2021-11-03:
Hi, 

I have just downloaded the latest version of the project and I see the exact same behaviour as initially reported by @kammoh .
It looks like the fixes done by @enjoy-digital are included in my codebase.
I tried all 3 linux versions listed [here](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164), but it didn't help.

I am using ./sim.py with no argument.