litex-hub/linux-on-litex-vexriscv #175: about using spi mode sdcard bootup problem

SanadaShinken, 2020-12-26:
Hi, All:

After some tests, booting from spi mode sdcard still failed. the following is what i done for sdcard boot tests.

1. create a fat16 partition on sd card. what I use is a Sandisk 32GB sd card. there is only one partition in the sd card.
2. copy images/* file into fat16 partition.
3. turn on fpga board, and wait the prompt shows boot from sd card failed.
4. after into prompt, using spisdcardboot to boot up, it shows the same error.

where am I doing wrong??

Many thanks.

BR, Sanada

enjoy-digital, 2020-12-28:
Hi @SanadaShinken,

can you provide the log of the boot and the list of files copied to the SDCard? Do you also have another SDCard you could test with? Also is it with an already supported board or a board you are adding support to?

SanadaShinken, 2020-12-28:
> Hi @SanadaShinken,
> 
> can you provide the log of the boot and the list of files copied to the SDCard? Do you also have another SDCard you could test with? Also is it with an already supported board or a board you are adding support to?

Hi @enjoy-digital :

the boot log will provide after I test in my office. the board which I used to test is qmtech wukong board. the SD is 32GB, and the used files is extracted from https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164. there are Image, rootfs.cpio, rv32.dtb, opensbi.bin, image.json.

BR, Sanda

SanadaShinken, 2020-12-29:
@enjoy-digital :

the below message is the sd card boot log. if you need more message, please tell how to provide. Many Thanks.

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 28 2020 13:39:26
 BIOS CRC passed (d5337e4f)

 Migen git sha1: d42aa6f
 LiteX git sha1: bddb1706

--=============== SoC ==================--
CPU:            VexRiscv SMP-LINUX @ 150MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            8-bit data
ROM:            64KiB
SRAM:           8KiB
L2:             0KiB
SDRAM:          262144KiB 16-bit @ 1200MT/s (CL-10 CWL-7)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00111111111111110000000000000000| delays: 09+-07
  m0, b2: |00000000000000000011111111111111| delays: 24+-06
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 08+-07
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00111111111111110000000000000000| delays: 08+-07
  m1, b2: |00000000000000000011111111111111| delays: 24+-07
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 08+-06
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB
   Read: 0x40000000-0x40200000 2MiB
Memtest OK
Memspeed at 0x40000000 (2MiB)...
  Write speed: 47MiB/s, 381Mbps
   Read speed: 39MiB/s, 312Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from flash...
Error: Invalid image length 0xcccccccc
Booting from SDCard in SPI-Mode...
Booting from boot.json...
Booting from boot.bin...
SDCard boot failed.
Booting from network...
Local IP : 192.168.1.50
Remote IP: 192.168.1.100
Booting from boot.json...
Booting from boot.bin...
Copying boot.bin to 0x40000000...
Network boot failed.
No boot medium found

--============= Console ================--

litex> help

LiteX BIOS, available commands:

leds                     - Set Leds value
flush_l2_cache           - Flush L2 cache
flush_cpu_dcache         - Flush CPU data cache
crc                      - Compute CRC32 of a part of the address space
ident                    - Identifier of the system
help                     - Print this help

sdcardboot               - Boot from SDCard
netboot                  - Boot via Ethernet (TFTP)
serialboot               - Boot from Serial (SFL)
flashboot                - Boot from Flash
reboot                   - Reboot

mem_speed                - Test memory speed
mem_test                 - Test memory access
mem_copy                 - Copy address space
mem_write                - Write address space
mem_read                 - Read address space
mem_list                 - List available memory regions

flash_erase              - Erase whole flash
flash_write              - Write to flash

sdram_mr_write           - Write SDRAM Mode Register
sdram_force_wrphase      - Force write phase
sdram_force_rdphase      - Force read phase
sdram_test               - Test SDRAM
sdram_cal                - Calibrate SDRAM
sdram_init               - Initialize SDRAM (Init + Calibration)

mdio_dump                - Dump MDIO registers
mdio_read                - Read MDIO register
mdio_write               - Write MDIO register

litex> sdcardboot

Booting from SDCard in SPI-Mode...
Booting from boot.json...
Booting from boot.bin...
SDCard boot failed.

litex>

BR, Sanada

enjoy-digital, 2021-01-04:
@SanadaShinken: I did some cleanups/changes while merging the `qmtech_wukong` board support in LiteX-Boards. I directly reused the `sdcard_pmod_io` function from the Arty: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/qmtech_wukong.py#L138. Could you do another test on your hardware with this updated code?

SanadaShinken, 2021-01-05:
> @SanadaShinken: I did some cleanups/changes while merging the `qmtech_wukong` board support in LiteX-Boards. I directly reused the `sdcard_pmod_io` function from the Arty: [...]

@enjoy-digital :

Thanks very much. I'll do the test as soon as possible. the test result will be upload in this issue.

BR, Sanada

SanadaShinken, 2021-01-07:
@enjoy-digital :

Thank you very much!! the SDCard boot is working!!!

after some bios code tracing, comparison, try and error, the root cause is the spi clock setting.
the clock of wukong board's spi sdcard is about 2MHz. when the clock setting is right, sdcard boot is work. Thank you very much.

BTW, I upload the latest setting of my current used. 
[qmtech_wukong.platforms.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5782345/qmtech_wukong.platforms.txt)
[qmtech_wukong.targets.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5782348/qmtech_wukong.targets.txt)

BR, Sanada

enjoy-digital, 2021-01-08:
Thanks @SanadaShinken for the feedback, great you got it working.

closed 2021-01-08 by enjoy-digital