litex-hub/linux-on-litex-vexriscv #176: Fix boot with kernel generated from litex-rebase branch.

enjoy-digital, 2020-12-29:
Boot is hanging when kernel is generated from litex-rebase branch (current kernel is generated from litex-rocket-rebase branch). Using upstream Linux 5.11-rc1 produces the same results with VexRiscv-SMP while working with mor1kx and Rocket.
Understand why and fix.

enjoy-digital, 2020-12-29:
The regression does not seem to be directly related to the litex patches now integrated in 5.11.rc1 but from another issue with 5.11.rc1 and our VexRiscv-SMP config (since 5.11.rc1 is not behaving correctly with mor1kx and Rocket). To workaround this, we are currently using a `linux-on-litex-vexriscv` branch with 5.10 stable + litex patches from 5.11.rc1 + litex-rebase patches... This is a temporary solution that should allow us to go further until the regression is understood...

geertu, 2021-01-04:
One issue is the "Failed to add a Kernel code resource at 40001000" message after Initmem setup.
Adding some debug info to init_resources() shows that the memblock.reserved list keeps on increasing, until memory corruption happens (pointers start to look like ASCII strings).
Changing L1_CACHE_SHIFT in arch/riscv/include/asm/cache.h to 5, as suggested in https://lore.kernel.org/linux-riscv/CAOnJCUL70g8PEQvB7B0F9t-XP8DS20Omp6NQiw4xDJRGc0_cdw@mail.gmail.com/ fixes that. Waiting for Atish patches to be posted...

Still, the kernel hangs after "sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver".  Usually, the next line is "Freeing unused kernel memory:", so this suggest another memory bookkeeping issue...

enjoy-digital, 2021-01-08:
Thanks for the additional info @geertu.

geertu, 2021-01-08:
You need the following two fixes from Atish:
  - "[PATCH 1/4] RISC-V: Do not allocate memblock while iterating reserved memblocks"
    https://lore.kernel.org/linux-riscv/20210107092652.3438696-2-atish.patra@wdc.com/
  - "[PATCH 3/4] RISC-V: Fix L1_CACHE_BYTES for RV32"
    https://lore.kernel.org/linux-riscv/20210107092652.3438696-4-atish.patra@wdc.com/

If you have applied Sebastien's patch to reduce kernel size:
"[PATCH] riscv: Align on L1_CACHE_BYTES when STRICT_KERNEL_RWX"
https://lore.kernel.org/linux-riscv/3ba42880e5b43fc238aac8d30da271516fb281ac.camel@gmail.com/
then you also need to apply my follow-up fix:
https://lore.kernel.org/linux-riscv/CAMuHMdXFf4=7D_SLBgju2qaUzwC8dMkxq3zWDTmZ3AgWkBMQig@mail.gmail.com/

After that Linux boot fine into the buildroot initramfs again.

enjoy-digital, 2021-01-08:
Awesome @geertu! Thanks for tracking this, being able to use `litex-rebase` will greatly simplify collaboration with other Linux projects (Rocket and mor1kx). I'll give it a try on next Monday.

enjoy-digital, 2021-01-11:
@geertu: I confirm this is working and just switched the project to it with https://github.com/litex-hub/linux-on-litex-vexriscv/pull/182. Thanks  again.

enjoy-digital, 2021-01-11:
The prebuilt images have also been shared at https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164.

closed 2021-01-11 by enjoy-digital