litex-hub/linux-on-litex-vexriscv #178: Fix LiteEth IRQ mode

enjoy-digital, 2020-12-29:
IRQ mode does not seems to be working currently with VexRiscv-SMP while working with mor1kx and Rocket. Check the IRQ integration/PLIC with VexRiscv-SMP to understand and fix.

dayjaby, 2021-01-04:
Is that the reason why ethernet is not working right now?

```
root@buildroot:~# ifconfig eth0 192.168.10.2
root@buildroot:~# ping 192.168.10.1
PING 192.168.10.1 (192.168.10.1): 56 data bytes
^C
--- 192.168.10.1 ping statistics ---
62 packets transmitted, 0 packets received, 100% packet loss
root@buildroot:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 1A:73:A7:EC:70:6B  
          inet addr:192.168.10.2  Bcast:192.168.10.255  Mask:255.255.255.0
          inet6 addr: fe80::1873:a7ff:feec:706b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:2
```

while the counterpart was definitely sending/receiving something:

```
$ ifconfig
enx00e04c415dfb: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.10.1  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 fe80::ccb8:f1de:fb9:3610  prefixlen 64  scopeid 0x20<link>
        ether 00:e0:4c:41:5d:fb  txqueuelen 1000  (Ethernet)
        RX packets 33  bytes 1518 (1.5 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 116  bytes 20281 (20.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

enjoy-digital, 2021-01-14:
@dayjaby: it's probably a different issue. Which board are you using? Were you booting Linux over netboot? (Just trying to see if Ethernet was working).

tgorochowik, 2021-01-14:
@enjoy-digital it looks as though VexRiscv-SMP changed the interrupt controller it uses to a sifive based one is that correct?

We are currently working on adding support for i2s, rebasing onto current master broke the interrupts for us too - the new controller type appears to be the culprit - we analyzed the desing and found out that even though the irqs from the hardware reach the IRQ controller its handler is never invoked.

It seems to be the same issue as the issue with liteeth irq mode not working.

Please let us know when you have any findings regarding the IRQ issues regarding liteeth, thanks!

enjoy-digital, 2021-01-14:
@tgorochowik: thanks for looking at this. The interrupt controller has indeed changed with VexRiscv-SMP for a implentation that should be compatible with PLIC+CLINT (https://sifive.cdn.prismic.io/sifive/0d163928-2128-42be-a75a-464df65e04e0_sifive-interrupt-cookbook.pdf). IIRC Ethernet IRQs were working in the the test repository were were using for VexRiscv-SMP: https://github.com/enjoy-digital/litex_vexriscv_smp

It could just be the related to the interrupt declaration in the `.dts`. We were explicitly attaching the interrupts of the peripherals to the `plic`: https://github.com/enjoy-digital/litex_vexriscv_smp/blob/master/json2dts.py#L149, we are not doing it currently in `litex_json2dts`: https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_json2dts.py#L206.

geertu, 2021-01-14:
There is an "interrupt-parent = <&intc0>;" in .dts pointing to the PLIC at the top, isn't it?
https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_json2dts.py#L31

enjoy-digital, 2021-01-27:
This is fixed by https://github.com/enjoy-digital/litex/commit/7abfbd9825188d1f6d97453838e18ed7af5526a7 and IRQs are re-enabled with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/1e05fad048a28520015a696d46499de6cbe15dee.

closed 2021-01-27 by enjoy-digital

zguig52, 2021-02-04:
Hi @enjoy-digital , I confirm you that with ECPIX-5 board, ethernet is not working when IRQ is enabled, while with polling everything is fine (option polling=True from soc_linux.py).

This is quite strange as it seems that IRQs are received but not processed when trying to manually config the IP and sending pings, I see the IRQs counter increasing (net_rx):
cat /proc/softirqs 
                    CPU0       
          HI:          0
       TIMER:     157864
      NET_TX:          0
      NET_RX:         10
       BLOCK:          0
    IRQ_POLL:          0
     TASKLET:          1
       SCHED:          0
     HRTIMER:          0
         RCU:       3460

And here I don't see any packets processed (tx or rx):
ifconfig -a
eth0      Link encap:Ethernet  HWaddr BE:BF:50:52:B8:C5  
          inet addr:XXX.XXX.XXX.XXX  Bcast:0.0.0.0  Mask:255.255.255.0
          inet6 addr: fe80::bcbf:50ff:fe52:b8c5/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:2