litex-hub/linux-on-litex-vexriscv #179: Specify Arty Device

wtfuzz, 2021-01-05:
I'm attempting to build for an Arty A7 100T and it doesn't seem the `--device` argument is working, as it's still selecting the 35T part.

```
$ ./make.py --board=arty --device=xc7a100tcsg324-1 --cpu-count=1 --build
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2021-01-05 05:40:57)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7a35ticsg324-1L.
```

wtfuzz, 2021-01-05:
Submitted #180 which allows the `--variant` argument to specified in make.py

```
$ ./make.py --board=arty --variant=a7-100 --cpu-count=1 --build
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2021-01-05 05:59:40)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7a100tcsg324-1.
```

enjoy-digital, 2021-01-06:
https://github.com/litex-hub/linux-on-litex-vexriscv/pull/180 has been merged, thanks @wtfuzz.

closed 2021-01-06 by enjoy-digital