litex-hub/linux-on-litex-vexriscv #181: litevideo: probe of f0006000.gpu failed with error -110

Chgavilana, 2021-01-09:
HI:

During boot process, I got a new error with litevideo driver.:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jan  9 2021 14:38:45
 BIOS CRC passed (40cda38f)

 Migen git sha1: d42aa6f
 LiteX git sha1: 1ce19400

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
SDRAM:		262144KiB 16-bit @ 800MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000011111111111111100| delays: 22+-07
  m0, b2: |00000000000000000000000000000001| delays: -
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 22+-07
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000001111111111111100| delays: 22+-07
  m1, b2: |00000000000000000000000000000001| delays: 31+-00
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 23+-07
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
Memtest OK
Memspeed at 0x40000000 (2MiB)...
  Write speed: 31MiB/s
   Read speed: 27MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading images/Image to 0x40000000 (7279384 bytes)...
[LXTERM] Upload complete (86.0KB/s).
[LXTERM] Uploading images/rv32.dtb to 0x40ef0000 (4738 bytes)...
[... 125 lines cut ...]
[    3.512655] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    3.535401] Freeing unused kernel memory: 176K
[    3.535956] Kernel memory protection not selected by kernel config.
[    3.537334] Run /init as init process
[    3.542767] mmc0: host does not support reading read-only switch, assuming write-enable
[    3.543915] mmc0: new SDHC card on SPI
[    3.559546] mmcblk0: mmc0:0000 SD32G 29.0 GiB 
[    3.718776]  mmcblk0: p1
[    4.048332] tmpfs: Unknown parameter 'mode'
mount: mounting tmpfs on /dev/shm failed: Invalid argument
[    4.056258] tmpfs: Unknown parameter 'mode'
mount: mounting tmpfs on /tmp failed: Invalid argument
[    4.062155] tmpfs: Unknown parameter 'mode'
mount: mounting tmpfs on /run failed: Invalid argument
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Saving random seed: [    4.978123] random: dd: uninitialized urandom read (512 bytes read)
OK
Starting network: OK

Welcome to Buildroot
buildroot login: 

```

Chgavilana, 2021-01-12:
Hi: debugging error, I found that problem happens when use litevideo_mmcm_write, over conditional 
```
if (!wait_event_timeout(prv->wq,
			litex_read8(prv->base + LITEVIDEO_MMCM_READY_OFF),
			HZ))
		return -ETIMEDOUT;

	return 0;
``
I can increase timeout value but same result, so, problem could be with wait_queue_head_t wq or reading o toogle LITEVIDEO_MMCM_READY_OFF. other drivers work fine (uart, spi, eth) but i can't fine what happens with litevideo.

enjoy-digital, 2021-01-14:
Hi @Chgavilana,

sorry for the delay on this. I'll try to have a closer look soon. It's indeed possible the current Linux driver relies on the MMCM to configure the Video clock. You could try to add a `mmcm` to your design (as done on Arty for example: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L71).

If not working I'll try to update/test the NeTV2 example soon.

Chgavilana, 2021-01-16:
Hi @ enjoy-digital

I checked and found that over kernel driver litevideo.h, some offset address are different respect to generated file  csr.json. Maybe like I try just HDMI OUT, some constant are different. when I change offset values, driver load without problem and error -110 disappear.

However, it still not working, I see that some variables are not used, for example all in function litevideo_clkgen_write use address that I can't find (It try to set M and D for clock generation, but these is hand coded); also framebuffer_core_underflow * over csr.json can't be readed or change. Also I can't check if LiteDRAMDMAReader used over core.py works. I Set mmcm module over project but this define others address that not affect litevideo registers.

I will appreciate if you test with NeTV2 and Linux, I will continue testing.

BR. Carlos.

mithro, 2021-01-18:
FYI - @kgugala

enjoy-digital, 2021-03-30:
LiteVideo driver is no longer required/used with current code.

closed 2021-03-30 by enjoy-digital