litex-hub/linux-on-litex-vexriscv #183: won't build for Genesys2

monniaux, 2021-01-12:
$ git rev-parse --short HEAD
48d7f3e
$ ./make.py --board=Genesys2 --build
[..]
ERROR: [Common 17-69] Command failed: File '/home/monniaux/progs/Risc-V/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw256.v' does not exist

enjoy-digital, 2021-01-12:
The VexRiscv-SMP cluster for the Genesys2 is not part of the pre-generated clusters. Scala/Spinal-HDL/VexRiscv tools are probably not installed correctly, uou can have a look at https://github.com/enjoy-digital/litex/issues/687 and ask there if you have troubles installing the tools.

Here is attached to cluster if you want to test without installing the tools to re-generate it:
[VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw256.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5803947/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw256.zip)

enjoy-digital, 2021-01-14:
The question has been answered, we can close I think. Feel free to ask in enjoy-digital/litex#687 if you have more questions @monniaux.

closed 2021-01-14 by enjoy-digital