litex-hub/linux-on-litex-vexriscv #184: nextpnr-ecp5 still not finished after 57 minutes

geertu, 2021-01-14:
Info: Logic utilisation before packing:
-Info:     Total LUT4s:     13244/24288    54%
-Info:         logic LUTs:  11244/24288    46%
-Info:         carry LUTs:   1022/24288     4%
-Info:           RAM LUTs:    652/12144     5%
-Info:          RAMW LUTs:    326/ 6072     5%
+Info:     Total LUT4s:     17892/24288    73%
+Info:         logic LUTs:  14298/24288    58%
+Info:         carry LUTs:   1446/24288     5%
+Info:           RAM LUTs:   1432/12144    11%
+Info:          RAMW LUTs:    716/ 6072    11%
 
-Info:      Total DFFs:     12956/24288    53%
+Info:      Total DFFs:     14894/24288    61%
 ...
-Info: Routing 57544 arcs.
+Info: Routing 76840 arcs.
 Info:            |   (re-)routed arcs  |   delta    | remaining|       time spent     |
 Info:    IterCnt |  w/ripup   wo/ripup |  w/r  wo/r |      arcs| batch(sec) total(sec)|
 ...
 Info:     105000 |    24332      80667 |  633   367 |     17562|     488.77    2476.01|

During my previous built for OrangeCrab on Jan 4, routing took 43.66s

I suspect commit 732371976ece9a21e2490322f6f6c556232c9883

geertu, 2021-01-14:
Aborted after 74 minutes.
Commit 7323719 is indeed to blame, as commit 48d7f3eb6ecc5 works fine.

Do I need a newer fpga-tooolchain? Using nightly-20201218.

enjoy-digital, 2021-01-14:
@enjoy-digital: The 25F is indeed almost full with VexRiscv SMP and the LiteSDCard core. With the following LiteSDCard patch you should be able to build it again. 

```
diff --git a/litesdcard/frontend/dma.py b/litesdcard/frontend/dma.py
index 839f6b7..b5e0609 100644
--- a/litesdcard/frontend/dma.py
+++ b/litesdcard/frontend/dma.py
@@ -26,7 +26,7 @@ class SDBlock2MemDMA(Module, AutoCSR):
         # # #
 
         # Submodules
-        fifo      = stream.SyncFIFO([("data", 8)], fifo_depth)
+        fifo      = stream.SyncFIFO([("data", 8)], fifo_depth, buffered=True)
         converter = stream.Converter(8, bus.data_width, reverse=True)
         self.submodules += fifo, converter
         self.submodules.dma = WishboneDMAWriter(bus, with_csr=True, endianness=endianness)
@@ -74,7 +74,7 @@ class SDMem2BlockDMA(Module, AutoCSR):
         # Submodules
         self.submodules.dma = WishboneDMAReader(bus, with_csr=True, endianness=endianness)
         converter = stream.Converter(bus.data_width, 8, reverse=True)
-        fifo      = stream.SyncFIFO([("data", 8)], fifo_depth)
+        fifo      = stream.SyncFIFO([("data", 8)], fifo_depth, buffered=True)
         self.submodules += converter, fifo
 
         # Flow
```

I'll have a closer look and fix the build soon.

enjoy-digital, 2021-01-15:
@geertu: thanks for the feedback. I'll do more specific tests with the OrangeCrab (I've been testing on Arty). For now you can revert to `spisdcard ` if you want to get back SPI-Mode SDCard.

enjoy-digital, 2022-03-18:
@geertu: The OrangeCrab now builds correctly with SD-Card and also works fine. Since I switched back ECP5/DDR3 boards to use native LiteDRAM interface, the following change is also required: https://github.com/litex-hub/litex-boards/commit/eb8657f515fe63ced13c487f7da537451d89d993.

As we saw in the past, it's possible that some OrangeCrab board will have troubles with the native LiteDRAM interface. --with-wishbone-memory argument can then be added to the build to use Wishbone Memory and L2 Cache (but is slower).

closed 2022-03-18 by enjoy-digital