litex-hub/linux-on-litex-vexriscv #186: sim.py hangs after sched_clock init

geertu, 2021-01-17:
LiteX simulation seems to be broken. It hangs after

> [    0.001749] sched_clock: 64 bits at 1000kHz, resolution 1000ns, wraps every 2199023255500ns

Same result with the latest prebuilds and my own kernel .config. Works fine on OrangeCrab, though.

Palmer and Atish could use the simulator to investigate the L1_CACHE_SHIFT issue, cfr.
https://lore.kernel.org/linux-riscv/CAMuHMdXoJ9-jM4sazFbHXEsaDFFMK1ybM53SDqy_2QqPMZEQ=g@mail.gmail.com/

enjoy-digital, 2021-01-20:
This was caused by the `sys_clk_freq` value passed to the `.dts` that was apparently too low and was causing a hang or slowing down too much the simulation.

This should be fixed with:
- https://github.com/enjoy-digital/litex/commit/acb6741b8a49a3ad2e7d68c9b135c8e3de144f39
- https://github.com/litex-hub/linux-on-litex-vexriscv/commit/4dd2ab4e5f28a2764db9d0c8147e47662a5879ff

With this, here is the simulation log I get:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS CRC passed (ac300d3c)

 Migen git sha1: 40b1092
 LiteX git sha1: 41964f94

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		32KiB
SRAM:		8KiB
L2:		0KiB
SDRAM:		65536KiB 32-bit @ 100MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

OpenSBI v0.8-1-gecf7701
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : LiteX / VexRiscv-SMP
Platform Features   : timer,mfdeleg
Platform HART Count : 8
Boot HART ID        : 0
Boot HART ISA       : rv32imas
BOOT HART Features  : time
BOOT HART PMP Count : 0
Firmware Base       : 0x40f00000
Firmware Size       : 124 KB
Runtime SBI Version : 0.2

MIDELEG : 0x00000222
MEDELEG : 0x0000b101
[    0.000000] Linux version 5.11.0-rc2 (florent@panda) (riscv32-buildroot-linux-gnu-gcc.br_real (Buildroot 2020.11-281-g69e5046e7b) 10.2.0, GNU ld (GNU Binutils) 2.33.1) #2 SMP Mon Jan 11 10:05:20 CET 2021
[... 66 lines cut ...]
[    4.526042] f0001000.serial: ttyLXU0 at MMIO 0x0 (irq = 0, base_baud = 0) is a liteuart
[    4.530880] printk: console [liteuart0] enabled
[    4.530880] printk: console [liteuart0] enabled
[    4.532751] printk: bootconsole [sbi0] disabled
[    4.532751] printk: bootconsole [sbi0] disabled
[    4.561755] libphy: Fixed MDIO Bus: probed
[    4.564898] i2c /dev entries driver
[    4.601498] NET: Registered protocol family 10
[    4.613033] Segment Routing with IPv6
[    4.614483] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    4.634361] Freeing unused kernel memory: 204K
[    4.635031] Kernel memory protection not selected by kernel config.
[    4.636334] Run /init as init process
[    4.988963] tmpfs: Unknown parameter 'mode'
mount: mounting tmpfs on /dev/shm failed: Invalid argument
[    4.995108] tmpfs: Unknown parameter 'mode'
mount: mounting tmpfs on /tmp failed: Invalid argument
[    5.001820] tmpfs: Unknown parameter 'mode'
mount: mounting tmpfs on /run failed: Invalid argument
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Saving random seed: [    6.311057] random: dd: uninitialized urandom read (512 bytes read)
```
Can you try to reproduce the results @geertu?

geertu, 2021-01-20:
Confirmed, booting to userspace again.
Note that wall-clock time is running really slow inside, but that may be expected?
E.g. "sleep 0.1" sleeps for 26s (on i7-8700K).

enjoy-digital, 2021-01-20:
Thanks for the test @geertu. So the simulation is 260X slower than 100MHz, so runs at 384KHz, which is realistic here yes.

closed 2021-01-20 by enjoy-digital