litex-hub/linux-on-litex-vexriscv #187: lxterm crash when pasting into Linux console

geertu, 2021-01-22:
Pasting a string of 12 or more characters into lxterm crashes serial communication, requiring a cold reboot (unplug) to recover.
Pressing the OrangeCrab button does not help.

Notes:

- It only happens when running Linux, not in the LiteX BIOS,
- It does not happen when using screen instead of lxterm.

Putting an upper limit on the number of iterations of the while loop in linux/drivers/tty/serial/liteuart.c:liteuart_timer() doesn't help.

geertu, 2021-01-22:
Seems to happens with screen, too, but needs a longer sequence (16 chars or more).

geertu, 2021-02-05:
The situation has improved:
  1. Pasting in screen seems to work fine now,
  2. Pasting in lxterm causes some (usually each 13th) characters to be dropped., but it no longer locks up.

You can still crash and lock-up usb_acm by e.g. holding the Enter key, and pressing the button.

gsomlo, 2022-11-08:
@geertu -- I had a similar experience with using `screen` (from `terminator`, fwiw) to connect to my LiteX + Rocket + Linux system. The problem simply disappeared when I finally got around to implement IRQ-driven RX in the LiteUART linux driver (see the top three commits in https://github.com/litex-hub/linux/tree/litex-rebase (or https://lore.kernel.org/all/20221107171500.2537938-1-gsomlo@gmail.com/). I wouldn't be surprised if this issue is related...