litex-hub/linux-on-litex-vexriscv #190: sbt fails to generate verilog

devobliquezer0, 2021-01-26:
I just updated all the repos, and now sbt can't seem to be able to build verilog.

Error:
```
VexRiscv cluster : VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ood_Wm
Generating cluster netlist
[info] Loading settings for project vexriscv-build from plugins.sbt ...
[info] Loading project definition from /build/FPGA/RISCV/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project
[info] Loading settings for project root from build.sbt ...
[info] Loading settings for project spinalhdl-build from plugin.sbt ...
[info] Loading project definition from /build/FPGA/RISCV/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/project
[info] Loading settings for project all from build.sbt ...
[info] Set current project to VexRiscv (in build file:/build/FPGA/RISCV/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running (fork) vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --cpu-count=1 --ibus-width=32 --dbus-width=32 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=128 --aes-instruction=False --out-of-order-decoder=True --wishbone-memory=True --netlist-name=VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ood_Wm --netlist-directory=/build/FPGA/RISCV/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog
[error] Error: Unknown option --out-of-order-decoder=True
[error] Error: Unknown option --wishbone-memory=True
[error] Try --help for more information.
[error] Exception in thread "main" java.lang.AssertionError: assertion failed
[error] 	at scala.Predef$.assert(Predef.scala:156)
[error] 	at spinal.core.package$.assert(core.scala:420)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.delayedEndpoint$vexriscv$demo$smp$VexRiscvLitexSmpClusterCmdGen$1(VexRiscvSmpLitexCluster.scala:72)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$delayedInit$body.apply(VexRiscvSmpLitexCluster.scala:59)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.collection.immutable.List.foreach(List.scala:392)
[error] 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error] 	at scala.App$class.main(App.scala:76)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.main(VexRiscvSmpLitexCluster.scala:59)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen.main(VexRiscvSmpLitexCluster.scala)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 2 s, completed Jan 26, 2021, 5:30:12 AM
INFO:ECP5PLL:Config:

```
Build: `./make.py --board=camlink_4k --cpu-count=1 --build`

devobliquezer0, 2021-01-26:
This fails on every build that requires sbt generating verilog:
Build: `./make.py --board=nexys4ddr --cpu-count=4 --build`
Error:
```
VexRiscv cluster : VexRiscvLitexSmpCluster_Cc4_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw64_Cdma_Ood
Generating cluster netlist
[info] Loading settings for project vexriscv-build from plugins.sbt ...
[info] Loading project definition from /build/FPGA/RISCV/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project
[info] Loading settings for project root from build.sbt ...
[info] Loading settings for project spinalhdl-build from plugin.sbt ...
[info] Loading project definition from /build/FPGA/RISCV/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/project
[info] Loading settings for project all from build.sbt ...
[info] Set current project to VexRiscv (in build file:/build/FPGA/RISCV/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running (fork) vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --coherent-dma --cpu-count=4 --ibus-width=64 --dbus-width=64 --dcache-size=8192 --icache-size=8192 --dcache-ways=2 --icache-ways=2 --litedram-width=64 --aes-instruction=False --out-of-order-decoder=True --wishbone-memory=False --netlist-name=VexRiscvLitexSmpCluster_Cc4_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw64_Cdma_Ood --netlist-directory=/build/FPGA/RISCV/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog
[error] Error: Unknown option --out-of-order-decoder=True
[error] Error: Unknown option --wishbone-memory=False
[error] Try --help for more information.
[error] Exception in thread "main" java.lang.AssertionError: assertion failed
[error] 	at scala.Predef$.assert(Predef.scala:156)
[error] 	at spinal.core.package$.assert(core.scala:420)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.delayedEndpoint$vexriscv$demo$smp$VexRiscvLitexSmpClusterCmdGen$1(VexRiscvSmpLitexCluster.scala:72)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$delayedInit$body.apply(VexRiscvSmpLitexCluster.scala:59)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.collection.immutable.List.foreach(List.scala:392)
[error] 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error] 	at scala.App$class.main(App.scala:76)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.main(VexRiscvSmpLitexCluster.scala:59)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen.main(VexRiscvSmpLitexCluster.scala)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 2 s, completed Jan 26, 2021, 5:39:39 AM

```

Dolu1990, 2021-01-26:
[error] Error: Unknown option --out-of-order-decoder=True
[error] Error: Unknown option --wishbone-memory=True

Are related to a very recent update where i aded those two options

Basicaly, did you also updated the repo submodules ?
I'm thinking specificaly about those two ones : 
https://github.com/litex-hub/pythondata-cpu-vexriscv_smp/tree/master/pythondata_cpu_vexriscv_smp/verilog/ext

devobliquezer0, 2021-01-26:
> [error] Error: Unknown option --out-of-order-decoder=True
> [error] Error: Unknown option --wishbone-memory=True
>
> Are related to a very recent update where i aded those two options
>
> Basicaly, did you [...]

I though `./litex-setup.py update` is supposed to take care of that. I just ran that, but still had to manually update the pythondata-cpu-vexriscv-smp submodules.
I just updated and it seems to be working now:
```
[sahaj@athena pythondata-cpu-vexriscv-smp]$ git submodule update
Submodule path 'pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL': checked out '1faadab42f64c8f7c3b77ab326588d9c5f45328d'
Submodule path 'pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv': checked out 'd6e8a5ef22042ddb962cfbf627efe909bb23da95'
```

enjoy-digital, 2021-01-27:
@devobliquezer0 with https://github.com/enjoy-digital/litex/commit/61034fe0f97e0ca012d10917ee7c1b1833a024fc `./litex-setup.py update` will now do the `git submodule update` which should slmplify VexRiscv-SMP updates.

closed 2021-01-27 by enjoy-digital