litex-hub/linux-on-litex-vexriscv #193: Illegal instruction on a simple binary

pcotret, 2021-01-30:
Hi,
I'm trying to execute a custom binary with pre-built Linux binaries + Nexys4DDR.
My C code:
```c
int main()
{
    int a=3,b=7,c=3;
    c=a+b;
    return 0;
}
```
Compile it with:
```bash
riscv64-unknown-elf-gcc -march=rv32im -mabi=ilp32 main.c -o main
```
After copying the binary on my SD card, if I execute `main`, I get an "Illegal instruction".
Am I missing something in the compilation process?

pcotret, 2021-02-03:
I'm gonna close my own issue.
I've been able to do what I want by putting my binary in the `rootfs_overlay` directory.

closed 2021-02-03 by pcotret

enjoy-digital, 2021-02-04:
@pcotret: good you've been able to do what you wanted.  Sorry for not  being responsive on this.