litex-hub/linux-on-litex-vexriscv #194: SDRAM initialization fails on ECPIX5

codebfu, 2021-01-30:
I seem to have the same issue #174 with an ecpix-5-85f (R02).
git hashs:
- linux-on-litex-vexriscv : 7cb6d79677db1b2c4f56182fcc06ff781fa8a924
- litex: 69307cfdde8b78aef3334f130545b87176bf6966
- litex-board: abccd12058717aa27e7174d59958dbbfe51102ba

build command used : ./make.py --board=ecpix5 --device=85F --cpu-count=1 --build

console output:
```
 BIOS built on Jan 30 2021 20:05:18
 BIOS CRC passed (0b7830b0)

 Migen git sha1: 40b1092
 LiteX git sha1: 69307cf

--=============== SoC ==================--
CPU:            VexRiscv SMP-LINUX @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            64KiB
SRAM:           8KiB
L2:             0KiB
SDRAM:          524288KiB 16-bit @ 200MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b0: |11100000| delays: 01+-01
  m0, b1: |00000000| delays: -
  m0, b2: |00000000| delays: -
  m0, b3: |00000000| delays: -
  best: m0, b00 delays: 01+-01
  m1, b0: |11100000| delays: 02+-01
  m1, b1: |00000000| delays: -
  m1, b2: |00000000| delays: 00+-00
  m1, b3: |00000000| delays: -
  best: m1, b00 delays: 01+-01
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB
   Read: 0x40000000-0x40200000 2MiB
  bus errors:  2/256
  addr errors: 32/8192
  data errors: 524279/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex> mem_read 0x40000000 128

Memory dump:
0x40000000  03 b9 20 c9 02 00 30 80 01 00 18 c0 03 00 2c 60  .. ...0.......,`
0x40000010  02 00 36 b0 01 00 1b d8 03 00 2d 6c 02 80 36 b6  ..6.......-l..6.
0x40000020  01 c0 9b db 03 60 ed 6d 02 b0 56 b6 01 d8 ab db  .....`.m..V.....
0x40000030  03 6c f5 6d 02 b6 5a b6 81 db ad db c3 6d f6 6d  .l.m..Z......m.m
0x40000040  62 b6 5b b6 b1 5b ad db db ad f6 6d 6e d6 5b b6  b.[..[.....mn.[.
0x40000050  b7 6b ad db d8 b5 f6 6d 6c da 7b b6 b6 6d bd 5b  .k.....ml.{..m.[
0x40000060  5b b6 de 2d ae db 4f 16 d7 6d a7 8b 68 b6 f3 45  [..-..O..m..h..E
0x40000070  b4 db 79 a2 da ed bc 51 6d f6 5e 28 b5 7b 0f 14  ..y....Qm.^(.{..

litex> mem_write 0x40000000 0 16

litex> mem_read 0x40000000 128

Memory dump:
0x40000000  00 3d 00 8a 00 00 00 00 00 00 00 00 00 00 00 00  .=..............
0x40000010  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x40000020  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x40000030  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x40000040  62 00 5b 00 b1 5b ad db db ad f6 6d 6e d6 5b b6  b.[..[.....mn.[.
0x40000050  b7 6b ad db d8 b5 f6 6d 6c da 7b b6 b6 6d bd 5b  .k.....ml.{..m.[
0x40000060  5b b6 de 2d ae db 4f 16 d7 6d a7 8b 68 b6 f3 45  [..-..O..m..h..E
0x40000070  b4 db 79 a2 da ed bc 51 6d f6 5e 28 b5 7b 0f 14  ..y....Qm.^(.{..

litex> mem_write 0x40000000 0x11223344

litex> mem_read 0x40000000 128

Memory dump:
0x40000000  44 3d 22 8a 00 33 00 11 00 00 00 00 00 00 00 00  D="..3..........
0x40000010  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x40000020  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x40000030  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x40000040  62 00 5b 00 b1 5b ad db db ad f6 6d 6e d6 5b b6  b.[..[.....mn.[.
0x40000050  b7 6b ad db d8 b5 f6 6d 6c da 7b b6 b6 6d bd 5b  .k.....ml.{..m.[
0x40000060  5b b6 de 2d ae db 4f 16 d7 6d a7 8b 68 b6 f3 45  [..-..O..m..h..E
0x40000070  b4 db 79 a2 da ed bc 51 6d f6 5e 28 b5 7b 0f 14  ..y....Qm.^(.{..

litex> mem_write 0x40000000 0x55667788

litex> mem_read 0x40000000 128

Memory dump:
0x40000000  88 3d 66 8a 00 77 00 55 00 00 00 00 00 00 00 00  .=f..w.U........
0x40000010  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x40000020  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x40000030  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0x40000040  62 00 5b 00 b1 5b ad db db ad f6 6d 6e d6 5b b6  b.[..[.....mn.[.
0x40000050  b7 6b ad db d8 b5 f6 6d 6c da 7b b6 b6 6d bd 5b  .k.....ml.{..m.[
0x40000060  5b b6 de 2d ae db 4f 16 d7 6d a7 8b 68 b6 f3 45  [..-..O..m..h..E
0x40000070  b4 db 79 a2 da ed bc 51 6d f6 5e 28 b5 7b 0f 14  ..y....Qm.^(.{..

litex>
```

enjoy-digital, 2021-01-31:
Thanks @codebfu for the feedback, I was expecting it to occur. The Wishbone/L2 cache has been reverted with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/83f92358ccb2b49226a43f3e9f8539fe92f85263, this is slower but should work on all boards.

codebfu, 2021-01-31:
Thanks, I'll give it a try asap; I did tinker with l2_size yesterday, but with l2_size > 0 it was missing the corresponding cpu verilog file; missing wishbone if I understand the naming scheme correctly.

codebfu, 2021-01-31:
I have no file ending with _Wm.v

```
-- Executing script file `ecpix5.ys' --

1. Executing Verilog-2005 frontend: /opt/litex-tools/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/RamXilinx.v
Parsing Verilog input from `/opt/litex-tools/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/RamXilinx.v' to AST representation.
Storing AST representation for module `$abstract\Ram_1w_1rs'.
Successfully finished Verilog frontend.
ERROR: Can't open input file `/opt/litex-tools/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ood_Wm.v' for reading: No such file or directory
```

List of available files :
```
VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw128_Cdma_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw128_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw16_Cdma_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw16_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw32_Cdma_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw32_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw64_Cdma_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw64_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is8192Iy2_Dw32Ds8192Dy2_Ldw16_Cdma_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is8192Iy2_Dw32Ds8192Dy2_Ldw16_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is8192Iy2_Dw32Ds8192Dy2_Ldw32_Cdma_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw32Is8192Iy2_Dw32Ds8192Dy2_Ldw32_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw128_Cdma_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw128_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw64_Cdma_Ood.v
VexRiscvLitexSmpCluster_Cc1_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw64_Ood.v
VexRiscvLitexSmpCluster_Cc2_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw128_Cdma_Ood.v
VexRiscvLitexSmpCluster_Cc4_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw128_Cdma_Ood.v
```

codebfu, 2021-01-31:
FYI, I've generated VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ood_Wm.v and now it works as expected

enjoy-digital, 2021-01-31:
@codebfu: This configuration is indeed not part for the pre-generated ones and needs to be generated. Great it's now working and thanks for the feedback. We'll switch back to the direct LiteDRAM interface on ECP5 when the issue will be fully understood and fixed.

closed 2021-01-31 by enjoy-digital