litex-hub/linux-on-litex-vexriscv #195: ./make.py --doc: contents.rst not found

geertu, 2021-02-01:
./make.py --board=orangecrab --doc gives:

> Running Sphinx v1.8.5
> making output directory...
> building [mo]: targets for 0 po files that are out of date
> building [html]: targets for 11 source files that are out of date
> updating environment: 11 added, 0 changed, 0 removed
> reading sources... [  9%] ctrl
> reading sources... [ 18%] ddrphy
> reading sources... [ 27%] i2c0
> reading sources... [ 36%] identifier_mem
> reading sources... [ 45%] index
> reading sources... [ 54%] interrupts
> reading sources... [ 63%] leds
> reading sources... [ 72%] sdram
> reading sources... [ 81%] spisdcard
> reading sources... [ 90%] timer0
> reading sources... [100%] uart
> 
> linux-on-litex-vexriscv/build/orangecrab/doc/identifier_mem.rst:14: WARNING: duplicate label identifier_mem, other instance in linux-on-litex-vexriscv/build/orangecrab/doc/identifier_mem.rst
> linux-on-litex-vexriscv/build/orangecrab/doc/sdram.rst:77: WARNING: Inline literal start-string without end-string.
> 
> Sphinx error:
> master file linux-on-litex-vexriscv/build/orangecrab/doc/contents.rst not found

Same issue with e.g. Arty.

enjoy-digital, 2021-02-01:
@geertu: just tested it successfully here with Sphinx v2.2.2.

geertu, 2021-02-02:
@enjoy-digital Thanks for verifiying!

https://www.sphinx-doc.org/ca/latest/usage/configuration.html?highlight=master_doc
says the default was changed from "contents" to "index" in v2 :-(

Forcing master_doc in LiteX:

> --- a/litex/soc/doc/__init__.py
> +++ b/litex/soc/doc/__init__.py
> @@ -28,6 +28,7 @@ offline_skin_js_path = "https://wavedrom.com/skins/default.js"
>  offline_wavedrom_js_path = "https://wavedrom.com/WaveDrom.js"
>  html_theme = 'alabaster'
>  html_static_path = ['_static']
> +master_doc = 'index'
>  """
>  
>  def generate_svd(soc, buildpath, filename=None, name="soc", **kwargs):

makes it work for me with Sphinx v1.8.5.  Would that still work with v2.2.2?
Thanks!

enjoy-digital, 2021-02-04:
FIxed with https://github.com/enjoy-digital/litex/pull/800, thanks @geertu.

closed 2021-02-04 by enjoy-digital