litex-hub/linux-on-litex-vexriscv #198: Legacy configuration in Buildroot config

pcotret, 2021-02-03:
```bash
git clone https://github.com/litex-hub/linux-on-litex-vexriscv
git clone http://github.com/buildroot/buildroot
cd buildroot
make BR2_EXTERNAL=../linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig
make
 ```
 Got the following error:
 ```
Makefile.legacy:9: *** "You have legacy configuration in your .config! Please check your configuration.".  Stop.
make: *** [Makefile:84: _all] Error 2
```
It should mean some parameters in the `defconfig` file may not be compliant with the latest Buildroot. Can't look into details for the moment.

enjoy-digital, 2021-02-04:
Thanks @pcotret, the explicit binutils was not required, I set it when trying to understand the issue with the 5.11 kernel but this is no longer required.

closed 2021-02-04 by enjoy-digital

duanjiulon, 2024-07-09:
> Thanks @pcotret, the explicit binutils was not required, I set it when trying to understand the issue with the 5.11 kernel but this is no longer required.

Hi, why did I add this line of code to the corresponding vex_default_config and then use the command, but it still has the same error

Dolu1990, 2024-07-09:
Replace BR2_GCC_VERSION_11_X with BR2_GCC_VERSION_13_X should fix it.
Let us know how it goes ^^

duanjiulon, 2024-07-09:
> Replace BR2_GCC_VERSION_11_X with BR2_GCC_VERSION_13_X should fix it. Let us know how it goes ^^

oh，dear，thanks so much！This kernel can also be used as the kernel of the SoC where the nax_litex is located, right? I remember your documentation saying it was compatible

duanjiulon, 2024-07-09:
> Replace BR2_GCC_VERSION_11_X with BR2_GCC_VERSION_13_X should fix it. Let us know how it goes ^^

I have another questions：
```
git clone https://github.com/Dolu1990/litex-linux.git
cd litex-linux
git checkout ae80e67c6b48bbedcd13db753237a25b3dec8301
# Increase speed by factor 10
sed -i 's/SD_SLEEP_US       5/SD_SLEEP_US       0/g' drivers/mmc/host/litex_mmc.c
cp $NAXRISCV/ext/NaxSoftware/debian2/linux/.config .config

export CROSS_COMPILE=/opt/riscv_rv64gc_linux/bin/riscv64-unknown-linux-gnu-
make -j$(nproc) ARCH=riscv oldconfig
make -j$(nproc) ARCH=riscv all

ls vmlinux               # Kernel elf (has symboles, usefull for debug)
ls arch/riscv/boot/Image # Kernel binary

cd ..
unset CROSS_COMPILE
```
I can not find the file /NaxSoftware/debian2 in your git,what's wrong with it?

Dolu1990, 2024-07-09:
> This kernel can also be used as the kernel of the SoC where the nax_litex is located, right?

Yes, there should be no problem with this

> I can not find the file /NaxSoftware/debian2 in your git,what's wrong with it?

It probably mean https://github.com/SpinalHDL/NaxSoftware/tree/main/debian_litex

Also note for vexii i redid the procedure, especialy about the debian image itself : 
https://github.com/SpinalHDL/VexiiRiscv/tree/dev/doc/litex/debian

duanjiulon, 2024-07-10:
> Also note for vexii i redid the procedure, especialy about the debian image itself : https://github.com/SpinalHDL/VexiiRiscv/tree/dev/doc/litex/debian

I get it， thank you so much,Recently I successfully rewrote ibus to axi-read only interface, dbus already has axi interface, but pbus only has axi4lite interface, so I can't successfully connect nax-core to axi-crossbar, I would like to ask you to help me generate a plungins wose pbus related axi-full interface, you can directly connect to axi-crossbar, like briey's soc. Thank you

Dolu1990, 2024-07-12:
You would just need to bridge AXI-lite -> AXI4 right ? If yes, this is a straight forward connection with no logic required, just connections + constants

commit 70db0edf (2021-02-04): buildroot/linux_vexriscv_defconfig: remove explicit binutils version (not required and no version no longer supported in upstream buildroot), thanks @pcotret.

Fixes #198.