litex-hub/linux-on-litex-vexriscv #200: Add to SkiffOS

paralin, 2021-02-11:
SkiffOS is a configuration package management tool for Buildroot which makes it easy to retarget a system across a wide variety of underlying hardware.

Considering that litex is already using buildroot, it would be easy to format the configs, patches, etc, into a Skiff package, and then use that to load other distros (Gentoo, Ubuntu, NixOS, ...) easily into the system while inheriting compiler information.

paralin, 2021-02-11:
The `litex/vexriscv` package has been implemented based on this repo:

https://github.com/skiffos/SkiffOS/pull/135

https://github.com/skiffos/SkiffOS/tree/litex/configs/litex/vexriscv

I don't have any hardware to test at the moment.

enjoy-digital, 2021-02-18:
Thanks @paralin, this seems really interesting and we'll explore this. From what I understand you've already been able to run the image with `sim.py` from `Linux-on-LiteX-VexRiscv`? If so, this should also run on hardware and I would be happy to do tests on hardware for you.

paralin, 2021-02-18:
Apparently a lot of additional Risc-v support just landed in Go and Docker as well. I've added a squashfs based boot mechanism that doesn't use an initramfs (to save memory) and most of the kernel options can probably be disabled. I know these chips have very low resources so as minimal of an image as possible is necessary. 

The current output from Skiff should work with the sim script, there's a TODO to add a "make" command to run it automatically (by downloading this repo & running it).

Will post again here once I've tested it with sim.py

paralin, 2022-05-10:
Allwinner Nezha support is in place in SkiffOS, which is a risc-v board commonly available with D1 by Allwinner.

Planning to revisit adding the litex config soon.

[image]

(This recently included adding support for riscv to Go in Buildroot).

enjoy-digital, 2022-05-10:
Great, thanks for the update @paralin! Really interesting. Compiling custom packages with buildroot is becoming a bit complicated when testing Linux-on-LiteX-VexRiscv and SkiffOS could be really interesting. @Dolu1990 (who might also be interested by this) also implemented a new NaxRiscv CPU that is 64-bit capable and could simplify your work since probably closer to the D1. We are also able to use Rocket as 64-bit CPU.
 
For NaxRiscv, add to your target/sim command: `--cpu-type=naxriscv --xlen=64`
For Rocket: `--cpu-type=rocket`

enjoy-digital, 2022-05-10:
Happy to help you getting started or do some tests for you in simulation or on hardware if needed.

paralin, 2023-09-15:
@enjoy-digital I'm looking to finish this work soon, circling back. We support the D1 in SkiffOS already today.

Could add a command to run the simulator.