litex-hub/linux-on-litex-vexriscv #202: project does not fit into de0-anno FPGA

mczerski, 2021-02-24:
The README page claims that the de0-nano board is supported but running this command:
```
./make.py --board=de0nano --cpu-count=1 --build
```
fails on fitting stage with this error:
```
Error (170011): Design contains 32900 blocks of type combinational node.  However, the device contains only 22320 blocks.
```

Does this project really supports de0-nano board ?

PS.
This is the content of the de0nano.map.summary:
```
Analysis & Synthesis Status : Successful - Wed Feb 24 07:41:46 2021
Quartus Prime Version : 20.1.1 Build 720 11/11/2020 SJ Lite Edition
Revision Name : de0nano
Top-level Entity Name : de0nano
Family : Cyclone IV E
Total logic elements : 68,298
    Total combinational functions : 32,900
    Dedicated logic registers : 37,848
Total registers : 37926
Total pins : 50
Total virtual pins : 0
Total memory bits : 384,706
Embedded Multiplier 9-bit elements : 8
Total PLLs : 1
```

Dolu1990, 2021-02-24:
One tip maybe to use the additional --without-out-of-order-decoder --with-wishbone-memory in order to reduce area. Especialy that the out of order decoder normaly require distributed ram that the de0nano do not have.

mczerski, 2021-02-24:
Thanks for fast response.
unfortunately running this command:
```
./make.py --without-out-of-order-decoder --with-wishbone-memory --board=de0nano --cpu-count=1 --build
```
gives exactly the same result.

Dolu1990, 2021-02-24:
Hooooo, got it. So, basicaly the issue is https://github.com/litex-hub/pythondata-cpu-vexriscv_smp/blob/master/pythondata_cpu_vexriscv_smp/verilog/RamXilinx.v

It is used to infer simple dual port ram with byte enable. This is used in the data cache of VexRiscv, and isn't properly infered by Quartus.

So the solution is to have an alternative version of this file that Quartus enjoy ^^

Dolu1990, 2021-02-24:
Here is the denano.fit.rpt line which demonstrate the issue : 

```
+--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------+
; Fitter Resource Utilization by Entity                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  ;
+------------------------------------------------------------------------------------------------------------------------------+-------------+---------------------------+---------------+-------------+--------------+---------+-----------+------+--------------+--------------+-------------------+------------------+---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------+-------------------------------------------------------------+--------------+
; Compilation Hierarchy Node                                                                                                   ; Logic Cells ; Dedicated Logic Registers ; I/O Registers ; Memory Bits ; DSP Elements ; DSP 9x9 ; DSP 18x18 ; Pins ; Virtual Pins ; LUT-Only LCs ; Register-Only LCs ; LUT/Register LCs ; Full Hierarchy Name                                                                                                                                                                                                                                                                               ; Entity Name                                                 ; Library Name ;
+------------------------------------------------------------------------------------------------------------------------------+-------------+---------------------------+---------------+-------------+--------------+---------+-----------+------+--------------+--------------+-------------------+------------------+---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------+-------------------------------------------------------------+--------------+
;             |Ram_1w_1rs:ways_0_data|                                                                                         ; 0 (0)       ; 32800 (32800)             ; 0 (0)         ; 0           ; 0            ; 0       ; 0         ; 0    ; 0            ; 0 (0)        ; 0 (0)             ; 0 (0)            ; |de0nano|VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1__Wm:VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1__Wm|VexRiscv:cores_0_cpu_logic_cpu|DataCache:dataCache_1|Ram_1w_1rs:ways_0_data                                                                                      ; Ram_1w_1rs                                                  ; work         ;
```

mczerski, 2021-02-24:
it seems that moving the loop over mask bits to the most outer block and merging wr_en and wr_mask signals into single if statement makes the quartus happy:
```
    reg [wrDataWidth-1:0] ram_block [(2**wrAddressWidth)-1:0];
    integer i;
    localparam COL_WIDTH = wrDataWidth/wrMaskWidth;
    always @ (posedge wr_clk) begin
        //if(wr_en) begin
        //    for(i=0;i<wrMaskWidth;i=i+1) begin
        //        if(wr_mask[i]) begin
        //            ram_block[wr_addr][i*COL_WIDTH +: COL_WIDTH] <= wr_data[i*COL_WIDTH +:COL_WIDTH];
        //        end
        //    end
        for (i=0;i<wrMaskWidth;i=i+1) begin
            if(wr_en && wr_mask[i]) begin
                ram_block[wr_addr][i*COL_WIDTH +: COL_WIDTH] <= wr_data[i*COL_WIDTH +:COL_WIDTH];
            end
        end
    end
```
even making separate ifs for wr_en and wr_mask  makes quartus to not infer ram properly, like this:
```
    reg [wrDataWidth-1:0] ram_block [(2**wrAddressWidth)-1:0];
    integer i;
    localparam COL_WIDTH = wrDataWidth/wrMaskWidth;
    always @ (posedge wr_clk) begin
        //if(wr_en) begin
        //    for(i=0;i<wrMaskWidth;i=i+1) begin
        //        if(wr_mask[i]) begin
        //            ram_block[wr_addr][i*COL_WIDTH +: COL_WIDTH] <= wr_data[i*COL_WIDTH +:COL_WIDTH];
        //        end
        //    end
        for (i=0;i<wrMaskWidth;i=i+1) begin
            if(wr_en) begin
                if (wr_mask[i]) begin
                    ram_block[wr_addr][i*COL_WIDTH +: COL_WIDTH] <= wr_data[i*COL_WIDTH +:COL_WIDTH];
                end
            end
        end
    end
```

How to ensure that this change does not brake anything ?

Dolu1990, 2021-02-25:
Chaning that for other synthesis tool is very very likely to break stuff. I think the best would be to make the flow flexible and select the right implementation depending the synthesis target.

Then to know if that break stuff, hmm i gess that's require synthesis on ecp5/artix7/cycloneIV and manual checks

enjoy-digital, 2021-03-11:
@mczerski, @Dolu1990: Thanks for looking at this. We also had similar issues with the verilog memories generated by Migen/LiteX in the past. In the long term, that would indeed be nice to provide specific implementation in LiteX for the different FPGA families, but that's possible we can find a quick workaround first. I'm going to look at https://github.com/enjoy-digital/liteeth/issues/63 first to refresh my mind :).

enjoy-digital, 2021-03-30:
Thanks @mczerski for the implementation tests, I used it to create a specific Intel/Altera version of the RAM here: https://github.com/litex-hub/pythondata-cpu-vexriscv_smp/commit/69e3c1e71b404b1677869cc9755db2df27d81ba9 and this version will now be used directly by LiteX on Intel/Altera plaforms with https://github.com/enjoy-digital/litex/commit/080ecad522ad2dec43295517f5111822022554f4. 

`./make.py --board=de0nano --cpu-count=1 --build` is now building correctly.

closed 2021-03-30 by enjoy-digital