litex-hub/linux-on-litex-vexriscv #203: Errors while running on ulx3s

soundnut, 2021-02-25:
Greetings,
I've managed to load the gateware onto a ulx3s board but run into some errors while booting linux.
While booting via serial port without an SD Card present I get these errors:

[   24.479545] mmc0: Skipping voltage switch
[   24.505508] mmc0: Skipping voltage switch
[   24.533908] mmc0: Skipping voltage switch
[   24.549724] mmc0: Skipping voltage switch
[   24.569306] mmc0: Skipping voltage switch
[   24.589921] mmc0: Skipping voltage switch
[   24.609325] mmc0: Skipping voltage switch
[   24.641615] mmc0: Skipping voltage switch

adding an SD Card avoids the Skipping voltage switch messages but then I get these errors

[     9.813537] Initramfs unpacking failed: invalid magic at start of compressed archive
[   10.885452] Freeing initrd memory: 8192K
[   10.963709] workingset: timestamp_bits=30 max_order=13 bucket_order=0
[   12.708147] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 253)
[   12.720843] io scheduler mq-deadline registered
[   12.729679] io scheduler kyber registered
[   12.816421] LiteX SoC Controller driver initialized: subreg:4, align:4
[   20.001164] f0001000.serial: ttyLXU0 at MMIO 0x0 (irq = 0, base_baud = 0) is a liteuart
[   20.028736] printk: console [liteuart0] enabled
[   20.028736] printk: console [liteuart0] enabled
[   20.037965] printk: bootconsole [sbi0] disabled
[   20.037965] printk: bootconsole [sbi0] disabled
[   20.265007] libphy: Fixed MDIO Bus: probed
[   20.289828] i2c /dev entries driver
**[   20.380371] litex-mmc f0004000.mmc: Requested clk_freq=12500000: set to 12500000 via div=4**
[   20.647790] NET: Registered protocol family 10
**[   21.731419] Command (cmd 8) failed, status 2
[   22.735655] Command (cmd 55) failed, status 2
[   23.739990] Command (cmd 55) failed, status 2
[   24.744328] Command (cmd 55) failed, status 2
[   25.748643] Command (cmd 55) failed, status 2
[   26.753056] Command (cmd 1) failed, status 2
[   26.757372] litex-mmc f0004000.mmc: Requested clk_freq=0: set to 195312 via div=256**
[   26.796348] Segment Routing with IPv6
[   26.804988] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   26.960523] Freeing unused kernel memory: 204K
[   26.964862] Kernel memory protection not selected by kernel config.
[   26.971821] Run /init as init process
[   30.405674] tmpfs: Unknown parameter 'mode'
mount: mounting tmpfs on /dev/shm failed: Invalid argument
[   30.469423] tmpfs: Unknown parameter 'mode'
mount: mounting tmpfs on /tmp failed: Invalid argument
[   30.533486] tmpfs: Unknown parameter 'mode'
mount: mounting tmpfs on /run failed: Invalid argument
**Starting syslogd: [   34.245435] litex-mmc f0004000.mmc: Requested clk_freq=12500000: set to 12500000 via div=4
[   35.281997] Command (cmd 8) failed, status 2
[   36.288165] Command (cmd 55) failed, status 2
[   37.292461] Command (cmd 55) failed, status 2
[   38.297067] Command (cmd 55) failed, status 2
[   39.301388] Command (cmd 55) failed, status 2
[   40.306086] Command (cmd 1) failed, status 2
[   40.311978] litex-mmc f0004000.mmc: Requested clk_freq=0: set to 195312 via div=256**
OK
Starting klogd: OK
Running sysctl: OK
Saving random seed: [   50.577315] random: dd: uninitialized urandom read (512 bytes read)
OK
**Starting network: [   52.365600] litex-mmc f0004000.mmc: Requested clk_freq=12500000: set to 12500000 via div=4
[   53.404990] Command (cmd 8) failed, status 2
[   54.409495] Command (cmd 55) failed, status 2
[   55.413821] Command (cmd 55) failed, status 2
[   56.419923] Command (cmd 55) failed, status 2
[   57.424750] Command (cmd 55) failed, status 2
[   58.429222] Command (cmd 1) failed, status 2
[   58.433883] litex-mmc f0004000.mmc: Requested clk_freq=0: set to 195312 via div=256**
OK

Welcome to Buildroot

The Command failed and clk_freq  messages cycle over and over. I can enter commands in between but would like to get rid of these messages to really do something with it.

During another boot attempt I get these errors
[   80.131476] mmc0: card has unknown MMCA version 15
[   80.136409] mmc0: error -22 whilst initialising MMC card
[   80.141037] litex-mmc f0004000.mmc: Requested clk_freq=0: set to 195312 via div=256
[   81.216574] litex-mmc f0004000.mmc: Requested clk_freq=12500000: set to 12500000 via div=4
[   81.256422] litex-mmc f0004000.mmc: card claims to support voltages below defined range
[   81.261612] litex-mmc f0004000.mmc: Requested clk_freq=0: set to 195312 via div=256
[   81.305284] litex-mmc f0004000.mmc: Requested clk_freq=12500000: set to 12500000 via div=4

I've tried with 3 different SD Cards - they all behave the same way. (2GB and 4GB - re-formated to fat16)

Any ideas?
Thanks

developandplay, 2021-03-29:
I encounter your second error on my ULX3S regardless of whether the SD-Card is inserted or not.
One warning that I noticed was this one `Max frequency for clock '$glbnet$main_clkout0'` which for me was under 50Mhz.

For now disabling sdcard support by removing these two lines fixed the problems for me.
https://github.com/litex-hub/linux-on-litex-vexriscv/blob/1da148d8e2b151bbb59ef816d8d5c517e58db811/make.py#L291-L292

soundnut, 2021-03-31:
Thanks for the feedback/confirmation. 
I've had more luck with SaxonSoc but will try again with your proposed change.

developandplay, 2021-06-10:
Just found out that seems to be a common issue: https://github.com/litex-hub/linux-on-litex-vexriscv/issues/171#issuecomment-763728692
I'm looking into it some more. Maybe we can apply a similar fix for the ulx3s.

PR #262 (2022-01-04): ulx3s: use spisdcard instead of sdcard
The ULX3S does not have a card detect (CD) pin for the SD interface,
which breaks the sdcard driver in Linux.

This patch changes the ULX3S to use the "spisdcard" interface instead
of the "sdcard" interface.  With this change the SD card works fine
from the BIOS and Linux in SPI-mode.

See also emard/ulx3s#15

Fixes: #203
closed 2022-01-04 by enjoy-digital