litex-hub/linux-on-litex-vexriscv #204: Error at ./sim.py in "Running litex simulation"

jhu960213, 2021-03-04:
Hello,

Each step of the: https://github.com/litex-hub/linux-on-litex-vexriscv works fine up until when I run: ./sim.py. Running the litex simulation of the linux kernel gave me these outputs. Please see attached outputs.  

[image]
[image]
[image]

I've looked into the source code and apparently in the "linux-on-litex-vexriscv/images" folder was missing 3 files:
1) Image
2) rootfs.cpio
3) opensbi.bin

However, rv32.dtb is there. The branch where I cloned the project was from the master branch if that matters. 

I'm not sure if those 3 files are supposed to be there or not but when I created dummy ones the error during compilation is gone then I get stuck at the  "Lift-Off!" part:

[image]

I'm running ubuntu 20.04.2 LTS if this helps

Thanks!

Best,
J.H

geertu, 2021-03-05:
Please find prebuilt images in #164.

enjoy-digital, 2021-03-06:
@jhu960213: Booting the simulation up to the Linux prompt can take some time yes, between 5-10 min depending your computer, unpacking initramfs is indeed a long step.

hasheddan, 2021-03-08:
@jhu960213 confirmed initramfs takes quite a while locally but should eventually pass if not showing any errors

enjoy-digital, 2021-03-11:
I think we can now close this, the error reporting in get_mem_data has been improved a bit with https://github.com/enjoy-digital/litex/commit/a81d1da98008e450cb8b11a4c9b9c379c1124fcd.

closed 2021-03-11 by enjoy-digital