litex-hub/linux-on-litex-vexriscv #206: OpenSBI hangs during serial boot on Arty A7

hasheddan, 2021-03-08:
After loading bitstream (`/make.py --board=arty --load`), when serial booting (`sudo lxterm --images=images/boot.json /dev/ttyUSB1 --serial-boot --speed=1e6`) on Digilent Arty A7 OpenSBI appears to hang forever on `sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver`. Tried it out with both kernel versions from #164. Let me know if there is any additional information I can provide. Thanks for y'all's awesome work on this!

```
--============= Liftoff! ===============--

OpenSBI v0.8-1-gecf7701
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : LiteX / VexRiscv-SMP
Platform Features   : timer,mfdeleg
Platform HART Count : 8
Boot HART ID        : 0
Boot HART ISA       : rv32imas
BOOT HART Features  : time
BOOT HART PMP Count : 0
Firmware Base       : 0x40f00000
Firmware Size       : 124 KB
Runtime SBI Version : 0.2

MIDELEG : 0x00000222
MEDELEG : 0x0000b101
[    0.000000] Linux version 5.11.0-rc2 (florent@panda) (riscv32-buildroot-linux-gnu-gcc.br_real (Buildroot 2020.11-281-g69e5046e7b) 10.2.0, GNU ld (GNU Binutils) 2.33.1) #2 SMP Mon Jan 11 10:05:20 CET 2021

...
[    1.606276] LiteX SoC Controller driver initialized: subreg:4, align:4
[    2.597710] f0001000.serial: ttyLXU0 at MMIO 0x0 (irq = 0, base_baud = 0) is a liteuart
[    2.602984] printk: console [liteuart0] enabled
[    2.602984] printk: console [liteuart0] enabled
[    2.605283] printk: bootconsole [sbi0] disabled
[    2.605283] printk: bootconsole [sbi0] disabled
[    2.623210] litex-spiflash f0006000.spiflash: s25fl128s1 (16384 Kbytes)
[    2.665384] libphy: Fixed MDIO Bus: probed
[    2.678509] liteeth f0004800.mac eth0: irq 2, mapped at a0415800
[    2.683099] i2c /dev entries driver
[    2.689960] i2c i2c-0: Not I2C compliant: can't read SCL
[    2.690861] i2c i2c-0: Bus may be unreliable
[    2.720638] litex-mmc f0006800.mmc: Requested clk_freq=12500000: set to 12500000 via div=8
[    2.749172] fpga_manager fpga0: LiteX ICAPBitstream FPGA Manager registered
[    2.773361] NET: Registered protocol family 10
[    2.788482] Segment Routing with IPv6
[    2.790187] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
```

Dolu1990, 2021-03-09:
Hi,

Does it hang systematicaly at the exact same point ?

About the hang, it seems to me it is in the kernel boot, not opensbi ^^

Maybe it is related to something broken in liteeth ? Can you try a boot without liteeth enabled ?

hasheddan, 2021-03-09:
@Dolu1990 thanks for following up! Yes it hangs at the exact same point every time. To boot without liteeth, should I just disable it [here](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/a3c6215ae30aec81f372a5bc82fcb6594569c79b/buildroot/board/litex_vexriscv/linux.config#L51) by setting to `n` and rebuild the linux images?

Dolu1990, 2021-03-09:
About litheth, i think there is an option into the make.py command line isn't it ? check make.py --help
(sorry i'm in travel, i can't check exactly)

This should be simpler than rebuilding a whole linux image.

hasheddan, 2021-03-09:
@Dolu1990 no worries at all, thank you for your help! I don't see a flag, but I'm guessing commenting out the `ethernet` capability would have the desired effect:

https://github.com/litex-hub/linux-on-litex-vexriscv/blob/7dde41de91aee99a52337b2deebc31c91b7ce862/make.py#L58

hasheddan, 2021-03-10:
Hmm that doesn't seem to change anything...

geertu, 2021-03-10:
Your issue looks very similar to #176. However, the images you used should contain the related fixes.

hasheddan, 2021-03-10:
@geertu is it possible the images were not updated after the fix went in? (https://github.com/litex-hub/linux-on-litex-vexriscv/issues/176#issuecomment-757876329)

enjoy-digital, 2021-03-11:
@hasheddan: Just to confirm, commenting out `ethernet `capability in the target will disable Ethernet (both in the gateware and linux (through the .dts)). I'm going to rebuild and test the Arty target with prebuilt linux image.

enjoy-digital, 2021-03-11:
@hasheddan: I don't see the issue on my setup with a fresh bitstream and using prebuilt OpenSBI/Linux images.
Can you test with the bitstream/images attached?

I just loaded the bistream and then the images with: `litex_term /dev/ttyUSBX --speed=1e6 --images=images/boot.json`

[arty_2021_03_11.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/6121598/arty_2021_03_11.zip)
[images_2021_03_11.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/6121600/images_2021_03_11.zip)

hasheddan, 2021-03-11:
@enjoy-digital thanks for following up! I'll give those a try. Do they match the ones on #164?

hasheddan, 2021-03-11:
@enjoy-digital those artifacts work! I wonder if the ones listed on #164 are stale?
[image]

enjoy-digital, 2021-03-11:
@hasheddan: Good, to be sure I reused the Linux/OpenSBI images from #164 (but rebuilt the bitstream). So the difference with your previous tests is the bistream and rv32.dtb.

enjoy-digital, 2021-03-26:
Not sure what happened exactly, but I think we can close this since has been tested sucesfully with the re-generated bitstream and prebuilt images.

closed 2021-03-26 by enjoy-digital