litex-hub/linux-on-litex-vexriscv #207: Loading bit stream error for kc705 board

jhu960213, 2021-03-08:
Hello,

I was able to build properly for the kc705 board for the Linux on vexriscv project but was unable to load the bitstream by running the command:

./make.py --board=kc705 --cpu-count=4 --load

I don't really understand what the debug output is referring and I feel like it might be a simple issue or permissions thing I've overlooked. Attached below is the debug output. 

[image]

please advise :) Thanks! 

P.S: I'm running ubuntu 20.04TS 64bit

Best,
J.H

jhu960213, 2021-03-10:
@enjoy-digital

enjoy-digital, 2021-03-11:
Hi @jhu960213,

this could be related to a change in OpenOCD. Can you try with this version: https://github.com/enjoy-digital/openocd? Also if you really have trouble with OpenOCD, you can use Vivado GUI and the Hardware Manager.

jhu960213, 2021-03-11:
@enjoy-digital 
Hello,

So I tried building with that version of opened and I ran into a lot of build errors when I was trying to install openocd. Also I noticed that the enjoy-digital's version is very old. 

Best,
J.H

enjoy-digital, 2021-03-11:
@jhu960213: OK, can you try loading the bistream with Vivado?

jhu960213, 2021-03-11:
Hello @enjoy-digital 

I was finally able to load the bit stream in the end to board = kc705 :). It was a permission issue with opencd, I added myself to the etc/udev/rules.d for accessing linux peripherals over Jtag/USB. But now I have a question, it seems like its taking a while to load the linux kernel over serial/Jtag. Is this the expected behaviour? Also this is the command I used:

[image]

Do you guys seem a problem with how I configured the /dev/ttyUSBX part? How do I know which usb port my fpga is connected to? 

Thanks! 

Best,
J.H

jhu960213, 2021-03-12:
Hello @enjoy-digital 

So indeed when I run the lxterm command, the bios doesn't boot up or show any outputs at all. It just hangs there. Please advise, I'm sure how to go from here. 

[image]

Best,
J.H

enjoy-digital, 2021-03-12:
Hi @jhu960213,

the UART is not capable of 1MBauds on the KC705, to it has been reduced to 500KBauds (--speed=500e3).

jhu960213, 2021-03-12:
@enjoy-digital 
Hello,

I tried it with the new speed but I still get stuck with no outputs and bios prompt. 

J.H

jhu960213, 2021-03-12:
Hello @enjoy-digital 

So i was able to load the linux image in the end with the lxterm command (I was using a wrong usb port and didnt have read/write persmission on it etc) but now when it finished booting up it sort of repeatedly prints out these debug outputs to console... Only when i hit enter it brings it back to a normal linux commandline style shown below but if I don't hit enter anymore it will just keep on printing these values. What's going on? 

[image]

I'm running linux on the KC705 board fyi. 

Thanks!
J.H

zhuzhizhan, 2021-06-16:
Hi, 
I have the same problem in nexys4ddr. I have solved it with: 1.remove the file with the ft2232h info in /etc/udev/rules.d/. 2.touch 99-openocd.rules with info "SUBSYSTEM=="tty", ATTRS{idVendor}=="0403",ATTRS{idProduct}=="6010", MODE="664", GROUP="plugdev"
SUBSYSTEM=="tty", ATTRS{idVendor}=="15ba",ATTRS{idProduct}=="002a", MODE="664", GROUP="plugdev"
SUBSYSTEM=="usb", ATTR{idVendor}=="0403",ATTR{idProduct}=="6010", MODE="664", GROUP="plugdev"
SUBSYSTEM=="usb", ATTR{idVendor}=="15ba",ATTR{idProduct}=="002a", MODE="664", GROUP="plugdev""
3.build the openocd with  https://github.com/enjoy-digital/openocd.

enjoy-digital, 2022-03-18:
I think we can now close this issue since the initial problem has been solved. For the SDCard issue, LiteSDCard core and drivers have been udpated since then and are working fine on other hardware, so it would need to be tested again.

closed 2022-03-18 by enjoy-digital

zhuzhizhan, 2022-03-18:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。