litex-hub/linux-on-litex-vexriscv #213: booting linux on zcu104

zld932, 2021-03-15:
hello
The bitstream for zcu104 has been generated. And I have load the bitstream using the following command successfully.
./make.py --board=zcu104 --cpu-count=1 --load

but when I run lxterm, the bios doesn't boot up or show any outputs. It just hangs there. 

[image]

what can I do to run linux, or how do I configure zcu104 to run
Thanks

closed 2021-03-15 by zld932

DWhitey, 2021-04-01:
Hi, I have the same issue at the moment. I built the bitsream and loaded it successfully. What did you do that lxterm doesn't get stuck?
Thanks

zld932, 2021-04-02:
> Hi, I have the same issue at the moment. I built the bitstream and loaded it successfully. What did you do that lxterm doesn't get stuck?
> 
> Thanks

ZCU104 has more than one port. So try other ports when you use lxterm. When you are struck, press enter. When you have response which  is same with terminal, type in reboot. 
Best wishes.

zld932, 2021-04-02:
> Hi, I have the same issue at the moment. I built the bitstream and loaded it successfully. What did you do that lxterm doesn't get stuck?
> 
> Thanks

The speed is 1e6

DWhitey, 2021-04-06:
> When you are struck, press enter.

Hi, thanks for your answer. That helped me. Well, I forgot to add a DIMM as well..