litex-hub/linux-on-litex-vexriscv #214: Buildroot defconfig point to a dead linux repo hash

Dolu1990, 2021-03-17:
https://github.com/litex-hub/linux-on-litex-vexriscv/blame/master/buildroot/configs/litex_vexriscv_defconfig#L29 hash is confusing buildroot flow.

If you look at https://github.com/litex-hub/linux/commit/7d6b20bd1aa8276d92504fa540bd0a903ef43610 you can notice the "This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository."

There is the buildroot build issue : 

```
wget --passive-ftp -nd -t 3 -O '/media/data/open/litex/buildroot/output/build/.linux-7d6b20bd1aa8276d92504fa540bd0a903ef43610-br1.tar.gz.6EFL7m/output' 'http://sources.buildroot.net/linux-7d6b20bd1aa8276d92504fa540bd0a903ef43610-br1.tar.gz' 
--2021-03-17 12:49:00--  http://sources.buildroot.net/linux-7d6b20bd1aa8276d92504fa540bd0a903ef43610-br1.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 172.67.72.56, 104.26.1.37, 104.26.0.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|172.67.72.56|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-03-17 12:49:00 ERROR 404: Not Found.
```

I'm not sure which branch should be used on https://github.com/litex-hub/linux

geertu, 2021-03-17:
This happened because@gsomlo rebased his branch, and this commit is no longer part of any branch/tag in the litex-hub/linux repository.
Running "git gc" (which may happen automatically from time to time) may even remove such commits from your local clone[1].
To avoid this, commits referenced from buildroot defconfigs should be tagged in litex-hub/linux.

[1] It did for me, but (for now) I could get it back using "git fetch git@github.com:litex-hub/linux.git 7d6b20bd1aa8276d92504fa540bd0a903ef43610"

Dolu1990, 2021-03-17:
ok thanks ^^

Hmm i will try with https://github.com/litex-hub/linux/commit/aa7ec90c54ee8555e7cfff33f0de599a002b90aa, it seems close enough

mithro, 2021-03-17:
I pushed a branch called litex-br-v0.0.0 so that commit remains referenced
in the repository.

Dolu1990, 2021-03-17:
Thanks.

Hmm maybe we got the wrong diagnostic, i'm quite confused XD

Basicaly, this issue seems to be related to buildroot itself, i tried to rebuild everything, got the same kind of error again and again. Then i swaped buildroot itself with an older version (1672e250106d5c9a2003f40685e57610238a732e) and there everything worked fine XD

There is a full log of the upstream  build error from buildroot (in the error.log)
[error.log](https://github.com/litex-hub/linux-on-litex-vexriscv/files/6158204/error.log)

So, maybe we should fix the buildroot version to a operationnal version.

Dolu1990, 2021-03-17:
(i'm now continuing on that older buildroot version)

enjoy-digital, 2021-03-29:
I just updated the defconfig with the latest litex-rebase version (5.12.rc4) with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/1da148d8e2b151bbb59ef816d8d5c517e58db811 and also added the updated images to https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164.

closed 2021-03-29 by enjoy-digital