litex-hub/linux-on-litex-vexriscv #215: FPU support

Dolu1990, 2021-03-17:
Hi,

Got the FPU added in VexRiscv and linux-on-litex-vexriscv  (not merged yet) via 
- https://github.com/enjoy-digital/litex/pull/850
- https://github.com/litex-hub/pythondata-cpu-vexriscv_smp/pull/4

There is some documentation about the FPU here : 
https://github.com/SpinalHDL/VexRiscv/tree/fiber#fpu

In short,
- support both F32/F64, subnormal and all 5 rounding mods
- can be shared between multiple CPU to save area
- can schedule most operations each cycle (as long there is no inter depedancies)
- so far, it shouln't impact FMax much (at least on Artix7)
- test with 2 CPU, if more is used, we might need to pipeline the connections between the FPU and the core a bit more.
- actualy cost 3662 LUT on Artix7

TODO : 

Getting the buildroot image to build (see https://github.com/litex-hub/linux-on-litex-vexriscv/issues/214)

The only requirements to get the FPU to work in buildroot is to add the following in the buildroot defconfig :
BR2_RISCV_ISA_CUSTOM_RVF=y
BR2_RISCV_ISA_CUSTOM_RVD=y

The DTS generation is already fixed, opensbi do not seems to require a rebuild.

mithro, 2021-03-17:
This is *super* cool!

Dolu1990, 2021-03-17:
Got it to work !

```sh
Welcome to Buildroot
buildroot login: root
                   __   _
                  / /  (_)__  __ ____ __
                 / /__/ / _ \/ // /\ \ /
                /____/_/_//_/\_,_//_\_\
                      / _ \/ _ \
   __   _ __      _  _\___/_//_/         ___  _
  / /  (_) /____ | |/_/__| | / /____ __ / _ \(_)__ _____  __
 / /__/ / __/ -_)>  </___/ |/ / -_) \ // , _/ (_-</ __/ |/ /
/____/_/\__/\__/_/|_|____|___/\__/_\_\/_/|_/_/___/\__/|___/
                  / __/  |/  / _ \
                 _\ \/ /|_/ / ___/
                /___/_/  /_/_/
  32-bit RISC-V Linux running on LiteX / VexRiscv-SMP.

login[77]: root login on 'console'
root@buildroot:~# cat /proc/cpuinfo 
processor	: 0
hart		: 0
isa		: rv32imafd
mmu		: sv32

processor	: 1
hart		: 1
isa		: rv32imafd
mmu		: sv32
```

It was also able to decode a MP3 using mpg123 with its default float backend, and write it into a wave file. So all should be good :)

Dolu1990, 2021-03-17:
So, one thing, it was tested on ArtyA7 35T using the following configuration : 
VexRiscvLitexSmpCluster_Cc2_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw128_Cdma_Ood_Fpu.v

The FPGA is quite full (89.07% LUT used), but the timings are ok.

Dolu1990, 2021-03-18:
@enjoy-digital Ready for merge

Also, i improved the SoC pipelining, and now one FPU will be created for each 4 cores, so it should scale. Synthesis with 8 cores pass timings on nexys_video

enjoy-digital, 2021-03-18:
Great work @Dolu1990! I'll review the integration code soon.

rdolbeau, 2021-03-20:
Excellent :-) Seems I've got the FPU running as well, rebuilding the buildroot with a hard-float ABI now.

@Dolu1990 : Is there a way to optionally configure more FPU than one for 4 cores ? (e.g. one-fortwo or one-for-one)

Dolu1990, 2021-03-20:
Currently, the code could do it, but the parameter isn't accessible. Basicaly it is done here : 
https://github.com/SpinalHDL/VexRiscv/blob/80f64f0f9f0e434796f72d583aa1eaaa47d863b9/src/main/scala/vexriscv/demo/smp/VexRiscvSmpLitexCluster.scala#L37

(group cpus by 4)

I will add the option to the litex integration

Dolu1990, 2021-03-23:
@rdolbeau There is the feature to change the ratio between CPU count and FPU : 
https://github.com/enjoy-digital/litex/pull/859

Dolu1990, 2021-03-23:
Did some mandelbrot tests. With F64, got 21 clock per iteration using https://github.com/SpinalHDL/buildroot-spinal-saxon/blob/main/package/mandelbrot/src/mandelbrot.c#L248

realy hard to ensure the ordering in which GCC schedule operations XD

rdolbeau, 2021-03-23:
@Dolu1990 Thanks :-) Now I need to figure out why 'init' crashes with a signal 11 when buildroot is compiled as ilp32d (hard-float)...

Dolu1990, 2021-03-23:
on which board with which config are you trying it ?

> ilp32d

You had to specify ilp32d manualy ?

Dolu1990, 2021-03-23:
There is the images i'm using on artyA7 for a dual corp system : 
[images.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/6192475/images.zip)

Can you give a try ? (maybe the DTB isn't compatible with your board, so keep your own ^^)

rdolbeau, 2021-03-24:
@Dolu1990 Thanks will try that ASAP (as I might have a HW problem, I'm trying the merge of my three-operands branch with dev to get RV32IMAFDCBK...)

Dolu1990, 2021-03-24:
Hoo i would say try without the merge first ^^

rdolbeau, 2021-03-24:
@Dolu1990 I'm an optimist :-) It seems your archive is soft-float (ilp32); the ELF signature have flags 0x0. Mine are 0x5 (I have both EF_RISCV_FLOAT_ABI_DOUBLE from using ilp32d and EF_RISCV_RVC as some test binaries from K used C and I enabled it by default).

With my own soft-float buildroot, I was able to run a trivial test code in soft-float at -O0 (the lone `fadd` in it worked), but the same test at -O3 would crash with a memory error (sig11) somewhere in a library, I _think_ *after* the `fadd` (maybe the printf). I figured using FP instructions in the soft-float ABI (`-march=rv32imafdc -mabi=ilp32`) might have issue, so I rebuilt the whole buildroot & the test binaries with the hard-float ABI (`-march=rv32imafdc -mabi=ilp32d` for my test code, BR2_RISCV_ABI_ILP32D=y along with F/D/C in the buildroot config, CONFIG_FPU=y for the kernel). In that configuration, init itself blows up with a sig11.

I'll have to retry on a pure 'dev' build, with none of my own stuff in it (the merge wasn't very smooth...) to make sure.

rdolbeau, 2021-03-24:
@Dolu1990 I seem to have the same issue when the core is generated from 'dev' (the only change is compressedGen = True):
```
[    7.446527] Run /init as init process                                                  
[    7.460148] init[1]: unhandled signal 11 code 0x1 at 0x95bd628c                        
[    7.461692] CPU: 0 PID: 1 Comm: init Not tainted 5.11.0-rc7 #2                         
[    7.462927] epc: 95bd628c ra : 95b017fe sp : 9d8fa610                                  
[    7.464346]  gp : c057a4a8 tp : 00000000 t0 : 0000000a                                 
[    7.465532]  t1 : 95af09cc t2 : 00000000 s0 : 9d8fac40                                 
[    7.466715]  s1 : 95b09a1c a0 : 9d8fa630 a1 : 00000000                                 
[    7.468260]  a2 : 9d8fa7d8 a3 : 00000024 a4 : 00000000                                 
[    7.469448]  a5 : 9d8fa628 a6 : 7efefeff a7 : 464d4824                                 
[    7.470648]  s2 : 9d8fa760 s3 : 000b2f08 s4 : 95b09a40                                 
[    7.472144]  s5 : 95b09a40 s6 : 00000000 s7 : 00000000                                 
[    7.473329]  s8 : 00000000 s9 : 9d8fa7cc s10: 9d8fa7d8                                 
[    7.474517]  s11: 95b08f30 t3 : 95b00638 t4 : 00000110                                 
[    7.475950]  t5 : 756e6900 t6 : 6ffffe35                                               
[    7.476734] status: 00000020 badaddr: 95bd628c cause: 0000000c                         
[    7.478851] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b    
[    7.480694] CPU: 0 PID: 1 Comm: init Not tainted 5.11.0-rc7 #2                         
[    7.481927] Call Trace:                                                                
[    7.482366] [<c000337e>] walk_stackframe+0x0/0xa6                                      
[    7.483410] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x000
0000b ]---                                                                                
```

Dolu1990, 2021-03-24:
> t seems your archive is soft-float (ilp32)

Hmm that's right, i'm confused, i'm trying to rebuild things now

>  compressedGen = True

XD that's it. I mean, at least it will not work with that.

The reason is that the FPU instruction of RVC isn't implemented yet. Will work on that.

rdolbeau, 2021-03-24:
@Dolu1990 OK good to know, I'll retry without RVC when I can - I will have to remove some stuff from my buildroot as RVC really helps with the size (and it's already over 60 MiB by now, I needed C++ and python for some test codes...)

Dolu1990, 2021-03-24:
Hoo i found something.

CONFIG_FPU=n was in the linux.defconfig

You need to set it to CONFIG_FPU=y too

Dolu1990, 2021-03-24:
[images.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/6195588/images.zip)
that kernel is with the CONFIG_FPU enabled, it has the save/restore fld fsd

Dolu1990, 2021-03-24:
So just tested on ArtyA7, works for me, let's me know how things goes for you

rdolbeau, 2021-03-24:
@Dolu1990 went a slightly different course - it's only 8 expansion patterns to add to the decompressor, I figured it would take less time to add them than recompile the full buildroot :-) I had (at least...) a small bug but init was fixed so it's definitely the issue. Second bitstream being generated, patch to VexRiscv to follow if it works-for-me.

Dolu1990, 2021-03-24:
Hoooo i was adding the feature too XD I just finished now, but was looking into how to verify it. Unfortunatly riscv-test do not has any case for it XD

So your PR is welcome, i will cross check with my implementation, that should give a good level of certainty.

rdolbeau, 2021-03-24:
Oups didn't think you'd get around to it so fast... anyway I can do a PR, but while I go much further than before (init no longer crashes) I have a(n apparent) hang after/around the time haveged starts :

```
[    7.516527] mmcblk0: mmc0:0001 EB1QT 29.8 GiB                                          
[    7.548444]  mmcblk0: p1                                                               
Starting syslogd: OK                                                                      
Starting klogd: OK                                                                        
Running sysctl: OK                                                                        
Saving random seed: [    9.030607] random: dd: uninitialized urandom read (512 bytes read)
OK                                                                                        
Starting haveged: haveged: can not open UNIX socket                                       
haveged: can not initialize command socket: Address family not supported by protocol      
```

Then nothing happens anymore and the console is not responsive :-(

Dolu1990, 2021-03-24:
> Starting haveged: haveged: can not open UNIX socket 

Most of the time i had a hang in that spot, was due to process to actively waiting on /dev/random, which isn't loaded because of no entropy generation.

The entropy generation in that system should be done by haveged, but  here a it can't open the unix socket, things are stuck.
Likely you need to add some package in order to enable it.

Do you have the CONFIG_UNIX=y in the linux.config ?

More broadly on saxonsoc ot have haveged working, i have  : 

CONFIG_NET=y
CONFIG_PACKET=y
CONFIG_PACKET_DIAG=y
CONFIG_UNIX=y
CONFIG_INET=y

rdolbeau, 2021-03-24:
@Dolu1990 Generally speaking that could be the problem, but this configuration works in soft-float and I _think_ that when waiting on /dev/urandom the console would still work as it doesn't rely on random data (but it was a while since I added haveged so I'm not 100% sure).

I have all those but CONFIG_UNIX, I'll add it and try again to make sure.

My current code in https://github.com/SpinalHDL/VexRiscv/pull/167

Dolu1990, 2021-03-24:
Before the look in root, there is no console, i mean, when i had that /dev/random  issue (not urandom), everything freezed.

Do you have the openssh package ?

rdolbeau, 2021-03-24:
no I didn't enable OpenSSH as this board has no ethernet (only micro-sd and serial; the new one crawling its way through the postal system should have ethernet ;-) )

Dolu1990, 2021-03-24:
So
I were able to test rv32imafdc on SaxonSoc, got it to boot and work fine (mandelbrot in double, mpg123 in float)

Just for sanity, did you checked the timings reports of the synthesis ? (as that's quite a lot of additionnal features XD)

Also your buildroot / linux config were working fine before the FPU addition right ?

Dolu1990, 2021-03-24:
Hoooo
#BR2_RISCV_ABI_ILP32=y
BR2_RISCV_ABI_ILP32D=y

That one i never tried.

Dolu1990, 2021-03-24:
(building with BR2_RISCV_ABI_ILP32D now)

rdolbeau, 2021-03-24:
> BR2_RISCV_ABI_ILP32D=y
> That one i never tried.

Yes, and this one (using the hard-float ABI) is quite important for code doing a lot of FP - passing FP values in registers rather than on the stack.

Timing was a bit borderline, re-synthesizing at lower frequency to make sure; except for the FP stuff everything was working before - and was booting on the FP hardware with the soft-float ABI (the software bug I had in this case was likely the missing C stuff).

Dolu1990, 2021-03-24:
One thing to notes, is that in addition of the BR2_RISCV_ISA_CUSTOM_RVC from the buildroot config,CONFIG_RISCV_ISA_C=y is required in the linux.config, else the kernel isn't compiled with RVC

Dolu1990, 2021-03-24:
I just tested on Saxon, got the ILP32D to work fine.

As far as i can see, the app seems to be ilp32d, as from speakertest, the sine wave generation use the C sin function, is used below as : 

```
000126e8 <generate_sine>:
   126e8:	1101                	addi	sp,sp,-32
   126ea:	a422                	fsd	fs0,8(sp)
   126ec:	2100                	fld	fs0,0(a0)
   126ee:	9301b507          	fld	fa0,-1744(gp) # 16130 <__SDATA_BEGIN__>
   126f2:	028477d3          	fadd.d	fa5,fs0,fs0
   126f6:	a026                	fsd	fs1,0(sp)
   126f8:	2504                	fld	fs1,8(a0)
   126fa:	cc22                	sw	s0,24(sp)
   126fc:	ce06                	sw	ra,28(sp)
   126fe:	842a                	mv	s0,a0
   12700:	12a7f7d3          	fmul.d	fa5,fa5,fa0
   12704:	1a97f7d3          	fdiv.d	fa5,fa5,fs1
   12708:	0aa7f553          	fsub.d	fa0,fa5,fa0
   1270c:	f55fe0ef          	jal	ra,11660 <sin@plt> //HERE !!! fa0 = sin(fa0)
   12710:	97c1a787          	flw	fa5,-1668(gp) # 1617c <generator_scale>
   12714:	40157553          	fcvt.s.d	fa0,fa0
   ```

Dolu1990, 2021-03-24:
hmm

You didn't tried with 
BR2_RISCV_ISA_CUSTOM_RVF=y  
BR2_RISCV_ISA_CUSTOM_RVD=y 
BR2_RISCV_ABI_ILP32D=n

Right ? Just to be sure i understand the case.

Dolu1990, 2021-03-24:
What i do normaly to debug this kind of cases where things hang, i run a simulation, to turn everything into whiteboxes ^^

Is there something to run a sim of your system ?

rdolbeau, 2021-03-24:
1) the kernel was built without and with CONFIG_RISCV_ISA_C (with it wouldn't load for me until the bootloader was configured with C as well, so early test did not have it, it doesn't seem to change the userland behavior)

2) I did not try a FP-enabled/soft-float ABI buildroot (RVF/D=y, ILP32 and not ILP32D)

3) to check the ABI declared in the binary you can use `readelf -h $BINARY` and look at the line Flags:

rdolbeau, 2021-03-24:
Not sure for the simulation, I know Vivado can do some stuff but I'm not enough of an expert. Anyway don't worry, I'll try to figure out what's wrong when I have more time. I can fall back on the soft-float ABI anyway, that one should have been fixex by the C stuff.

rdolbeau, 2021-03-24:
Update; it might not be the ABI for me. Reloaded the soft-float buildroot, simple test binary works, but the 'real' binary (a numerical mini-app) hangs for me. Good news is, that's an easier test to share.

Dolu1990, 2021-03-24:
> (so early test did not have it, it doesn't seem to change the userland behavior)

which behaviour wasn't changed ? crashing behaviour ? or correct behaviour ?

> readelf -h 

Cool thanks, didn't know :)

rdolbeau, 2021-03-24:
Didn't change either behavior :-) init crashes in hard-float w/ or w/o RVC in the kernel, works fine in soft-float w/ or w/o RVC in the kernel; as far as I can tell it only changes whether the kernel uses C instructions (with, it is much smaller).

rdolbeau, 2021-03-24:
New update; partial PIBKAC - I had forgotten (again) to update that loathsome DTB, and that seem to have caused the hang of init, as I reach login and can log now. It seems RVC was the primary 'real' problem to fully boot, but my test code still crashes the SoC:

a) soft-float test code 'hydro' hangs the system on the soft-float buildroot;
b) hard-float test code 'hydro' doesn't-hull-hang on the hard-float buildroot:
```
root@buildroot:~# ./hydro -i input.nml                                                    
+-------------------+                                                                     
|GlobNx=80          |                                                                     
|GlobNy=80          |                                                                     
|nx=80              |                                                                     
|ny=80              |                                                                     
|nxystep=80         |                                                                     
[   62.830654] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:                        
[   62.831893] rcu:     1-...0: (0 ticks this GP) idle=50e/1/0x40000000 softirq=522/522 fq
s=2625                                                                                    
[   62.833793]  (detected by 0, t=5252 jiffies, g=-47, q=4)                               
[   62.834823] Task dump for CPU 1:                                                       
[   62.835386] task:hydro           state:R  running task     stack:    0 pid:   83 ppid: 
   75 flags:0x00000008                                                                    
[   62.837479] Call Trace:                                                                
[   62.837886] [<c00020f0>] ret_from_fork+0x0/0xc                                         
```

The kernel output then repeats every ~63 seconds (with t=XYZ jiffies the only thing that changes beyond the kernel timestamp); I can't stop or background the process or regain control, but the terminal acknowledge my actions (carriage return, displayin ^Z or ^C).

Dolu1990, 2021-03-24:
Is hydro using some of the custom instruction added in the CPU ?

I mean, we are currently debugging a system with many tunned parameters at once.
Would it be possible to try the system with a "vanilla" VexRiscv ? without the additional custom extentions ?

rdolbeau, 2021-03-24:
Normally no, it was compiled with the buildroot compiler:
```
[dolbeau@localhost Src]$ cat make.inc 
R5IMA_TOOLCHAIN=/home/dolbeau2/LITEX/buildroot-rv32/output/host
R5IMA_BUILDROOT=/home/dolbeau2/LITEX/buildroot-rv32/output/target

CC=$(R5IMA_TOOLCHAIN)/bin/riscv32-buildroot-linux-gnu-gcc
#CC=/opt/riscv64bk/bin/riscv64-unknown-elf-gcc
CFLAGS=-O3 -march=rv32imafdc -mabi=ilp32
LDFLAGS=-I$(R5IMA_BUILDROOT)include -L$(R5IMA_BUILDROOT)/lib -lnuma -lm -latomic
```

Does it work for you? (I don't think it's ever been tried on RV32GC, though it is known to work in RV64GC under QEmu). If it works, then the problem might be specific to my hardware build.

I was generating the core from a 'vanilla' dev branch, none of my stuff there to remove a variable.

I also tried the same code with -mno-fdiv (this avoids all fdiv and fsqrt instructions, as experience has taught me they can be troublesome), but as far as I can tell they are not the culprit.

I'll try again with a completely clean slate this week-end to make sure ; disabling my stuff might not have cleaned up everything properly. Hopefully, I'll also have access to a bigger FPGA - and if ethernet works in that one, I could have NFS to access binaries and speed up the testing considerably (although making NFS works from buildroot might be a project in itself...)

rdolbeau, 2021-03-25:
OK, so starting with a clean slate (current GitHub repos with just enough patches to generate for my board):

a) adding FD instructions in the core, buildroot, kernel, hydro, retaining the soft-float ABI (ilp32): hydro works
b) switching buildroot & hydro to the hard-float ABI (ilp32d): hydro works
c) adding C to the core, buildroot, kernel, hydro & bootloader: linux boots but hydro causes the system to hang

So while https://github.com/SpinalHDL/VexRiscv/pull/167 helped by enabling boot (init no longer crashes), there's still a C+FD issue as C itself was not causing issue for my crypto tests (including running large binaries such as python3).

Other than that one issue, fantastic job to have a numerical code like hydro working fine on the SoC :-)

Dolu1990, 2021-03-25:
Cool ^^

> there's still a C+FD issue

So, here, try to pull the vexriscv dev, i did one fix, which fix one bad interraction between wfi and the fpu which could in some very specific condition (i don't think that the issue here) hang the system.

By the way what is that hydro binary doing ?

Dolu1990, 2021-03-25:
https://github.com/HydroBench/Hydro/tree/master/HydroC

Rawrrrrr ^^

rdolbeau, 2021-03-25:
Moving the core to dev (to get the fix) did not help unfortunately.

For Hydro:

> HydroC is not an application, but rather a mini-application (...)It is a simplified version of the astrophysical code RAMSES. It is a 2-dimensional CFD using the Finite Volume Method with a Godunov’s scheme and a Riemann solver at each interface on a regular 2D mesh.

It's a numerical code really designed to run on multiple big HPC-oriented servers with high-speed interconnect, not inside a small SoC, so it's impressive it actually works :-) (though I've not enabled any parallelism yet).

Dolu1990, 2021-03-25:
Ok ^^ 

Hmm, how many CPU core do you have in the SoC, and how many core per FPU ?

Because that may be a important delta with my actual configuration.

rdolbeau, 2021-03-25:
1 core so far. Will try with 2 cores/1 FPU (I don't think I can fit 2 cores/2 FPUs in my 35T).

Dolu1990, 2021-03-25:
ok, I'm trying now to build the whole system on litex

Dolu1990, 2021-03-25:
Localy, i added the possibility to have RVC on linux-on-litex-vexriscv, got it to work :

```
buildroot login: root
                   __   _
                  / /  (_)__  __ ____ __
                 / /__/ / _ \/ // /\ \ /
                /____/_/_//_/\_,_//_\_\
                      / _ \/ _ \
   __   _ __      _  _\___/_//_/         ___  _
  / /  (_) /____ | |/_/__| | / /____ __ / _ \(_)__ _____  __
 / /__/ / __/ -_)>  </___/ |/ / -_) \ // , _/ (_-</ __/ |/ /
/____/_/\__/\__/_/|_|____|___/\__/_\_\/_/|_/_/___/\__/|___/
                  / __/  |/  / _ \
                 _\ \/ /|_/ / ___/
                /___/_/  /_/_/
  32-bit RISC-V Linux running on LiteX / VexRiscv-SMP.

login[76]: root login on 'console'
root@buildroot:~# ./hydro -i 
hydro      input.nml
root@buildroot:~# ./hydro -i input.nml 
+-------------------+
|GlobNx=80          |
|GlobNy=80          |
|nx=80              |
|ny=80              |
|nxystep=80         |
|tend=20000.000     |
|nstepmax=10        |
|noutput=0          |
|dtoutput=0.000     |
+-------------------+
DMalloc: allocating 28224
Centered test case : 42 42
sh: cpupower: not found
Hydro starts in double precision.
Hydro: Main process running on buildroot
Hydro: standard build
Hydro starts main loop.
Page offset 8
IMalloc: allocating 6720
IMalloc: allocating 6720
DMalloc: allocating 26880
DMalloc: allocating 6720
DMalloc: allocating 6720
DMalloc: allocating 80
DMalloc: allocating 80
Hydro: init mem 0.052840s
Hydro computes initial deltat: 1.336306e-03
--> step=   1,  1.33631e-03, 1.33631e-03 0.007 MC/s {4.45 Mflops 4187396 Ops} (0.942s)
--> step=   2,  2.67261e-03, 1.33631e-03 0.007 MC/s {4.55 Mflops 4020988 Ops} (0.884s)
--> step=   3,  5.70914e-03, 3.03653e-03 0.007 MC/s {4.48 Mflops 4187396 Ops} (0.935s)
--> step=   4,  8.74568e-03, 3.03653e-03 0.007 MC/s {4.54 Mflops 4020988 Ops} (0.885s)
--> step=   5,  1.24942e-02, 3.74854e-03 0.007 MC/s {4.47 Mflops 4187396 Ops} (0.936s)
--> step=   6,  1.62428e-02, 3.74854e-03 0.007 MC/s {4.54 Mflops 4020988 Ops} (0.886s)
--> step=   7,  2.06309e-02, 4.38811e-03 0.007 MC/s {4.47 Mflops 4187396 Ops} (0.937s)
--> step=   8,  2.50190e-02, 4.38811e-03 0.007 MC/s {4.53 Mflops 4020988 Ops} (0.887s)
--> step=   9,  2.95700e-02, 4.55100e-03 0.007 MC/s {4.47 Mflops 4187396 Ops} (0.938s)
--> step=  10,  3.41210e-02, 4.55100e-03 0.007 MC/s {4.53 Mflops 4020988 Ops} (0.888s)
Hydro ends in 00:00:09.134s (9.134) <4.50 MFlops>.
Average MC/s: 0.007059 min 0.006825, max 0.007221 sig 0.000189
GATCON    CONPRI    EOS       SLOPE     TRACE     QLEFTR    RIEMAN    CMPFLX    UPDCON    COMPDT    MAKBOU    ALLRED    
PE0_DP 0.4217    0.5333    0.2485    1.2615    0.9557    0.2455    4.4893    0.2837    0.4056    0.2537    0.0142    0.0000    
%      4.6275    5.8521    2.7272    13.8429   10.4878   2.6945    49.2646   3.1128    4.4510    2.7839    0.1556    0.0000    
Average MC/s: 0.007
root@buildroot:~# 

``` 

But i have to try again with one specific config disabled

Dolu1990, 2021-03-25:
I successfully recreated a hang on the hydro binary.
To do that i had to keep IBusCachedPlugin.injectorStage on its default (false). when things were working with rvc it was to true to improve timings.
But, that injectorStage feature impact a very specific spot which may impact how the instructions are forked to the FPU.
Only issue in the diagnostic is that right now without that option i get a -4ns slack, which is way to much.

So i'm trying to setup something simlar in SaxonSoc but with a slower main clock.

Dolu1990, 2021-03-25:
@rdolbeau So right, as a workaround for now, just turn injectorStage to true, it should solve the issue. I'm working now on recreating the hang in simulation.

Dolu1990, 2021-03-25:
Thanks for the catch ^^

rdolbeau, 2021-03-25:
@Dolu1990 Hehe I was synthesizing so soon as I saw the prospective fix :-) So I can confirm yes, injectorStage=true solves the problem; hydro is working with RV32IMAFDC/ILP32D now! It also seem to improve WNS, but performance is a bit lower (the MFlops in the output is a rough estimate but is useful to compare configurations with the same binary).

as a mini-app, hydro is really made to test and evaluate systems, so it did its job once more :-)

Next step, back to my own merged branch and testing OpenMP parallelism.

Dolu1990, 2021-03-25:
Cool ^^

So, about the fpu, got 42 MFLOPS single threaded with mandelbrot test 100 Mhz. This could probably be pushed higher if that was properly scheduled in assembly.

rdolbeau, 2021-03-25:
@Dolu1990 I tried my merged version with injectorStage=true, hydro works in that configuration as well, alongside @mjosaarinen test codes from https://github.com/rvkrypto/rvkrypto-fips .

So I now have a RV32IMAFDCBK core booting Linux and running codes :-) Perhaps the first one as FPUs aren't common in RV32 FPGA cores and B/K are still under review...

Won't be able to fit 2 cores in the 35T, I'm already at > 83% of LUTs:

```
+----------------------------+-------+-------+-----------+-------+
|          Site Type         |  Used | Fixed | Available | Util% |
+----------------------------+-------+-------+-----------+-------+
| Slice LUTs                 | 17319 |     0 |     20800 | 83.26 |
|   LUT as Logic             | 16507 |     0 |     20800 | 79.36 |
|   LUT as Memory            |   812 |     0 |      9600 |  8.46 |
|     LUT as Distributed RAM |   776 |     0 |           |       |
|     LUT as Shift Register  |    36 |     0 |           |       |
| Slice Registers            | 11385 |     0 |     41600 | 27.37 |
|   Register as Flip Flop    | 11385 |     0 |     41600 | 27.37 |
|   Register as Latch        |     0 |     0 |     41600 |  0.00 |
| F7 Muxes                   |   307 |     0 |     16300 |  1.88 |
| F8 Muxes                   |    62 |     0 |      8150 |  0.76 |
+----------------------------+-------+-------+-----------+-------+
```

Dolu1990, 2021-03-25:
@rdolbeau https://github.com/SpinalHDL/VexRiscv/commit/6f481f51ef2bc9ec15866682087eb982ce865ae9  fix the issue properly. Got RVC without injectorStage to work on Saxon (lower frequancy to meet timings)

So i would say, keep the injectorStage to ensure timing pass :)

Dolu1990, 2021-03-25:
> So I now have a RV32IMAFDCBK core booting Linux and running codes :-) Perhaps the first one as FPUs aren't common in RV32 FPGA cores and B/K are still under review...

> I tried my merged version with injectorStage=true, hydro works in that configuration as well, alongside @mjosaarinen test codes from https://github.com/rvkrypto/rvkrypto-fips .
 
 Greate :D
 
 Hmm one question that i have is how to verify that the output of the simulation is right ?
 Because even on X86, seems like there is some seed in it, which made the result different each run XD

[6 more replies]

closed 2021-04-01 by Dolu1990