litex-hub/linux-on-litex-vexriscv #219: Question : LFE5U 25F  or  LFE5U 45F and DDR3 RAM and RGMII support

akshar001, 2021-03-24:
I am totally new to this world and found this project most amazing thing,

I have one question.

If we make custom boards with DDR3 RAM and 512MB or 1GB RAM and also wants to support RMGII interface can we do that with this both FPGA devices?

enjoy-digital, 2021-03-26:
Hi @akshar001,

a 25F will probably be almost full with the CPU + DDR3 + 1Gbps Ethernet. I would recommend using a 45F if you want to be able to have others peripherals around the base SoC.

akshar001, 2021-03-27:
I have one idea. @enjoy-digital i want to make one open source hardware very cheap like raspberry pi one and support all of these functionalities.

Your rics-v core fpga and the hardware. Imaging a 50-60$ board which will do all of this thing!

akshar001, 2021-03-27:
It's just the pin problem right otherwise we can run it easily!

enjoy-digital, 2021-03-28:
The issue is more related to the resource usage than the IOs. The 25F is already using almost all the resources with CPU + DDR3 + 1Gbps so with a 25F the design will not be able to evolve a lot.

akshar001, 2021-03-29:
Can we also have wifi and hdmi and usb peripherals alongs with this with 45f or then we need to go for 85f

akshar001, 2021-03-29:
I want to build FPGA SBC same kind of raspberry pi with all of this system but under 80$!

enjoy-digital, 2021-03-29:
A 45F will be fine to add Wifi, HDMI and USB  in single core. If you want to support mult-icore (2, 4) I would recommend a 85F.

akshar001, 2021-03-29:
Thanks, i would go single core first.
One last question what max frequency we can get here?

enjoy-digital, 2021-03-29:
Probably something between 60-75MHz.

akshar001, 2021-03-29:
Can we use external vco or pll's and put up a high frequencies for system clocks like up to 600-700mhz

enjoy-digital, 2021-03-29:
No, the limitation is related to the FPGA fabric, the CPU will not run at more than ~75MHz on a ECP5.

akshar001, 2021-03-30:
Okay. Can we support 8 core SMP using 85F ?

enjoy-digital, 2021-03-30:
4 CPU cores is probably max with a 85F. Here is device utilization report with 4 cores on the ECPIX5 built with `./make.py --board=ecpix5 --cpu-count=4 --build`. You can probably pack more CPU cores, but you'll have to lower the frequency, so the advantage will be limited. 8 and more cores is more a demonstration of the scalability than something really useful for now.
```
Info: Device utilisation:
Info: 	       TRELLIS_SLICE: 29142/41820    69%
Info: 	          TRELLIS_IO:    90/  365    24%
Info: 	                DCCA:     4/   56     7%
Info: 	              DP16KD:    61/  208    29%
Info: 	          MULT18X18D:    16/  156    10%
Info: 	              ALU54B:     0/   78     0%
Info: 	             EHXPLLL:     1/    4    25%
Info: 	             EXTREFB:     0/    2     0%
Info: 	                DCUA:     0/    2     0%
Info: 	           PCSCLKDIV:     0/    2     0%
Info: 	             IOLOGIC:    53/  224    23%
Info: 	            SIOLOGIC:     6/  141     4%
Info: 	                 GSR:     0/    1     0%
Info: 	               JTAGG:     0/    1     0%
Info: 	                OSCG:     0/    1     0%
Info: 	               SEDGA:     0/    1     0%
Info: 	                 DTR:     0/    1     0%
Info: 	             USRMCLK:     1/    1   100%
Info: 	             CLKDIVF:     1/    4    25%
Info: 	           ECLKSYNCB:     1/   10    10%
Info: 	             DLLDELD:     0/    8     0%
Info: 	              DDRDLL:     1/    4    25%
Info: 	             DQSBUFM:     2/   14    14%
Info: 	     TRELLIS_ECLKBUF:     1/    8    12%
Info: 	        ECLKBRIDGECS:     0/    2     0%
```

enjoy-digital, 2021-03-30:
I think we can now close this, but feel free to play with the tools to test various configurations/implementations  before selecting the ECP5 component for your board.

closed 2021-03-30 by enjoy-digital

akshar001, 2021-03-30:
Okay.

enjoy-digital, 2021-03-30:
Sorry I try to avoid having too much issues opened to be able to keep track of things, but happy to discuss more or answer specific questions for your project.

akshar001, 2021-03-30:
You are very helpful. I want to create one super cheap board which an be bought with students hobbyists and some AI stuff. kind of raspberry pi. Do you think idea is viable? Cause you know more fpga then me.

enjoy-digital, 2021-03-30:
This is interesting yes, but you have to be aware that it will not compete with the the SoC we use nowadays in a RPI or smartphone for generic applications. But this is clearly becomes interesting for students to learn about SoC/Linux/RISC-V and create a full open-source SoC (with also open-source toolchain in the case of the ECP5) or to create  specific applications with custom peripherals and associated kernel drivers where using the specialized resources of the FPGA can allow things that are simply not possible with off-the-shelf or provides better performance than non specialized SoCs.

akshar001, 2021-03-30:
That i know now, my idea is totally changed. But at-least providing specific hardware interfaces could help. like in some system integrating if we can introduce two soft cores with risc-v with atleast running standard protocols like mosquitto or that kind of thing for internet, and high end ADC channel or PWM or that kind of thing can be integrated.

akshar001, 2021-03-30:
We can give 2 core cpu and rest of the custom blocks that can be programmed easily, the hardware board that we build should be easily modified with external modules like, is someone wants to build 16 UART interfacing device they can do it easily with our boards. And can use linux as well. Low end ofcourse but it would be sufficient for them.