litex-hub/linux-on-litex-vexriscv #220: Settings to speed up simulation

developandplay, 2021-03-26:
Are there any flags/settings that can be utilized to speed up the simulation?
The simulation does work for me but it takes about 30m to get to the intial buildroot login.

enjoy-digital, 2021-03-28:
Hi @developandplay,

Linux generally boots in 15 minute on the machine using for the tests (an I7 from 2014). Which machine are you testing with? Verilator is already configured to compile the code in O3 and is a SoC with medium complexity (with the DRAM core also simulated), so it's not really possible to speed the simulation.

developandplay, 2021-03-28:
Thank you!
I'm using a pretty powerful ryzen 7 mobile cpu. However I guess that doesn't help if it's all about high single core clock speed.
Boot times are maybe 30m and then entering a character on the console takes 30s.
I was just surprised by the performance difference compared to a simple temu riscv simulation.

enjoy-digital, 2021-03-29:
The simulation will indeed be a lot slower than software emulation (since here the whole logic of the SoC is simulated). It should still be faster than 30min, but this time is probably related to the configuration of your machine.

closed 2021-03-29 by enjoy-digital

developandplay, 2021-06-07:
When using the normal litex-sim command for other cores the `--threads` option was exactly what I was looking for.
I now created a quick PR for it.