litex-hub/linux-on-litex-vexriscv #225: Support for Debian or Fedora RISC-V Linux

cjearls, 2021-04-14:
Is there any support planned for Debian or Fedora RISC-V ports instead of Buildroot Linux? What would need to be accomplished to allow these to run properly?

mithro, 2021-04-14:
Debian or Fedora RISC-V would need to support 32bit RISC-V platform.

cjearls, 2021-04-14:
I see. Would it work on the Rocket configuration, or does linux-on-litex only use 32-bit Rocket?

mithro, 2021-04-14:
I believe the repo @ https://github.com/litex-hub/linux-on-litex-rocket is 64bit and should support "off the shelf" Linux if I understand.

enjoy-digital, 2021-04-15:
@cjearls: That's in fact one of the aims of @gsomlo with Linux-on-LiteX-Rocket and think it's already possible:
https://open-src-soc.org/2021-03/program-riscv-meetings.html#T-SOMLO
https://open-src-soc.org/2021-03/media/slides/3rd-RISC-V-Meeting-2021-03-31-13h30-Gabriel-Somlo.pdf

gsomlo, 2021-04-15:
On Wed, Apr 14, 2021 at 11:15:09PM -0700, enjoy-digital wrote:
> @cjearls: That's in fact one of the aims of @gsomlo with Linux-on-LiteX-Rocket

Indeed, I picked 64-bit RocketChip specifically to be able to run an
existing distro (Fedora is the one I care about most) -- which tend
to only target rv64gc.

Last time I tried (as of a couple of weeks ago), systemd hangs or
times out while trying to boot the Fedora root filesystem (from initrd).
I used to blame that on bugs with the sdcard gateware, but that
situation has improved significantly over just the last few days
(thanks again @enjoy-digital)!

ATM, I can reliably chroot to the Fedora root partition (ripped from a
qemu rv64gc VM) from Busybox, and run programs from it (e.g. yosys,
nextpnr-ecp5, etc.).

Still working on getting systemd to boot the whole thing natively.

closed 2021-04-22 by cjearls

Dolu1990, 2021-04-22:
@cjearls 

There is a few interresting video about it ^^
https://www.youtube.com/watch?v=EjJZb-Ayeys&ab_channel=FOSDEM

Basicaly, the main issue as far as i understand it is to handle packages which do not support cross-compilation, and to handle package which aren't implemented in a portable way using patches.