litex-hub/linux-on-litex-vexriscv #234: Memory Initialization Failed on DE10-Nano

Albert-Hu, 2021-06-01:
Hi, I was trying to run the Linux images on DE10-Nano, but it stops at the message *Liftoff!* after the Linux images are uploaded.

>         __   _ __      _  __
>        / /  (_) /____ | |/_/
>       / /__/ / __/ -_)>  <
>      /____/_/\__/\__/_/|_|
>    Build your hardware, easily!
> 
>  (c) Copyright 2012-2021 Enjoy-Digital
>  (c) Copyright 2007-2015 M-Labs
> 
>  BIOS built on Jun  1 2021 15:55:47
>  BIOS CRC passed (5078eda3)
> 
>  Migen git sha1: 3ffd64c
>  LiteX git sha1: 26db1070
> 
> --=============== SoC ==================--
> CPU:		VexRiscv SMP-LINUX @ 50MHz
> BUS:		WISHBONE 32-bit @ 4GiB
> CSR:		32-bit data
> ROM:		64KiB
> SRAM:		8KiB
> L2:		2KiB
> SDRAM:		65536KiB 16-bit @ 50MT/s (CL-2 CWL-2)
> 
> --========== Initialization ============--
> Initializing SDRAM @0x40000000...
> Switching SDRAM to software control.
> Switching SDRAM to hardware control.
> Memtest at 0x40000000 (2.0MiB)...
>   Write: 0x40000000-0x40200000 2.0MiB     
>    Read: 0x40000000-0x40200000 2.0MiB     
>   bus errors:  0/256
>   addr errors: 8191/8192
>   data errors: 524288/524288
> Memtest KO
> Memory initialization failed
> 
> --============= Console ================--
> 
> litex> serialboot
> 
> Booting from serial...
> Press Q or ESC to abort boot completely.
> sL5DdSMmkekro
> [LXTERM] Received firmware download request from the device.
> [LXTERM] Uploading images/Image to 0x40000000 (7279384 bytes)...
> [LXTERM] Upload complete (81.7KB/s).
> [LXTERM] Uploading images/rv32.dtb to 0x40ef0000 (2168 bytes)...
> [LXTERM] Upload complete (80.1KB/s).
> [LXTERM] Uploading images/rootfs.cpio to 0x41000000 (4002304 bytes)...
> [LXTERM] Upload complete (81.7KB/s).
> [LXTERM] Uploading images/opensbi.bin to 0x40f00000 (53640 bytes)...
> [LXTERM] Upload complete (81.6KB/s).
> [LXTERM] Booting the device.
> [LXTERM] Done.
> Executing booted program at 0x40f00000
> 
> --============= Liftoff! ===============--

I found the SDRAM is 64MB rather than 32MB, and the memory initialization is failed.

enjoy-digital, 2021-06-29:
Hi @Albert-Hu,

sorry for the delay. On the DE10-Nano, the SDRAM is provided as a MiSTer expansion board and there are probably different variants. The one that is currently supported is the AS4C32M16 based one (64MB ) (https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/terasic_de10nano.py#L26) If yours is 32MB, the SDRAM module should be updated in the LiteX target. Can you check this which Variant you have and eventually try to modify the SDRAM module?

Thanks.

Albert-Hu, 2021-07-01:
Hi @enjoy-digital,

Thanks for your reply. I don't have the MiSTer SDRAM expansion board.

And I tried to build it without MiSTer SDRAM expansion board but failed.

I set the variable "with_mister_sdram" to False in make.py at line [446](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L446).

Then I got the error message:

> INFO:SoC:        __   _ __      _  __  
> INFO:SoC:       / /  (_) /____ | |/_/  
> INFO:SoC:      / /__/ / __/ -_)>  <    
> INFO:SoC:     /____/_/\__/\__/_/|_|  
> INFO:SoC:  Build your hardware, easily!
> INFO:SoC:--------------------------------------------------------------------------------
> INFO:SoC:Creating SoC... (2021-07-01 22:14:46)
> INFO:SoC:--------------------------------------------------------------------------------
> INFO:SoC:FPGA device : 5CSEBA6U23I7.
> INFO:SoC:System clock: 50.00MHz.
> INFO:SoCBusHandler:Creating Bus Handler...
> INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
> INFO:SoCBusHandler:Adding reserved Bus Regions...
> INFO:SoCBusHandler:Bus Handler created.
> INFO:SoCCSRHandler:Creating CSR Handler...
> INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
> INFO:SoCCSRHandler:Adding reserved CSRs...
> INFO:SoCCSRHandler:ctrl CSR added at Location 0.
> INFO:SoCCSRHandler:uart CSR added at Location 2.
> INFO:SoCCSRHandler:timer0 CSR added at Location 3.
> INFO:SoCCSRHandler:CSR Handler created.
> INFO:SoCIRQHandler:Creating IRQ Handler...
> INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
> INFO:SoCIRQHandler:Adding reserved IRQs...
> INFO:SoCIRQHandler:IRQ Handler created.
> INFO:SoC:--------------------------------------------------------------------------------
> INFO:SoC:Initial SoC:
> INFO:SoC:--------------------------------------------------------------------------------
> INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
> INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
> CSR Locations: (3)
> - ctrl   : 0
> - uart   : 2
> - timer0 : 3
> INFO:SoC:IRQ Handler (up to 32 Locations).
> INFO:SoC:--------------------------------------------------------------------------------
> INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
> INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
> INFO:SoCBusHandler:plic Region added at Origin: 0xf0c00000, Size: 0x00400000, Mode: RW, Cached: False Linker: False.
> INFO:SoCBusHandler:plic added as Bus Slave.
> INFO:SoCBusHandler:clint Region added at Origin: 0xf0010000, Size: 0x00010000, Mode: RW, Cached: False Linker: False.
> INFO:SoCBusHandler:clint added as Bus Slave.
> INFO:SoCIRQHandler:uart IRQ added at Location 0.
> INFO:SoCIRQHandler:timer0 IRQ added at Location 1.
> INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00010000, Mode: R, Cached: True Linker: False.
> INFO:SoCBusHandler:rom added as Bus Slave.
> INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00010000, Mode: R, Cached: True Linker: False.
> INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
> INFO:SoCBusHandler:sram added as Bus Slave.
> INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
> INFO:SoCIRQHandler:uart IRQ added at Location 0.
> INFO:SoCIRQHandler:timer0 IRQ added at Location 1.
> INFO:CycloneVPLL:Creating CycloneVPLL, speedgrade -I7.
> INFO:CycloneVPLL:Registering Single Ended ClkIn of 50.00MHz.
> INFO:CycloneVPLL:Creating ClkOut0 sys of 50.00MHz (+-10000.00ppm).
> INFO:CycloneVPLL:Creating ClkOut1 sys_ps of 50.00MHz (+-10000.00ppm).
> INFO:CycloneVPLL:Creating ClkOut2 vga of 40.00MHz (+-10000.00ppm).
> INFO:SoCBusHandler:opensbi Region added at Origin: 0x40f00000, Size: 0x00080000, Mode: RW, Cached: True Linker: True.
> INFO:SoCBusHandler:sdblock2mem added as Bus Master.
> INFO:SoCBusHandler:sdmem2block added as Bus Master.
> INFO:SoCIRQHandler:sdirq IRQ allocated at Location 2.
> INFO:SoCCSRHandler:switches CSR allocated at Location 1.
> Traceback (most recent call last):
>   File "./make.py", line 662, in <module>
>     main()
>   File "./make.py", line 640, in main
>     builder.build(run=args.build, build_name=board_name)
>   File "/home/albert/litex/litex/litex/soc/integration/builder.py", line 249, in build
>     self.soc.finalize()
>   File "/home/albert/litex/migen/migen/fhdl/module.py", line 156, in finalize
>     subfragments = self._collect_submodules()
>   File "/home/albert/litex/migen/migen/fhdl/module.py", line 149, in _collect_submodules
>     r.append((name, submodule.get_fragment()))
>   File "/home/albert/litex/migen/migen/fhdl/module.py", line 102, in get_fragment
>     self.finalize()
>   File "/home/albert/litex/migen/migen/fhdl/module.py", line 157, in finalize
>     self.do_finalize(*args, **kwargs)
>   File "/home/albert/litex/litex/litex/soc/cores/cpu/vexriscv_smp/core.py", line 432, in do_finalize
>     self.specials += Instance(self.cluster_name, **self.cpu_params)
>   File "/home/albert/litex/migen/migen/fhdl/module.py", line 136, in __getattr__
>     raise AttributeError("'"+self.__class__.__name__+"' object has no attribute '"+name+"'")
> AttributeError: 'VexRiscvSMP' object has no attribute 'cluster_name'

Thanks.

enjoy-digital, 2021-07-01:
Hi @Albert-Hu,

without the  MiSTer SDRAM, you will be able to build a LiteX SoC, but won't be able to run Linux on it (seems you need at least 32MB of SDRAM for this). Is this what you are trying to do?

Albert-Hu, 2021-07-03:
Hi @enjoy-digital,

I'm just studying about the RISC-V and learning the boot flow.

So the first thing that I think is following this project instruction to build up a RISC-V SoC on FPGA and try something I want to do.

I don't have much experience with FPGA and hardware.

Could you give me suggestions on how to do that?

Thanks a lot.

Albert.

benoit1995, 2021-08-13:
Hi @Albert-Hu 
By reading the de10nano manual you will find that the SOC consists of an FPGA and an ARM (Cortex A9), where the Uart/Eth/Flash/SDCard and SDRAM on the development board are hardwired to the ARM part, so the FPGA cannot access them directly. 
So you need to purchase these modules separately. The litex project supports the MiSTer expansion board and SDRAM by default (**/litex-board/platform/terasic_de10nano.py**), so you can buy the MiSTer expansion board (with UART/SDCard/VGA) and the MiSTer SDRAM (128M).
The vexRiscv will point the PC to the ROM address (ROM is implemented by the on-chip SRAM of the FPGA) after booting, and the Bootloader source code will be stored in this ROM (**litex/litex/soc/integration/soc_core.py +200**). 
You can get more information about the bootloader by reading **litex/soc/software/bios/main.c**. The bootloader will download the Linux kernel to SDRAM, so if you do not use an external SDRAM you will not be able to download the Linux kernel.

bon courage!

Albert-Hu, 2021-08-14:
Hi @benoit1995

Thank you so much for your information.

Albert~

enjoy-digital, 2022-03-18:
The question has been answered, we can probably close this.

closed 2022-03-18 by enjoy-digital