litex-hub/linux-on-litex-vexriscv #237: Booting with Custom root file system on nexys4DDR

chriswils95, 2021-06-28:
I am trying to boot linux from an sd card with a custom root file system on my nexys4ddr board. 
Steps taken was:
 * create a custom root file system that contains startup scripts(.tar or .cpio file)
* edit the boot.json file to use my root file system instead of the rootfs.cpio file with the same address
Anytime i boot up it get stuck. I attached the uart output.
[image]

enjoy-digital, 2021-06-29:
Hi @chriswils95,

we'll need more information to be able to help. Please make sure to read https://github.com/enjoy-digital/litex/wiki/Feedback-Contribution-Support to have a better understanding of the minimal information we need to simplify support.

From what I understand, you are adapting a blog/tutorial written for another board to the nexys4ddr, so we to provide help we need:

    The command line that has been used to generate the Nexys4DDR design.
    A link to the blog/tutorial.
    The way you are booting the kernel images, is it also from SDCard?
    The way you built the kernel images.

In addition to this, if you want to share the bitstream + linux images, it can help us reproduce the issue on a local Nexys4DDR.

Thanks.

Dolu1990, 2021-06-29:
Seems like the kernel is waiting on the rootfs to be mounted at /dev/ram0.

Can you send all the console logs ?

chriswils95, 2021-06-29:
The command line that has been used to generate the Nexys4DDR design.
   *    ./make.py --board=Nexys4DDR --cpu-count=4 --build

The way you are booting the kernel images, is it also from SDCard?
   * Yes booting everything from SD Card.
The way you built the kernel images.
   *  I used the prebuilt kernel images
   
I also attached files in the SD Card,

chriswils95, 2021-06-29:
I figure it out, all i had to do with edit the rootfs_overlays in buildroot, regenerate the binaries and use the files in output/images, and it worked like charm. you guys are awesome.

closed 2021-06-29 by chriswils95

enjoy-digital, 2021-07-02:
Great you got it working, thanks for the feedback.