litex-hub/linux-on-litex-vexriscv #243: prebuit bitstreams: include dtb files into distribution

frantony, 2021-08-10:
Hi @enjoy-digital !

I'm trying to use barebox for booting Linux.
I use prebuilt [arty_2020_12_30.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5753339/arty_2020_12_30.zip) bitstream for Arty from https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164.

I need dtb file for booting linux and barebox. There is neither dtb-file nor even ```csr.json``` file attached to the issue https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164.

The issue https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164 recommends dtb regeneration, but at the moment dtb regeneration for Arty does not work (e.g. 'switches_interrupt' KeyError, see https://github.com/litex-hub/linux-on-litex-vexriscv/issues/239).

I used quick-and-dirty hack for ```litex_json2dts.py``` to suppress the 'switches_interrupt' KeyError, the generated file ```linux-on-litex-vexriscv/build/arty/arty.dts``` contains

```
  mac0: mac@f0006800 {
        compatible = "litex,liteeth";
        reg = <0xf0006800 0x7c>,
              <0xf0007000 0x0a>,
              <0xb0000000 0x2000>;
        ...
  };
```

these liteetch base addresses are inapplicable for [arty_2020_12_30.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5753339/arty_2020_12_30.zip) bitstream. More appropriate addresses are:

```
  mac0: mac@f0004800 {
          compatible = "litex,liteeth";
          reg = <0xf0004800 0x7c /* base */
                 0xf0004000 0x0a /* mdio_base */
                 0xb0000000 0x2000>; /* buf_base */
        ...
  };
```

It is not obvious how to regenerate correct dtb for the prebuilt bitstream.

Could you please regenerate prebuit bitstreams, linux stuff and include dtb files into distribution?

frantony, 2021-08-10:
> I used quick-and-dirty hack for litex_json2dts.py to suppress the 'switches_interrupt' KeyError

Thanks to @MateuszKosmala I can drop quick-and-dirty hack and just use https://github.com/litex-hub/linux-on-litex-vexriscv/pull/242

frantony, 2021-08-13:
@enjoy-digital 

I have just published my version of pre-built images for booting linux with barebox on Arty:
https://github.com/frantony/arty-linux-on-litex-vexriscv-prebuilt

* git lfs is used;
* dtb files are included.

zoro269, 2022-06-09:
@frantony 
Dear sir
I took a look about your prebuilt image version. Can you show me the steps to build it? Especially the file fw_jump.bin in the folder opensbi.
https://github.com/frantony/arty-linux-on-litex-vexriscv-prebuilt/tree/main/opensbi
Thank you