litex-hub/linux-on-litex-vexriscv #244: There is no ethphy in kcu105, only ethmac.

benoit1995, 2021-08-12:
I successfully downloaded the kernel (https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164) to kcu105 using tftp, but I don't see eth0 after using ifconfig -a, only Io and sit0 exist.
[image]
I checked . /build/kcu105/kcu105.dts, found no mac0 node, then in litex/tools/litex_json2dts.py I found that if I want to generate a mac0 node, I need both ethphy and ethmac.
`if "ethphy" in d["csr_bases"] or "ethmac" in d["csr_bases"]:`
I found that when enabling the ethernet of kcu105 in make.py, it only generates ethmac but not ethphy.
[image]
[image]
Why does this situation arise? Is it because kcu105 is using sfp instead of rj45?
Looking forward to your reply.

Benoit.

enjoy-digital, 2021-09-21:
Thanks @benoit1995 for the feedback. The 1000 BASE X PHY used on the KCU105 is indeed a bit different and currently does not have CSRs. I'll look at this soon.

enjoy-digital, 2022-03-23:
Sorry for the delay, this is fixed by https://github.com/enjoy-digital/liteeth/commit/4cd0a99187677553078fcefbf78eb15863b6ed61.

closed 2022-03-23 by enjoy-digital