litex-hub/linux-on-litex-vexriscv #249: Notes on loading vexriscv linux to acorn_cle_215 over pcie

sajattack, 2021-09-08:
As requested:
- flashing the bitstream works with litepcie_util flash_write and litepcie_util flash_reload
- make.py:
 ```
class AcornCLE215(Board):
    soc_kwargs = {"with_pcie": True, "uart_name":"crossover"}
    def __init__(self):
        from litex_boards.targets import acorn
        Board.__init__(self, acorn.BaseSoC, soc_capabilities={
            # Communication
            "serial",
            # Storage
            #"sata",
        }, bitstream_ext=".bit")
```

- loading linux over crossover uart works with litex_server and lxterm
  - litex_server --pcie --pcie-bar 01:00.0
  - litex_term --image=../../images/boot.json --serial-boot bridge
  - comm_pcie.py must be modified to subtract the base address of CSRs in litex_server
  - it uploads at 8-10KB/s (max I've managed modifying lxterm to have no delay, no sfl_outstanding and 255 packet size.
 
There's probably a few other mods I had to make that I'll update this with later, just wanted to get something down for you.

AEW2015, 2021-09-23:
Here is some python software I wrote for my acorn builds (non-litex).  Some sample code to read the XADC temp and to use the ICAP WBStar command to boot from a different address in flash.
https://github.com/byuccl/fiate/tree/main/host_sw

enjoy-digital, 2021-11-19:
Thanks @sajattack, @AEW2015,

when building the acorn design with:
`./make.py --board=acorn_pcie --build`

The UART can now be directly used with the LitePCIe driver:
```
cd kernel
sudo ./init.sh
insmod liteuart.ko
litex_term /dev/ttyLXU0
```

The upload speed can be up to 600KB/s when enabling 256-bytes payload transfer in litex_term (Not yet integrated but tested in https://github.com/enjoy-digital/litex/issues/1105).

closed 2021-11-19 by enjoy-digital