litex-hub/linux-on-litex-vexriscv #253: ECP5 -- L2 cache slows things down signifcantly

essele, 2021-11-26:
I'm just trying a few config experiments on an icesugar_pro (will try some others soon) and I've noticed some unexpected behavior with the l2_size parameter.

If I use the standard default settings (l2_size = 2048) I get a memory speed of 19MiB/s write and 14.1MiB/s read, and a boot time to the last timed kernel message of 17.57s.

if I set l2_size to 0, I would have thought we should get worse performance, but I get a memory speed of 19MiB/s write and 18.7MiB/s read, and a stunning boot time of 6.54 seconds, and it generally feels much snappier.

Is that expected, or am I missing something?

enjoy-digital, 2022-01-04:
Hi @essele,

with `l2_size` set to `0`, the SoC will use the direct memory path to LiteDRAM instead of going through the main bus of the SoC/L2 Cache. This is indeed faster but using the direct memory path requires DRAM's DM to be handled correctly on the DRAM and we saw some issues with this on some boards (the same bitstream was working on some boards, failing on others) This still has to be properly investigated and that's the reason the direct memory path is not enabled by default. If this is working on your board, it's indeed more interesting to use it.

enjoy-digital, 2022-03-18:
EPC5/DDR3 targets now default to LiteDRAM native memory interface with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/81014de6738c903cdf6710a81d6e239649543990. For boards where `mem_test `is failing  with LiteDRAM native memory (related to direct DM use), use of Wishbone Memory/L2 Cache can be enabled by adding `--with-wishbone-memory`.

closed 2022-03-18 by enjoy-digital