litex-hub/linux-on-litex-vexriscv #254: Linux fails to boot on Orangecrab (85F) with cpu-count=2

essele, 2021-11-27:
I've got a new 85F Orangecrab and have got it booting nicely with a single core, but when I try cpu-count=2 it doesn't get past the "==== Liftoff! ====" message, which I presume is the opensbi bit.

I have already set l2_size to 0 as per some of the discussions in other logged issues, and that got the bios running ok.

Any ideas, or anywhere I can look for problems?

Dolu1990, 2021-11-27:
Hi,

Are the synthesis timings ok ?

essele, 2021-11-28:
Yes, I think so ....

`Info: Max frequency for clock                   '$glbnet$sys_clk': 68.96 MHz (PASS at 64.00 MHz)`
`Info: Max frequency for clock '$glbnet$main_crg_ecp5pll1_clkout1': 95.79 MHz (PASS at 12.00 MHz)`
`Info: Max frequency for clock       '$glbnet$clk48$TRELLIS_IO_IN': 309.02 MHz (PASS at 48.00 MHz)`
`Info: Max frequency for clock '$glbnet$main_crg_ecp5pll1_clkout0': 230.57 MHz (PASS at 48.00 MHz)`
`Info: Max frequency for clock                  '$glbnet$init_clk': 281.37 MHz (PASS at 24.00 MHz)`

Dolu1990, 2021-11-28:
Hmmm, and is it booting with the same arguements but in the simulation ?

What exact command line + arguements are you using ?

essele, 2021-11-28:
So the sim.py and make.py do wildly different things in terms of what hardware is included, so I'm not sure how that's a fair comparison, am I missing some other way of simulating?

My make.py command line:  `./make.py --board orangecrab --device 85F --cpu-count 2 --build`
(and I have edited make.py to alter l2_size to 0)

If I run `./sim.py --cpu-count 2` it works fine.

essele, 2021-11-28:
Further info ... interestingly I get the same behaviour single cpu if l2_size is 0. So with l2_size=0 neither single or dual core will boot .. both stop at "==== Liftoff ====".

With l2_size=2k then the single core one is fine, but the dual-core one doesn't even load the bios (no output at all.)

Dolu1990, 2021-12-01:
that's weird, because the CPU netlists were not updated since a while (+6 months) : 
https://github.com/litex-hub/pythondata-cpu-vexriscv_smp/tree/master/pythondata_cpu_vexriscv_smp/verilog

So it maybe something broken more recently in the SoC itself ?

enjoy-digital, 2022-03-18:
The OrangeCrab has DM pins swapped and with VexRiscv-SMP/Native LiteDRAM interfaces, this need to be fixed in the gateware (since VexRiscv-SMP do partial writes on the DRAM which is not the case when using the L2). I just reverted this in: https://github.com/litex-hub/litex-boards/commit/eb8657f515fe63ced13c487f7da537451d89d993

@essele; Could you do a test with up to date LiteX repos and with 1 then 2 CPUs? (I just tested with 1 CPU here, but only have a 25F, so can't test with 2).

enjoy-digital, 2022-03-23:
@essele: I just tested a very similar configuration on the ECPIX5: 2 CPUs with similar FPGA/DDR3. The initial issues you had were:
- The missing DM remapping on the OrangeCrab when using the direct LiteDRAM interfaces (fixed with https://github.com/litex-hub/litex-boards/commit/eb8657f515fe63ced13c487f7da537451d89d993).
- ECP5/DDR3 memtest was randomly failing (fixed with https://github.com/enjoy-digital/litex/commit/dd7a04a5c07068e3f10288c049ebb15f97d51c80).
So things should now be working on the OrangeCrab too.

One thing that still need to be investigated is the behavior with -`-with-wishbone-memory` and `--cpu-count > 1` but I'll investigate this separately (and this is not a common use-case).

closed 2022-03-23 by enjoy-digital