litex-hub/linux-on-litex-vexriscv #256: VC707 board support

Logicwax, 2021-12-04:
How hard would it be able to get this project running on VC707 boards?

enjoy-digital, 2022-01-04:
Hi @Logicwax,

the VC707 is supported by LiteX-Boards, so adding support for it should only be matter of adding the definition in `make.py`. To run Linux-On-LiteX-Vexriscv on an FPGA you only need a fabric large enough + UART + 32MB of RAM. The main difficulty is generally to get the DRAM working, but DDR3 SDRAM is already there on the VC707.

Logicwax, 2022-01-06:
I tried that already: https://github.com/Logicwax/linux-on-litex-vexriscv/blob/271ac243ca6ff5aacb40bed76c23fddfff17bc5d/make.py#L175-L192

but I run into issues that I'm not sure what to do about:
```
Traceback (most recent call last):
  File "./make.py", line 797, in <module>
    main()
  File "./make.py", line 740, in main
    soc.add_sdcard()
  File "/home/logicwax/litex/litex/soc/integration/soc.py", line 1591, in add_sdcard
    self.submodules.sdphy  = SDPHY(sdcard_pads, self.platform.device, self.clk_freq, cmd_timeout=10e-1, data_timeout=10e-1)
  File "/home/logicwax/litesdcard/litesdcard/phy.py", line 615, in __init__
    self.submodules.io = sdphy_cls(clocker, sdpads, pads)
  File "/home/logicwax/litesdcard/litesdcard/phy.py", line 543, in __init__
    io  = pads.data[i],
AttributeError: 'Record' object has no attribute 'data'
```

Logicwax, 2022-01-06:
Ignore my previous message, it looks like it does work when I fix what I assume is a typo.  I've made two PR's to fix this:
https://github.com/litex-hub/linux-on-litex-vexriscv/pull/263
https://github.com/litex-hub/litex-boards/pull/318

Logicwax, 2022-01-06:
My last issue is that I don't see eth0 interface.   Is there something else I need to do to be able to see the ethernet interface?

enjoy-digital, 2022-01-06:
@Logicwax Thanks for the PR, that's already a first good step if you are able to boot Linux on the VC707. It seems the Ethernet support hasn't yet been added to the VC707 target (https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/xilinx_vc707.py), probably because we don't yet have SGMII support in LiteEth. If you really need Ethernet, an option could be to use a RJ45 to SFP module and use the K7 1000BASE-X PHY: https://github.com/enjoy-digital/liteeth/blob/master/liteeth/phy/k7_1000basex.py.

Logicwax, 2022-02-02:
Thanks!   I'll try that if its my only option for VC707.    Do I need to enable anything to get that working?   I'm not sure how to use a module from kc705 board for the vc707.

enjoy-digital, 2022-02-02:
You can find an integration example of the A7 1000BASE-X PHY here: https://github.com/litex-hub/litex-boards/blob/fccb952c4b1a878473d751a5a6c4d4da1949b635/litex_boards/targets/xilinx_ac701.py#L94-L120 Integration of the K7 1000BASE-X PHY will be very similar.

enjoy-digital, 2022-03-18:
The initial question has been answered and VC707 is now supported, thanks @Logicwax. We can probably close this. Feel free to keep us updated with progression on K7 1000BASE-X PHY integration if you go further.

closed 2022-03-18 by enjoy-digital