litex-hub/linux-on-litex-vexriscv #258: Running sim.py hangs after OpenSBI

cbrune, 2021-12-19:
First off - big thanks to you all for doing this and for taking the time to read this.
I'm new to this and am excited to see things work :)

I'm seeing the same symptom as #173 from last year.  I'm using the latest master branch, commit a6719b9468d043c59b827d16a149360bf4db6f4c.

versions of things:
- verilator: 4.216
- litex master (321b91d551ae58f32c14596985d096d178c32be6)
- the pre-built OpenSBI and Linux images from #164.

I am running `sim.py` with no arguments.

This is what I see:
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS CRC passed (6e93ba71)

 Migen git sha1: ac703010eaa0
 LiteX git sha1: 321b91d551ae

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		32KiB
SRAM:		8KiB
L2:		0KiB
SDRAM:		65536KiB 32-bit @ 100MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

OpenSBI v0.8-1-gecf7701
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : LiteX / VexRiscv-SMP
Platform Features   : timer,mfdeleg
Platform HART Count : 8
Boot HART ID        : 0
Boot HART ISA       : rv32imas
BOOT HART Features  : time
BOOT HART PMP Count : 0
Firmware Base       : 0x40f00000
Firmware Size       : 124 KB
Runtime SBI Version : 0.2

MIDELEG : 0x00000222
MEDELEG : 0x0000b101
```

and then it hangs.

If you need more info let me know.

Dolu1990, 2021-12-20:
Hi ^^

Which image of linux did you tried ?

cbrune, 2021-12-20:
I tried the Image from the linux_2021_03_29.zip (5.12-rc4) link: https://github.com/litex-hub/linux-on-litex-vexriscv/files/6220823/linux_2021_03_29.zip
```
$ sha256sum images/Image
5bc329c79d0f9c0944ca628156adf66bb578f5d9997c2b780b1f6fb5eeceb6d0  images/Image
$ sha256sum images/rootfs.cpio
f72849001b91b59ad069448bc7d9d81bcf2e0d03fda5912b30b9a8716a314324  images/rootfs.cpio
```

cyntem, 2021-12-24:
I got same issue. Ubuntu 21.04, I tried all available linux images and opensbi. Is any solution available?
```

[xgmii_ethernet] loaded (0x55ada2718090)
[serial2tcp] loaded (0x55ada2718090)
[ethernet] loaded (0x55ada2718090)
[clocker] loaded
[serial2console] loaded (0x55ada2718090)
[spdeeprom] loaded (addr = 0x0)
[gmii_ethernet] loaded (0x55ada2718090)
[clocker] sys_clk: freq_hz=1000000, phase_deg=0

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS CRC passed (0c9d48a2)

 Migen git sha1: ac70301
 LiteX git sha1: a6ed4c5c

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		32KiB
SRAM:		8KiB
L2:		0KiB
SDRAM:		65536KiB 32-bit @ 100MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

OpenSBI v0.8-2-ga9ce3ad
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : LiteX / VexRiscv-SMP
Platform Features   : timer,mfdeleg
Platform HART Count : 8
Boot HART ID        : 0
Boot HART ISA       : rv32imas
BOOT HART Features  : time
BOOT HART PMP Count : 0
Firmware Base       : 0x40f00000
Firmware Size       : 120 KB
Runtime SBI Version : 0.2

MIDELEG : 0x00000222
MEDELEG : 0x0000b101
```

Dolu1990, 2021-12-24:
I will check the sim to see what happen once i'm back home, the linux 5.10 image may give better results ?

Dolu1990, 2021-12-26:
I tried, basicaly, seems to be a DTS / earlycon issue, if you let the sim run veeeery long, sudently, all the printk pop.
Got it to print things early by forcing in the dts the following : 

```
        chosen {
            bootargs = "console=hvc0 earlycon=sbi rootwait root=/dev/ram0";
            linux,initrd-start = <0x41000000>;
            linux,initrd-end   = <0x41800000>;
        };       ```

Dolu1990, 2021-12-26:
So, the change which "broke" the dts came from : 
https://github.com/enjoy-digital/litex/commit/b1b1e92ad03171b9ae823d31a657c271ee2533fc#diff-1768452e8130aa2aa4f0b1ff4b81a29a656cae6491b8078d90f87e8884120d9cR63

earlycon=sbi   is the main thing, i guess we don't have to care about "console=hvc0"

@enjoy-digital What was the reason to switch from earlycon=sbi to earlycon=liteuart ?

enjoy-digital, 2021-12-26:
Thanks @Dolu1990 for looking at this,

the idea was to use the same command for the different LiteX/Linux SoCs: https://github.com/enjoy-digital/litex/pull/1032/commits/ec2f2a6af5e55b4b7ad2766cebe379135c5e047e but that's indeed possible it breaks the early console. Not we we'll notice it that much on hardware, but it can indeed seems to hang in simulation, which is why I initially wanted to switch to earlycon.

I would recommend using @Dolu1990's change for now and I'll have a look later to re-enable it properly.

Dolu1990, 2021-12-26:
Ahhh, it make sense ^^
thanks

cbrune, 2021-12-27:
Confirming adding `earlycon=sbi` to the bootargs does the trick.  Thanks @Dolu1990 and @enjoy-digital for looking into it. I put together a proposed fix in #260 and enjoy-digital/litex#1145

cbrune, 2021-12-29:
Looks like the pre-built Linux images are out of date wrt to enjoy-digital/litex@ec2f2a6.

Rebuilding the Linux image following the buildroot instructions produces a Linux image that does *not* hang in `sim.py`, i.e. the earlycon is working.

This is noticeable both in `sim.py` and on real ULX3S hardware.  Using the older Linux image from https://github.com/litex-hub/linux-on-litex-vexriscv/files/6220823/linux_2021_03_29.zip, there is a noticeable lag before the Linux console begins printing.

Using a fresh Linux image from buildroot, `earlycon` works for both `sim.py` and real HW.

Minor note on the buildroot -- out of the box it complained about a kernel header version mismatch (5.15.x vs. 5.14.x), but you can run `make menuconfig` and navigate to `Toolchain -> Custom kernel headers series` and set it to `5.14.x`.

enjoy-digital, 2022-01-04:
Thanks @cbrune, I'll try to look at this soon and update things.

enjoy-digital, 2022-03-23:
Sorry for the delay, the prebuilt images have been updated and the issue is not seen with these images.

closed 2022-03-23 by enjoy-digital